FastLED/FastLED #1471: FastLED and Arduino TinyUSB issues on RP2040

solidDoWant, 2023-02-02:
Hello! I have a strange issue when using **both** FastLED and the [Arduino TinyUSB port](https://github.com/adafruit/Adafruit_TinyUSB_Arduino) on a Raspberry Pi Pico (RP2040). When including both the FastLED and TinyUSB library some LEDs attached will be a random color. They are usually towards the beginning of the string, but occasionally will be a few (I'm guessing 3-7) LEDs in. Here's a pic of the issue:
[image]

The strings having the issue, the position of the wrong color LEDs, and what colors they are varies update (`show` call) to update, as well as unrelated changes to code. For example, if I add a dead simple counter (`uint8_t counter = 0;` in a global scope, and `counter++;` every `update` loop, the color, position, strip, and frequency of the issue may change.

I have tested the following to verify that the hardware is not the issue:
* Swapped out MCU boards (Raspbery Pi Picos all bought from Digi-Key sealed in cut tape)
* Tried with and without a level shifter, multiple models
* With and without the "breakout" PCB I built
* Tried other LED strips from multiple manufacturers and lengths
* Used other controllers from various manufacturers

Here's some example code demonstrating the issue:
```
#include <FastLED.h>
#include <Adafruit_TinyUSB.h>

#define LED_COUNT 35

CRGB ChannelLEDs[LED_COUNT];
void setup() {
  FastLED.addLeds<WS2812B, 0, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 1, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 2, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 3, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 4, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 5, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 6, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 7, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 8, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 9, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 10, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 11, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 12, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 13, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 14, GRB>(ChannelLEDs, 0, LED_COUNT);
  FastLED.addLeds<WS2812B, 15, GRB>(ChannelLEDs, 0, LED_COUNT);
  fill_solid(ChannelLEDs, LED_COUNT, CRGB::Red);
}

void loop() {
  FastLED.show();
}
```

If you comment out the `#include <Adafruit_TinyUSB.h>` line the issue goes away.

Note that I'm using the same CRGB array for all controllers, but the issue happens whether or not I use the same array for all of them. The color is arbitrary too - I can even turn all the LEDs off (set color to black) and I will _still_ have the issue.
[image]

Any idea what what the issue could be? I am guessing that both the libraries use the same hardware resource for something, but I have no idea where to start looking.

solidDoWant, 2023-02-02:
I _think_ this issue is somehow related to the PIO state machines. From what I can tell the TinyUSB code for the RP2040 makes use of some of the state machines (not sure how many). It appears that setting `#define FASTLED_RP2040_CLOCKLESS_PIO 0` before importing the FastLED library fixes the issue, but I am unsure what the impact is. Can anybody comment on the tradeoff of setting the macro to 0?

ademuri, 2023-08-25:
I'm not completely sure, but here's my understanding: with `FASTLED_RP2040_CLOCKLESS_PIO` set to `1`, FastLED will use the PIO state machines for writing out the LEDs. This means that the output process will happen asynchronously - that is, in the background, and so your application code can still run during this. With it set to `0`, it will do the output process in software. This means that the call to `FastLED.show()` or similar will block for as long as it takes to write out the LEDs. This is approximately 30 LEDs/1ms.

Dantali0n, 2023-08-27:
This is caused by the stream of LED data never being interrupted for more than 50ms which is required by ws2812b LEDS to determine 'end of data'. 

So by putting `.show()` directly in a loop without any delay in between, its to the LEDS as if you are sending data over an infinitely long strand of LEDS.

Simply but `delay(25);` after `.show()`

zackees, 2024-09-20:
> This is caused by the stream of LED data never being interrupted for more than 50ms which is required by ws2812b LEDS to determine 'end of data'. 
> 
> 
> 
> So by putting `.show()` directly in a loop without any delay in between, its to the LEDS as if you are sending data over an infinitely long strand of LEDS.
> 
> 
> 
> Simply but `delay(55);` after `.show()`

It's 50 microseconds

zackees, 2026-09-06:
Status update — the diagnostics half of this landed on `master` in 5ec9375a24 (#4163).

To be precise about what was and was not wrong, since the thread has a few competing theories:

**@solidDoWant's original diagnosis was correct.** TinyUSB claims PIO state machines, and so does FastLED. That contention is the cause.

**The contention itself is already handled** and has been for a while: `ClocklessController` acquires its resources through `RpPioDmaResourceManager`, which uses the SDK claims system via non-panicking `pio_claim_unused_sm(pio, false)`. If nothing is free it releases whatever it already took and returns. FastLED does not steal TinyUSB's state machines, so the random-color corruption should not reproduce on current master.

**What was still wrong is that it told you nothing.** All three claim-failure paths returned silently, so the symptom was a dark strip with zero diagnostic output — which is most of why this issue stayed confusing for three years. They now warn, naming the pin:

```
[RP PIO] no free PIO state machine for pin 2; another library holds them all.
         Set FASTLED_RP2040_CLOCKLESS_PIO 0 to bit-bang instead.
```

That last part is the workaround @solidDoWant found empirically, and @ademuri's explanation of the tradeoff in this thread is right: `FASTLED_RP2040_CLOCKLESS_PIO 0` moves output to blocking bit-bang instead of asynchronous PIO+DMA.

For completeness on the other theory raised here: the 50 µs WS2812 latch @Dantali0n described is a real constraint, but it is a separate issue from this one — it does not depend on TinyUSB being linked, and @zackees already corrected the 50 ms figure to 50 µs.

I have not reproduced the original fault on hardware — no RP2040/RP2350 is currently enumerated on this bench — and reproducing it needs a TinyUSB sketch that actually exhausts the state machines. So I am leaving this open rather than closing it on inference. If anyone still hitting it can retest on master, the warning should now say plainly whether contention is what you are looking at.

Generated with [Claude Code](https://claude.com/claude-code)

https://claude.ai/code/session_01KkufoNxfnNRU9psT3R9F51

zackees, 2026-09-08:
**Now reproduced on hardware.** My previous comment said I had not run this on silicon because no RP board was enumerated. One is now attached (RP2350W), so here is the actual result.

### The contention handling is correct

I starved every PIO state machine directly rather than linking TinyUSB — what matters for this bug is that the state machines are owned by *someone else*, not who owns them. Then asked FastLED to build a clockless controller on the starved block:

| Check | Result |
|---|---|
| `starvedSms` / `freeAfterStarve` | 10 / **0** — block fully starved |
| `noTheft` | **True** — FastLED claimed nothing |
| `claimsIntact` | **True** — the starving owner keeps all 10 |
| `survivedShow` | **True** — `show()` safe on the disabled controller |

So @solidDoWant's original diagnosis was right about the cause, and the claims-based acquisition does now hold the line: FastLED does not take a state machine TinyUSB owns.

Incidentally, `baselineFreeSms` is 11 of 12 on a Pico 2 W — CYW43 WiFi already holds one before any FastLED code runs.

### But the same run found a real leak, and it is a cause of this issue

`ClocklessController` has **no destructor** on the RP PIO path. Every `release*` call in `clockless_rp_pio.h` is an error-path rollback inside `init()`. On a *successful* init it permanently holds the PIO state machine, its instruction-memory allocation, a DMA channel, the pin claim, and the `dma_buf` allocation.

Worse than a leak: `dma_chan_waits[dma_channel]` keeps pointing at the destroyed controller's `mWait`, and the shared DMA ISR dereferences it on every completion.

**This produces the exact symptom reported here.** A sketch that adds and drops controllers starves its own PIO block, and then the driver correctly reports "no free PIO state machine" — contention FastLED created itself, with no other library involved. That is a second, independent route to a dark strip, and it would look identical from the outside.

Measured on the control leg (state machines free, so the claim must succeed and then be returned):

```
before: control.claimedOne=True, control.released=False, noLeak=False
```

### Fix

#4214 adds the destructor and the hardware test. After it:

```
RESULT: RPC smoke PASS (discovery, help, ping, payload, status, drivers,
        testNoSerial, RP concurrency, RP PIO contention, error handling)
```

The test is wired into the existing `--rpc-smoke` RP path so it runs on every RP bench pass, and has two legs — a control leg with state machines free (must claim one and give it back) and the starved leg (must claim nothing). Either alone proves nothing: without the control leg, "FastLED claimed nothing" is indistinguishable from the driver being compiled out. The control leg is what caught the leak.

### Still not reproduced

The *original* 2023 report — random-coloured LEDs with TinyUSB linked, on an RP2040. I have an RP2350, not an RP2040, and I did not link TinyUSB. What is now evidenced is that the resource arbitration behaves correctly under starvation and that a leak existed which could starve it. If anyone still hitting the original corruption can retest on master, the warning added in #4163 should say plainly whether contention is what you are looking at.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

https://claude.ai/code/session_01KkufoNxfnNRU9psT3R9F51

zackees, 2026-09-09:
## Update: the earlier evidence had a hole, now closed

My previous comment reported the contention handling as correct on hardware. Review of the PR found that result was **not sufficient on its own**, and the fix is worth recording here because it changes what the evidence proves.

The old result asserted `noTheft`, `claimsIntact`, `survivedShow` and `noLeak` under a fully starved block. Every one of those holds **vacuously** if the clockless backend is simply not compiled in: a driver that does not exist also claims nothing, also leaves the other owner's claims intact, and also survives `show()`. The test could not distinguish "FastLED correctly declined" from "FastLED was never there".

The test now runs a **control leg first**, with state machines available:

```
baselineFreeSms: 11
control:  proxyValid=True  freeDuring=10  claimedOne=True  freeAfter=11  released=True
```

`freeDuring=10` against a baseline of 11 — FastLED took exactly one state machine when one was free, and returned it. That establishes the driver is present and functioning before the starved case is interpreted, and `control.proxyValid`/`control.claimedOne` are now terms in the verdict rather than reported fields.

With that in place, the starved result means what it claims:

```
contention: starvedSms=11  freeAfterStarve=0  noTheft=True
            claimsIntact=True  survivedShow=True  freeAfter=11
noLeak: True
```

**3 runs, 3 PASS**, identical each time:

```
RESULT: RPC smoke PASS (discovery, help, ping, payload, status, drivers,
        testNoSerial, RP concurrency, RP PIO contention,
        RP PIO0+PIO1 resources, error handling)
```

### Scope, stated plainly

- The board is an **RP2350W**, not an RP2040. The contention mechanism is the same — PIO state machines owned by other code — and the test starves them directly rather than through TinyUSB, since what matters is that they are owned by someone else, not who owns them. But this is not an RP2040 measurement, and I am not presenting it as one.
- TinyUSB itself is not in the loop. Direct starvation reproduces the resource condition, not the specific interaction with that library.

Both caveats argue the same way: this closes the "FastLED behaves correctly when the PIO block is fully claimed" question with live evidence, and leaves "RP2040 + Adafruit TinyUSB specifically" resting on the mechanism being identical.

Fix and test: #4214.

zackees, 2026-09-11:
## Regression check: the contention fix holds on current master

Incidental confirmation rather than new investigation — the PIO contention test rides inside `--rpc-smoke`, which I have been running repeatedly today on an RP2350W (`2DCB876B587EA334`) while working unrelated issues.

**7 separate `--rpc-smoke` runs on 2026-09-11 against current master, all naming the contention test in the pass line, zero failures:**

```
RESULT: RPC smoke PASS (discovery, help, ping, payload, status, drivers, testNoSerial,
                        RP concurrency, RP PIO contention, RP PIO0+PIO1 resources,
                        error handling)
```

Those runs span several firmware builds and branches over the day, so the fix from #4214 is holding across rebuilds rather than in one lucky configuration.

Worth noting the contrast with four days earlier: logs from 09-07 in the same scratch directory show `RESULT: RPC smoke FAIL: testRpPioParallelResources returned unsuccessful response` five times. I surfaced those earlier today while tallying gate stability and flagged them as a possible live concern — checking the timestamps showed they predate the fix, and every run since has passed. Recording that here so the 09-07 failures are not mistaken for a current regression by anyone reading those logs.

The scope caveats in the previous comment are unchanged and still the right ones: this is an RP2350W, TinyUSB is not in the loop, and the RP2040 + Adafruit TinyUSB combination specifically still rests on the mechanism being identical rather than on a measurement. I have no RP2040 attached, so I cannot close that gap.