FastLED/FastLED #1473: Platform isn't recognized

325racer, 2023-02-04:
Hopin get some help to sort out a Platform isn't recognized issue.

I'm using a Longan RP2040 Canbed board and some Amazon WS2812B Leds.

http://docs.longan-labs.cc/1030018/ and https://www.amazon.com/dp/B01CDTEBKA?psc=1&ref=ppx_yo2ov_dt_b_product_details

```astLED-master\src/led_sysdefs.h:61:2: error: #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
   61 | #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."```

325racer, 2023-02-05:
I have attempted to add the following to the led_sysdefs.h but continute to get the same error.

``#elif defined(ARDUINO_ARCH_RP2040) || defined(LONGAN_RP2040) || defined(RP2040) || defined(CANBed_2040) || defined(CAN_BUS_Shield) || defined(Aruino_CAN_BUS_MCP2515) // not sure a pico-sdk define for this
// RP2040 (Raspberry Pi Pico etc)
#include "platforms/arm/rp2040/led_sysdefs_arm_rp2040.h"``

zackees, 2024-09-19:
RP2040 generic support was put in a month ago. Closing this as stale.

closed 2024-09-19 by zackees