FastLED/FastLED #1474: Driving WS2812B Leds with ESP32, palette is not continuous after addressing more than 63 leds

McMathewson, 2023-02-06:
#include <FastLED.h>
#define NUM_LEDS1 64
#define LED_PIN1 22
CRGB leds1[NUM_LEDS1];
uint8_t paletteIndex = 0;

DEFINE_GRADIENT_PALETTE(redLight){
0, 255, 0, 0,
//102, 0, 0, 0,
127, 0, 0, 0,
//153, 0, 0, 0,
255, 0, 0, 0
};

CRGBPalette16 redLightPalette = redLight;

void setup() {
FastLED.addLeds<WS2812B, LED_PIN1, GRB>(leds1, NUM_LEDS1);
FastLED.setBrightness(255);
FastLED.clear(true);
}
void loop(){
fill_palette(leds1, NUM_LEDS1, paletteIndex, 255 / NUM_LEDS1, redLightPalette, 255, LINEARBLEND);

EVERY_N_MILLISECONDS_I(speedTimer, 1) {
paletteIndex++;
}
speedTimer.setPeriod(20); //use purpleVal for GPS
FastLED.show();
}

Im relatively new to coding on the ESP32 and I cant find anybody else with this specific issue but when I address more than 63 leds on my strip I have a sort of delay between the start and end of the strip. I have only had issues when using the fill_palette function.

I have a circular strip of WS2812B leds which total to 99 leds which id like to run a continuous moving palette on for a project.

In the arduino board manager im using version 2.0.6 eps32 board by Espressif Systems and version 3.5.0 of the FastLed library.

Let me know if this is the wrong place to post this issue.

dmadison, 2023-02-11:
The third argument of the `fill_palette()` function is how much to increment the starting index of the palette per LED. If you're using a circular strip you will run into problems where 255 is not evenly divisible by the number of LEDs in the strip and the palette won't be continuous.

[Try using the current master branch of the library](https://github.com/FastLED/FastLED/archive/refs/heads/master.zip) (this function was added after 3.5) and using the `fill_palette_circular()` function instead:

    fill_palette_circular(leds1, NUM_LEDS1, paletteIndex, redLightPalette, 255, LINEARBLEND);

McMathewson, 2023-02-11:
Thank you @dmadison !. This fixed up the problem I was having. Ill let you know if I run into additional problems with this funcition.

closed 2023-02-11 by McMathewson