FastLED/FastLED #1476: Weird Fastled behaviour on nodeMCU

illusionmanager, 2023-02-08:
I have wifi running on the nodeMCU (just to get the time) and this is part of my code
 ring[8][second] = 0;
    if (second % 2 ) {
       fill_solid(ring[1],ringsize[1],CHSV((hour + minute / 60.0) *256 / 12.0 , 255,255));
       fill_solid(ring[0],ringsize[0],CHSV((hour + minute / 60.0) *256 / 12.0 , 255,255));
    } else {
      fill_solid(ring[1],ringsize[1],CHSV((hour + minute / 60.0) *256 / 12.0 , 255,40));
       fill_solid(ring[0],ringsize[0],CHSV((hour + minute / 60.0) *256 / 12.0 , 255,40));
    }
    FastLED.show();
that works just fine, but when I change the code and make the if statement always true, it doesn't work reliable anymore and most of the time skips 5 seconds in the animation. 
It is just an example. Other code often only animates the first led of the strip and when I change some totally unrelated part of the code just to animate some other leds as well, it suddenly starts working. It basically prevents me from making the animations I really want. As mentioned the code above is just an example, where I make very minimal change to the code and it goes from behaving fine to behaving badly.

guillaume-dorczynski, 2023-02-09:
>when I change some totally unrelated part of the code just to animate some other leds as well, it suddenly starts working

Hello, it sounds like a memory problem, maybe a buffer overflow somewhere in your code

https://en.wikipedia.org/wiki/Buffer_overflow

illusionmanager, 2023-02-09:
i don't think it is a buffer overflow. As a quick test I sized every array, including the array that holds the leds, way bigger, and the behaviour is exactly the same. Changing the compiler optimization option to -O1 changes the behaviour, but doesn't resolve it. Even with fairly simple code it goes wrong. Perhaps it has to do with the enabled WiFi on the esp2866. I don't know. I've now changed to the adafruit_neopixel library and that one performs without these issues. Too bad. I liked the FastLed lib.

zackees, 2023-02-09:
This isn't complete code and you are using weird constructs for the led layout.

If you are using just one pin to write out the leds then just use a single linear array of leds (not a multi dimensional array). See if that fixes your problem.

illusionmanager, 2023-02-09:
because it is organized in rings I have:
 const int ringsize[] = {1,8,12,16,24,32,40,48,60};
  ring[0] = &leds[0];
  ring[1] = ring[0]+ringsize[0];
  ring[2] = ring[1]+ringsize[1];
  ring[3] = ring[2]+ringsize[2];
  ring[4] = ring[3]+ringsize[3];
  ring[5] = ring[4]+ringsize[4];
  ring[6] = ring[5]+ringsize[5];
  ring[7] = ring[6]+ringsize[6];
  ring[8] = ring[7]+ringsize[7];
I am using a single array leds[NUM_LEDS]. It's just a way to more easily tell which ring I'm using.

illusionmanager, 2023-02-09:
CRGB leds[NUM_LEDS]; 
should define the array for the leds (I don't care about the contents at this point)
ring[0] = &leds[0]; should be valid or not?? 

I do execute:
```
  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setBrightness(20); 
  FastLED.setMaxRefreshRate(240);
  fill_solid(leds,NUM_LEDS,CRGB(0,0,0));
```
before filling it with other content:
fill_solid(ring[0],ringsize[0],CHSV((hour + minute / 60.0) *256 / 12.0 , 255,255)); 
sets all of the content of ring[0] to the given value.
all the other ring's content is set in similar statements.

zackees, 2023-02-09:
Change:

```C++
CRGB leds[NUM_LEDS];
CRGB *ring[9];
```

to

```
CRGB leds[NUM_LEDS] = {0}
CRGB *ring[9] = {0};
```

And it will zero initialize your values.

closed 2024-10-12 by zackees