FastLED/FastLED #1477: Reducing memory usage for LEDs colors

guillaume-dorczynski, 2023-02-09:
Hello :)

( Sorry if this was already suggested and declined, I did a small search but there are so many issues... )

Sometimes, it's unecessary to have an array of CRGB for the LEDs

As an extreme example, let's say I have a strip of 1000 LEDs, and I want only white and black LEDs. I have to create an array of 1000 CRGB, that is a lot of memory wasted, because the LEDs will only have two colors and that can be done with one bit per LED, so an array of 1000 bits (125 bytes), versus 3000 bytes for an equivalent CRGB array.

The idea would be to have another way of setting the LEDs colors, without having to use a CRGB array parameter in `addLeds`, instead have an user-defined callback function returning a CRGB color, and called <leds_count> times by FastLED.show() instead of looking into a CRB array

With the above example, it could look like this:
```
uint8_t leds[125];

void setup()
{
	FastLED.addLeds<NEOPIXEL, 12>( colorCallback, 1000 );
}

void loop()
{
	// user fill 'leds' array as he wants...
	
	FastLED.show();
}

CRGB colorCallback( size_t led_id )
{
	// return White if bit is set, or Black if bit is clear
	return ( ( leds[ led_id / 8 ] >> ( led_id % 8 ) ) & 1 ) == 1 ? CRGB::White : CRGB::Black;
}
```

In another case, I may need 16 colors, and use 4 bits per LED and a lookup CRGB array, still ~6 times less memory than a big CRGB array.

I have another idea to further reduce memory usage in certain cases (for example, when only one pixel of the strip is lit), but first let's see what you think about it :)

zackees, 2023-02-09:
1000 lights = 3000 bytes = 3k memory.

Most MCU's have much much more than 3k of memory. Are you actually using an MCU that doesn't have this much memory? Maybe consider switching to an ESP32 which does and is extremely cheap.

The problem here is that FastLED supports clockless chips like the WS2812 and as such there are tight timing requirements and therefore the frame buffer needs to be declared up front so that you don't have to deal with this complexity of bitbanging the leds out to whatever flavor of led chipset you are using.

So if 3k is too much, then you should just modify one of the other led libraries to give you the streaming behavior that you want.

zackees, 2023-10-03:
This should be closed.

closed 2023-10-11 by guillaume-dorczynski