FastLED/FastLED #1480: UCS1912 support

pelim, 2023-02-20:
I did't find that much information on UCS1912 and if it is compatible to WS2811 

here is a translated version of the chinese specs https://docdro.id/iKyrKek

marmilicious, 2023-02-20:
Here's a data sheet for WS2811 to compare.
http://www.world-semi.com/DownLoadFile/129
or
https://cdn-shop.adafruit.com/datasheets/WS2811.pdf

Timings look a bit different so not sure it would be compatible with WS2811, but if you have some UCS1912 LEDs on hand you can certainly try it out. If it doesn't work it won't hurt anything, it just won't light up.

Thank you for providing the link to the data sheet.

zackees, 2024-09-13:
Compile support added. Timings were set from the datasheet. No idea if it actually works.

https://github.com/FastLED/FastLED/commit/5616c05d822443605189a4eeffb21108f2b11000

closed 2024-09-13 by zackees