FastLED/FastLED #1481: Pico-SDK CMake RP2040 Support

cinderblock, 2023-02-20:
I've been trying to get FastLED working with the PicoSDK for a custom RP2040 based board.

I've got it mostly compiling but I'm getting what I think is a bug in the m0 clockless assembly implementation.

I was originally working on this build chain in a [pico-sdk issue](https://github.com/raspberrypi/pico-sdk/issues/1234), if you care to look at that.
To summarize:

The main `CMakeLists.txt`, setup mostly as the pico-sdk prescribes:

```cmake
cmake_minimum_required(VERSION 3.13)

include(lib/SDK/pico_sdk_init.cmake)

project(my_project)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

# TODO: Use `add_subdirectory(src)` instead of the following
add_executable(my_executable
    src/main.cpp
)

target_link_libraries(my_executable FastLED)

target_link_libraries(my_executable pico_stdlib)

target_link_libraries(my_executable
    pico_stdio
    hardware_spi
    # hardware_uart
    hardware_i2c
    hardware_gpio
)
 
add_subdirectory(lib/FastLED)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(my_executable)
```

A `lib/FastLED/CMakeLists.txt` for the FastLED "`INTERFACE`" library:

```cmake
add_library(FastLED INTERFACE)

target_sources(FastLED INTERFACE
    compat/FastLED.cpp
)

# Required pico SDK libraries
target_link_libraries(FastLED INTERFACE
    hardware_sync
    hardware_gpio
    hardware_clocks
    hardware_dma
    hardware_pio
    cmsis_core
    pico_malloc
    pico_time
)

# Ensure our compatiblity header is in the include search path before FastLED's sources
target_include_directories(FastLED INTERFACE
    compat/include
)

target_compile_definitions(FastLED INTERFACE
    # FASTLED_RP2040_CLOCKLESS_PIO=1
)
```

Notably, since FastLED expects to be built in the Arduino environment, some functions need to be declared (and defined inline).
We can declare and implement what's needed with `compat/Adapter.hpp`:

```hpp
#include <stdlib.h>
#include <pico/time.h>
#include <RP2040.h>

// FastLED tests for this to enable RP2040 platform
#define ARDUINO_ARCH_RP2040

static inline uint32_t micros(void) { return to_us_since_boot(get_absolute_time()); }
static inline uint32_t millis(void) { return to_ms_since_boot(get_absolute_time()); }
static inline void yield(void) {};
```

*Note: include guards shouldn't be needed since this should only be included by the two compatibility files*

Instead of building the FastLED sources directly with `-include compat/Adapter.hpp`, I build a single `compat/FastLED.cpp` that brings in the compatibility layer and all the sources into a single translation unit. This was easier and, imho, cleaner than setting compile flags in CMake for only specific files:

```cpp
#include "Adapter.hpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/bitswap.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/colorpalettes.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/colorutils.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/FastLED.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/hsv2rgb.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/lib8tion.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/platforms.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/noise.cpp"

#undef FASTLED_INTERNAL
#include "../FastLED/src/power_mgt.cpp"
```

When library consumers `#include <FastLED.h>`, they need to include a shim first. So we make `compat/include/FastLED.h`:

```c
#pragma once

// This file is used to include the FastLED library in a way that is compatible with the Pico SDK

// Include the compability layer
#include "../Adapter.hpp"

// Prevent warnings about SPI support
#define FASTLED_INTERNAL

#include "../../FastLED/src/FastLED.h"
```

----

Onto the issue at hand.

After setting up CMake to build FastLED, I'm getting an error whenever I include `FastLED.addLeds<NEOPIXEL, 6>(leds, NUM_LEDS);` in my `main()`:

```log
/usr/bin/arm-none-eabi-g++ -DLIB_CMSIS_CORE=1 -DLIB_PICO_BIT_OPS=1 -DLIB_PICO_BIT_OPS_PICO=1 -DLIB_PICO_DIVIDER=1 -DLIB_PICO_DIVIDER_HARDWARE=1 -DLIB_PICO_DOUBLE=1 -DLIB_PICO_DOUBLE_PICO=1 -DLIB_PICO_FLOAT=1 -DLIB_PICO_FLOAT_PICO=1 -DLIB_PICO_INT64_OPS=1 -DLIB_PICO_INT64_OPS_PICO=1 -DLIB_PICO_MALLOC=1 -DLIB_PICO_MEM_OPS=1 -DLIB_PICO_MEM_OPS_PICO=1 -DLIB_PICO_PLATFORM=1 -DLIB_PICO_PRINTF=1 -DLIB_PICO_PRINTF_PICO=1 -DLIB_PICO_RUNTIME=1 -DLIB_PICO_STANDARD_LINK=1 -DLIB_PICO_STDIO=1 -DLIB_PICO_STDIO_UART=1 -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_UTIL=1 -DPICO_BOARD=\"pico\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_TARGET_NAME=\"my_executable\" -DPICO_USE_BLOCKED_RAM=0 -I../lib/SDK/src/common/pico_stdlib/include -I../lib/SDK/src/rp2_common/hardware_gpio/include -I../lib/SDK/src/common/pico_base/include -Igenerated/pico_base -I../lib/SDK/src/boards/include -I../lib/SDK/src/rp2_common/pico_platform/include -I../lib/SDK/src/rp2040/hardware_regs/include -I../lib/SDK/src/rp2_common/hardware_base/include -I../lib/SDK/src/rp2040/hardware_structs/include -I../lib/SDK/src/rp2_common/hardware_claim/include -I../lib/SDK/src/rp2_common/hardware_sync/include -I../lib/SDK/src/rp2_common/hardware_irq/include -I../lib/SDK/src/common/pico_sync/include -I../lib/SDK/src/common/pico_time/include -I../lib/SDK/src/rp2_common/hardware_timer/include -I../lib/SDK/src/common/pico_util/include -I../lib/SDK/src/rp2_common/hardware_uart/include -I../lib/SDK/src/rp2_common/hardware_resets/include -I../lib/SDK/src/rp2_common/hardware_clocks/include -I../lib/SDK/src/rp2_common/hardware_pll/include -I../lib/SDK/src/rp2_common/hardware_vreg/include -I../lib/SDK/src/rp2_common/hardware_watchdog/include -I../lib/SDK/src/rp2_common/hardware_xosc/include -I../lib/SDK/src/rp2_common/hardware_divider/include -I../lib/SDK/src/rp2_common/pico_runtime/include -I../lib/SDK/src/rp2_common/pico_printf/include -I../lib/SDK/src/common/pico_bit_ops/include -I../lib/SDK/src/common/pico_divider/include -I../lib/SDK/src/rp2_common/pico_double/include -I../lib/SDK/src/rp2_common/pico_float/include -I../lib/SDK/src/rp2_common/pico_malloc/include -I../lib/SDK/src/rp2_common/pico_bootrom/include -I../lib/SDK/src/common/pico_binary_info/include -I../lib/SDK/src/rp2_common/pico_stdio/include -I../lib/SDK/src/rp2_common/pico_stdio_uart/include -I../lib/SDK/src/rp2_common/pico_int64_ops/include -I../lib/SDK/src/rp2_common/pico_mem_ops/include -I../lib/SDK/src/rp2_common/boot_stage2/include -I../lib/SDK/src/rp2_common/hardware_spi/include -I../lib/SDK/src/rp2_common/hardware_i2c/include -I../lib/FastLED/compat -I../lib/FastLED/FastLED/src -I../lib/SDK/src/rp2_common/hardware_dma/include -I../lib/SDK/src/rp2_common/hardware_pio/include -I../lib/SDK/src/rp2_common/cmsis/stub/CMSIS/Core/Include -I../lib/SDK/src/rp2_common/cmsis/stub/CMSIS/Device/RaspberryPi/RP2040/Include -mcpu=cortex-m0plus -mthumb -O3 -DNDEBUG -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-rtti -fno-use-cxa-atexit -MD -MT CMakeFiles/my_executable.dir/src/main.cpp.obj -MF CMakeFiles/my_executable.dir/src/main.cpp.obj.d -o CMakeFiles/my_executable.dir/src/main.cpp.obj -c ../src/main.cpp
/tmp/ccasthV7.s: Assembler messages:
/tmp/ccasthV7.s:812: Error: invalid offset, value too big (0x00000454)
```

With `--save-temps` I'm able to see the intermediate `.s` file that is being compiled and can inspect around line `812`:

```s
; ...
	.syntax divided
@ 279 "../lib/FastLED/compat/../FastLED/src/platforms/arm/rp2040/../common/m0clockless.h" 1
	  loadleds3 ip, r4, #1 ,r3;  loaddither7 r4, r5, r6, #1;  dither5 r4, r5;  scale4 r4, r6, #12, r3;  adjdither7 r6,r5,#1,#4,r3;  swapbbn1 r2, r4;
@ 0 "" 2
	.thumb
	.syntax unified
	ldr	r6, .L131                     ; <------- Line 812: Error: invalid offset, value too big
	movs	r0, r2
	str	r6, [sp, #8]
	ldr	r6, [r6]
	mov	r2, ip
	str	r6, [sp]
	b	.L124
.L130:
	cmp	r1, #0
	bne	.LCB824
	b	.L129	@long jump
.LCB824:
.L124:
	mov	r10, r2
	mov	r7, r8
	ldr	r2, [sp, #4]
	mov	r6, r9
	.syntax divided
; ...
```

This tells me something from `m0clockless.h` isn't generating valid assembly output...

Looking at that code, I'm not very familiar with ARM assembly and marcos, so I'm not sure what's going on.

In any case, is this a FastLED bug? Or am I doing something stupid? Or is this a "won't fix" because this isn't Arduino?

AlopexXx, 2023-03-02:
It seems like I have the same problem.
But it occurs when I try to debug in PlatformIo.

`C:\Users\XXX\AppData\Local\Temp\ccgTx4nG.s: Assembler messages:
C:\Users\XXX\AppData\Local\Temp\ccgTx4nG.s:4202: Error: invalid offset, value too big (0x000004B0)
*** [.pio\build\pico\src\main.cpp.o] Error 1
======================================================================================================================== [FAILED] Took 2.60 seconds ========================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
pico           FAILED    00:00:02.604`

cinderblock, 2023-03-02:
@AlopexXx I've seen this bug happen when enabling debug symbols. I'm not using any but maybe you are?

Are you able to manually build and capture the intermediate assembly file?

AlopexXx, 2023-03-04:
@cinderblock yes it only happens when I try to debug.
I was able to get the following:
```
@ .pio\libdeps\pico\FastLED\src/platforms/arm/rp2040/../common/m0clockless.h:338:         if ((ticksBeforeInterrupts - currentTicks) > kTicksIn45us) {
	.loc 19 338 36 view .LVU1118
	subs	r3, r5, r3	@ _10, ticksBeforeInterrupts, currentTicks
.LVL420:
	.loc 19 338 36 view .LVU1119
	mov	r8, r3	@ _10, _10
@ .pio\libdeps\pico\FastLED\src/platforms/arm/rp2040/../common/m0clockless.h:338:         if ((ticksBeforeInterrupts - currentTicks) > kTicksIn45us) {
	.loc 19 338 9 view .LVU1120
	ldr	r3, .L166	@ tmp163,                                      ; <- main.s:4714: Error: invalid offset, value too big (0x000004B0)
	cmp	r8, r3	@ _10, tmp163
	bls	.LCB4249	@
	b	.L164	@long jump	@
.LCB4249:
.LVL421:
.L161:
	.loc 19 338 9 view .LVU1121
.LBE1258:
	.loc 19 281 5 is_stmt 1 view .LVU1122
	cmp	r4, #0	@ counter,
	bne	.LCB4256	@
	b	.L165	@long jump	@
.LCB4256:
.LVL422:
.L162:
.LBB1263:
@ .pio\libdeps\pico\FastLED\src/platforms/arm/rp2040/../common/m0clockless.h:314:       );
	.loc 19 314 8 is_stmt 0 view .LVU1123
	ldr	r3, [sp]	@ scratch, %sfp
	ldr	r6, [sp, #4]	@ bn, %sfp
	ldr	r5, [sp, #8]	@ b, %sfp
	mov	r8, r2	@ leds, leds
	mov	r2, ip	@ pData, pData
.LVL423:
	.loc 19 314 8 view .LVU1124
	mov	r1, r10	@ _bitmask, _bitmask
	.syntax divided
```

cinderblock, 2023-03-04:
Interesting. I wonder if debugging is enabling certain code paths that are otherwise disabled/removed with `-ffunction-sections`...

Trojaner, 2023-03-09:
I can confirm the same issue when using [Arduino-Pico](https://arduino-pico.readthedocs.io/en/latest/) in combination with platform.io and FastLED. Works fine if I use `platform = raspberrypi` instead.

```
C:\Users\troja\AppData\Local\Temp\ccoiOoQK.s: Assembler messages:
C:\Users\troja\AppData\Local\Temp\ccoiOoQK.s:1046: Error: invalid offset, value too big (0x00000530)
C:\Users\troja\AppData\Local\Temp\ccoiOoQK.s:1066: Error: invalid offset, value too big (0x000004EC)
*** [.pio\build\pico\src\main.cpp.o] Error 1
```

hermgreider, 2023-04-30:
Similar problem, no Platform IO. Lastest Ardunio IDE 2.1. Earle Philhower core. FastLED 3.5 does not support. Switched to FastLED master. First had errors with cli and sei in led_sysdefs_arm_rp2040.h. Commented those out (correct?). Warnings about register deprecated and forcing software SPI. But, the main one identical to above:

{standard input}: Assembler messages:
{standard input}:2944: Error: invalid offset, value too big (0x00000530)
{standard input}:2971: Error: invalid offset, value too big (0x000004EC)

hermgreider, 2023-04-30:
> Ah, curious. Don't comment out cli and sei. Works fine. Just ignore all those pesky warning.

miketanct, 2023-08-18:
I'm facing the same error compiling in Arduino IDE but only when more than one pin is set up with FastLED.addLEDs, and only when compiling with any of the arduino-pico repository boards. See my post about it over at the arduino-pico thread - https://github.com/earlephilhower/arduino-pico/discussions/1649#discussion-5532985

cromefire, 2023-10-02:
Does anyone have a solution to this? Not compiling with debugging doesn't seem to work in the Arduino IDE. This kinda reduces my dual RGB controller only to one, which is kinda a shame, because the hardware is certainly capable of 2 channels...

zackees, 2023-10-05:
I think I read recently that cli and sei are defined through arduino for the multi core stuff.

You might want to look there and ask chat gpt.

SeongGino, 2023-11-17:
Hi! I'm testing out implementing FastLED in my project, hoping to simplify use of different possible LEDs, and I'm running into the same problem here building for the RP2040 with the Arduino IDE.

I was able to get it to build (with some warnings provided, some seemingly about overwriting bits from the Arduino.h library) by setting the compile optimizations to `-Os`, and it seems to work fine after the fact.

At least for my project, for the 2040, this is acceptable and causes no noticeable performance impact--but since we would've normally called for an `-O3` opt level before me implementing FastLED, and this library isn't compatible with these at all it seems right now, I do hope this gets resolved sooner rather than later.

(FWIW, current build flags in my implementation only calls `FastLED.addLeds` once, so this isn't with multiple LEDs atm - but since the whole point of me adding FastLEDs is to simplify adding more/external LEDs, it's at least interesting to note.)

JonnyHaystack, 2023-12-16:
Having the same issue using arduino-pico building with PlatformIO (same as #1571), I am using `-O3` optimizations, and I'm only trying to add a single LED strip:
```
CRGB leds[72];
...
FastLED.addLeds<NEOPIXEL, 7>(leds, 72);
```

I would rather not use `-Os`, and I don't even want this assembly code to be built into my firmware because I'm planning to use the RP2040 clockless controller that uses PIO and DMA.

I would've hoped/expected that it would not compile in the common ARM implementation, if I'm using `#define FASTLED_RP2040_CLOCKLESS_PIO 1`

closed 2024-01-15 by kriegsman

coledeck, 2024-01-29:
Hi,

I'm still having this issue, but in a slightly different form possibly.

Originally, I ran into the bug provided, and after updating to the latest version with the fix from #1576, my issues went away, and I was able to compile for 8 strips.

Now, I'm encountering the following:

With 8 identical strips: I cannot compile in debug mode at all `Error: invalid offset, value too big (0x000004B0)`, but in regular build mode it works fine. -Os, -O3, does not matter for either mode.

With 1 strip: I cannot compile **at all**, in any setup, I'm not sure what changed here. My code is relatively complex and does a lot of stuff,but with just 1 strip defined, I cannot compile, regardless of debug, -Os, -O3, or **version of fastLED** - the above fix does not change the result at all. It's slightly different in that I get 2 errors this time:

```
/tmp/ccG7to2c.s: Assembler messages:
/tmp/ccG7to2c.s:4026: Error: invalid offset, value too big (0x00000530)
/tmp/ccG7to2c.s:4046: Error: invalid offset, value too big (0x000004EC)
```

So I'm not sure where to go from here, any help appreciated. I'm on the master branch of arduino-pico for some unrelated issues I was encountering, base Pi Pico, WS2812.

Thanks,
Cole

JonnyHaystack, 2024-01-30:
@coledeck that's weird. Did you definitely add that new define I added in my PR? I am personally only using one (NeoPixel/WD2812) strip, and I have no issues compiling. Debugging works fine too. Can you share a minimal project example to reproduce your issue?

coledeck, 2024-01-30:
@JonnyHaystack Yes I added that define manually, but double checked that it was indeed getting set by default as well. 

Unfortunately it's part of quite a large code base, so it'll take me a bit to see if I can repro with a simpler setup, but the same code does compile and debug fine without FastLED so it's definitely an issue with FastLED+arduino-pico still and not another library.

JonnyHaystack, 2024-02-14:
Based on the fact that you seem to be still getting the errors from the m0 clockless assembly code, you must not have the define in the right place for it to be picked up by the library. How are you setting it? It should be set to 0, which it *isn't* by default.

I don't think 8 strips will work with just the DMA implementation anyway though, because there wouldn't be enough channels... It's ideal to use DMA, and the m0 implementation seems to have a lot of build issues for RP2040, so if you do have the option of changing your hardware setup to use less strips that would be my advice.

PR #1576 (2024-01-15): feat: option to disable M0 clockless fallback on RP2040
Fixes: #1481, #1571

When building for RP2040, at least with -O3 optimizations and the arduino-pico core, the M0 clockless assembly code fails to be assembled for RP2040. This code is only included to be used as a fallback if it fails to claim a DMA channel, which is an undesirable situation to be in anyway.

To keep everyone happy, I've added an option `FASTLED_RP2040_CLOCKLESS_M0_FALLBACK` which is 1 by default but can be set to 0 to disable this fallback and skip compiling `m0clockless.h`. This means the default behaviour remains the s [... cut]