FastLED/FastLED #1482: ESP32, blur2d not supported

m4gg1, 2023-02-26:
im getting the following error when using blur2d like this: 

```
CRGB leds[NUM_LEDS];
const uint8_t MATRIX_Y = 8;
const uint8_t MATRIX_X = 8;
uint8_t amount = 32;

...

blur2d(leds, MATRIX_X, MATRIX_Y, amount);

```

```
/Users/magnus/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/magnus/Documents/Arduino/libraries/FastLED-3.5.0/src/colorutils.cpp:425: undefined reference to `XY(unsigned char, unsigned char)'
/Users/magnus/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: libraries/FastLED-3.5.0/colorutils.cpp.o: in function `blurColumns(CRGB*, unsigned char, unsigned char, unsigned char)':
/Users/magnus/Documents/Arduino/libraries/FastLED-3.5.0/src/colorutils.cpp:440: undefined reference to `XY(unsigned char, unsigned char)'
/Users/magnus/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: libraries/FastLED-3.5.0/colorutils.cpp.o:/Users/magnus/Documents/Arduino/libraries/FastLED-3.5.0/src/colorutils.cpp:445: more undefined references to `XY(unsigned char, unsigned char)' follow
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```

marmilicious, 2023-03-15:
It might be useful if you could include a link to your full program on pastebin.com or gist.github.com

zackees, 2023-03-15:
It's not defined in FastLED and needs to be defined in your application, from the docs:

```C++
/// Forward declaration of the function "XY" which must be provided by
/// the application for use in two-dimensional filter functions.
uint16_t XY( uint8_t, uint8_t);// __attribute__ ((weak));
```

This is because FastLED does not know what your grid of pixels looks like because you define it. The grid could also snake around where the direction of each column is flipped.

This is how it's defined in one of the examples:

```C++
#define NUM_LEDS (kMatrixWidth * kMatrixHeight)

CRGB leds[kMatrixWidth * kMatrixHeight];

uint16_t XY( uint8_t x, uint8_t y)
{
  uint16_t i;

  if( kMatrixSerpentineLayout == false) {
    i = (y * kMatrixWidth) + x;
  }

  if( kMatrixSerpentineLayout == true) {
    if( y & 0x01) {
      // Odd rows run backwards
      uint8_t reverseX = (kMatrixWidth - 1) - x;
      i = (y * kMatrixWidth) + reverseX;
    } else {
      // Even rows run forwards
      i = (y * kMatrixWidth) + x;
    }
  }

  return i;
}
```

This is done as a linker error as linker errors just really suck in C++. It would be nice if there could be a helpful error message attached that says "please define this in your project" but instead you just git this missing function declaration.

closed 2023-08-30 by m4gg1