FastLED/FastLED #1484: fastled support for ws2812-2020

nk63, 2023-03-06:
Controlling a ws2812-2020 smd device with ws2812 preferences cause a changing bahaviour by accidence in lighting. When I compare the datasheets there are a diferent timing for T0H, T0L, T1H, T1L. Especially the higher reset time of >280µs is obviously. 
How I can adjust the timing?

zackees, 2023-03-06:
You'll need to write a custom `CPixelLEDController` for your chipset type.

http://fastled.io/docs/3.1/class_c_pixel_l_e_d_controller.html

nk63, 2023-03-07:
Thanks for the answer, but I already checked the *.h file for any parameters to adjust timing and I cannot find the right place.
For the WS2812 timing is: T0H 350µs, T0L 800µs, T1H 700µs, T1L 600µs (+/- 150 µs), Reset >50µs.
For the WS2812-2020 is: T0H 220-380µs, T0l 580-1000µs, T1H 580-1000µs, T1L 220-420µs, Reset > 280µs.

Normally the 2812 values are inside the 2812-2020 rage, only the reset time is different. Where I can see which timing is
programmed in available controllers. Has another user already integrated 2020 led chips?

zackees, 2023-03-07:
```C++
template <uint8_t DATA_PIN, int T1, int T2, int T3, EOrder RGB_ORDER = RGB, int XTRA0 = 0, bool FLIP = false, int WAIT_TIME = 50>
class ClocklessController : public CPixelLEDController<RGB_ORDER>
```

```C++
/// Frequency multiplier for each clockless data interval.
/// @see Notes in @ref ClocklessChipsets
#define FMUL (CLOCKLESS_FREQUENCY/8000000)
```

```C++
/// WS2812 controller class @ 800 KHz.
/// @tparam DATA_PIN the data pin for these LEDs
/// @tparam RGB_ORDER the RGB ordering for these LEDs
template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class WS2812Controller800Khz : public ClocklessController<DATA_PIN, 2 * FMUL, 5 * FMUL, 3 * FMUL, RGB_ORDER> {};
```

zackees, 2023-03-07:
The weird template stuff is because when this library was made the amount of chips that were able to reliably hit the precise timing requirements for the chipsets was not and instead required assembly code. However, given how everything has speed up by orders of magnitude in the last decade you might be able to get away with just directly subclassing CPIxelLEDController and controlling the microcontroller with `delayMicros()` for now.

nk63, 2023-03-10:
Thank you for these helpfull hints. I changed in file "chipsets.h" the values for an existing type (WS2812) to 
    class WS2812Controller800Khz : public ClocklessController<DATA_PIN, 4, 9, 5, RGB_ORDER>
I couldn't use the metrice with FMUL variables due to this values are to rough. In my case (arduino nano 16MHZ) FMUL=2.
That's why I decided to test different values also with non integer FMUL. Above mentioned values resulted nearly a perfect result. In my case there were after 10 or more cycles some errors. 
Now I will check 2 alternatives. First a better capacitive blocking directly at chip. Second. Usage of a processor with a higher frequency to adjust the cycles better. Maybe the minimum cycle of 62.5ns is to high. 
But I believe that is now a issue of high frequencies, i.e. cabling, capacitive blocking, ...

Thank you for discussion and support. Now I will learn a better understanding of fastLED library. It is a fine thing.

phreakocious, 2023-06-12:
> ```c++
> template <uint8_t DATA_PIN, int T1, int T2, int T3, EOrder RGB_ORDER = RGB, int XTRA0 = 0, bool FLIP = false, int WAIT_TIME = 50>
> class ClocklessController : public [...]

I see occasional color issues with WS2812-2020.  Is this possibly a viable solution for that problem?  Could you provide some details on how to apply this properly?  Thanks!

nk63, 2023-06-12:
Hi phreakocious,
I solved my problem. It was definitively not a timing problem in the driver. I created new pcb's with a near located smd 0603 100nF capacitor (distance to LED lower 2mm). And immediately all libraries including FastLED works properly.
Only the RGB order is sometime to change to GRB. 
Recently I use the setting for WS2811 and WS2812 without changes. 
Nico

closed 2024-09-19 by zackees