FastLED/FastLED #1485: The declaration of the template is not as per the standards.

BabaYaga1221, 2023-03-14:
Recently, I am working on a robotics project where I found that the iterator name is redeclared. As a result, I found that the name of the template class in ``FastLED/src/lib8tion.h 827:39`` is declared as ``q`` which is not as per the standards initialed. I thought that people are facing the same errors as I do. If you would like to adjust the standard, assign this issue to me. Thanks

~Regards

[image]

zackees, 2023-03-14:
What's the error message and what are repro steps including your operating system and other setup parameters?

zackees, 2023-08-29:
This symbols are missing on some toolchain setup. That's why they are defined here. There should be a pre processor define to conditionally compile these symbols

vortigont, 2023-08-31:
actually there are "[Naming Conventions in C++](https://www.geeksforgeeks.org/naming-convention-in-c/)" where classes are declared with first uppercase letter and vars are lowercase'd. Not mandatory for sure, but a respectful style de facto. So would be better to use `Q` or `Qfx`.

BabaYaga1221, 2023-08-31:
@robertlipe is right. While it does shift the issue, my approach was to replace "q" with "qfx" as a temporary fix. I believe it's important to create documentation for the repository and provide clear explanations about these changes.

zackees, 2024-09-19:
I see that q was actually replaces with qfx so this should be resolved and thus closing the issue.

closed 2024-09-19 by zackees