FastLED/FastLED #1487: I2S0O_DATA_OUT0_IDX I2S1O_DATA_OUT0_IDX not declared for ESP32S3

rsiemens77, 2023-03-17:
I was migrating some code from ESP32e to ESP32S3 and getting errors with FastLED3.5 compilation.  I fixed the F_CPU issue already reported.  I also have other missing declarations:

C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:456:34: error: 'I2S0O_DATA_OUT0_IDX' was not declared in this scope

C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:461:34: error: 'I2S1O_DATA_OUT0_IDX' was not declared in this scope

For ESP32, seems these come from gpio_sig_map.h.  Attached are the versions for ESP32 and ESP32S3 for comparison.  Notice the above are missing from the S3 version.  I'm not sure if this is an issue for FastLED or the Arduino-ESP32 component, but thought I'd report it, just in case.

ESP32:

[gpio_sig_map.h.txt](https://github.com/FastLED/FastLED/files/10997996/gpio_sig_map.h.txt)

ESP32S3:

[gpio_sig_map.h.txt](https://github.com/FastLED/FastLED/files/10997997/gpio_sig_map.h.txt)

rsiemens77, 2023-03-17:
I'm not familiar with I2S details, but am wondering if the "SD" declaration versions are equivalent? Like I2S1O_SD_OUT_IDX?

rsiemens77, 2023-03-18:
I tried defining them to the SD versions to see how the compile would go.  There's other missing declarations and problems as below.

In file included from C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/fastled_esp32.h:10,
                 from C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms.h:42,
                 from C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/FastLED.h:62,
                 from C:/esp4/dmcs/main/NeoFactory.hpp:34,
                 from C:/esp4/dmcs/main/Execute.cpp:25:
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:90:17: note: #pragma message: NOTE: ESP32 support using I2S parallel driver. All strips must use the same chipset
 #pragma message "NOTE: ESP32 support using I2S parallel driver. All strips must use the same chipset"
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/fastled_esp32.h:10,
                 from C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms.h:42,
                 from C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/FastLED.h:62,
                 from C:/esp4/dmcs/main/NeoFactory.hpp:34,
                 from C:/esp4/dmcs/main/Execute.cpp:25:
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::i2sInit()':
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:470:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_msb_right = 1;
              ^~~~
              rx_conf
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:471:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_mono = 0;
              ^~~~
              rx_conf
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:472:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_short_sync = 0;
              ^~~~
              rx_conf
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:473:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_msb_shift = 0;
              ^~~~
              rx_conf
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:474:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_right_first = 1; // 0;//1;
              ^~~~
              rx_conf
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:475:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_slave_mod = 0;
              ^~~~
              rx_conf
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:478:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf2'
         i2s->conf2.val = 0;
              ^~~~~
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:479:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf2'
         i2s->conf2.lcd_en = 1;
              ^~~~~
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:480:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf2'
         i2s->conf2.lcd_tx_wrx2_en = 0; // 0 for 16 or 32 parallel output
              ^~~~~
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:481:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf2'
         i2s->conf2.lcd_tx_sdx2_en = 0; // HN
              ^~~~~
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:484:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'sample_rate_conf'
         i2s->sample_rate_conf.val = 0;
              ^~~~~~~~~~~~~~~~
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:485:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'sample_rate_conf'
         i2s->sample_rate_conf.tx_bits_mod = 32; // Number of parallel bits/pins
              ^~~~~~~~~~~~~~~~
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:486:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'sample_rate_conf'
[... cut ...]
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:776:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.rx_start = 0;
              ^~~~
              rx_conf
C:/esp4/frameworks/esp-idf-v4.4.4/components/arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:777:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_start = 0;
              ^~~~
              rx_conf

rsiemens77, 2023-03-28:
I noticed there's this branch for S3, I will take a look at that.

https://github.com/hpwit/I2SClockLessLedDriveresp32s3

netmindz, 2024-01-07:
Did you make any further progress with this? I'm just seeing a similar issue with the I2S code with SmartMatrix

closed 2025-01-13 by zackees