FastLED/FastLED #1489: FastLED is uncompatible with new Arduino Giga R1 Wifi

donemuhendislik, 2023-03-24:
Hi all, last week we needed to carry our project from Arduino mega to new Arduino Giga R1 Wifi. Neither our written FastLED animation code nor FastLED example codes stopped to work. Could we get support about this ? Or is there any update about this  in mean time ?

We have an error while we compile any codes about this.

In file included from C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/FastLED.h:51:0,
                 from C:\Users\softw\AppData\Local\Temp\.arduinoIDE-unsaved2023224-22508-83kkw5.q8q4y\Blink\Blink.ino:5:
C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/led_sysdefs.h:61:2: error: #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
 #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
  ^~~~~

exit status 1

Compilation error: exit status 1

zackees, 2023-03-24:
You have a new variant on the STM32 microcontroller that isn't cataloged yet in the FastLED service.

However there are 4 other similar variants in the code:

```C++
#elif defined(STM32F10X_MD) || defined(__STM32F1__) || defined(STM32F2XX) || defined(STM32F1)
```

You could try defining this in your ino, or forcing the define in your `platform.ini` if using platformio ide plugin.

```C++
#define STM32F10X_MD
#include <FastLED.h>
```

Then if that doesn't compile, try the next define. If it works let us know. If it doesn't work, also let us know.

donemuhendislik, 2023-03-25:
Hi again, thank you so much for your answer. We tried everything that you wrote yesterday... But we have different errors on every solutions... I'm writing them :

*** We tried them in simple instant FastLED examples -> Fire2012.ino (All solutiıons paste at the start of ino)

ADVISED SOLUTIO 1 : 

#define STM32F10X_MD 
include <FastLED.h>

CONSOLE ERROR 1 :

**In file included from C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/led_sysdefs.h:31:0,
                 from C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/FastLED.h:51,
                 from C:\Users\softw\AppData\Local\Temp\.arduinoIDE-unsaved2023225-15000-1wyrog0.l3st\Fire2012\Fire2012.ino:6:
C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/led_sysdefs_arm_stm32.h:6:10: fatal error: application.h: No such file or directory
 #include <application.h>
          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1**

-----------------------------------------------------------------------------------------------------------------------------------

ADVISED SOLUTION 2 : 
#define __STM32F1__
#include <FastLED.h>

CONSOLE ERROR 2 :

In file included from C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/led_sysdefs_arm_stm32.h:18:0,
                 from C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/led_sysdefs.h:31,
                 from C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/FastLED.h:51,
                 from C:\Users\softw\AppData\Local\Temp\.arduinoIDE-unsaved2023225-15000-1wyrog0.l3st\Fire2012\Fire2012.ino:6:
C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/cm3_regs.h:21:3: error: conflicting declaration 'typedef struct CoreDebug_Type CoreDebug_Type'
 } CoreDebug_Type;
   ^~~~~~~~~~~~~~
In file included from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h747xx.h:242:0,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h7xx.h:136,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/cmsis.h:22,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/PinNamesTypes.h:33,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\variants\GIGA/pinmode_arduino.h:24,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/Arduino.h:26,
                 from C:\Users\softw\AppData\Local\Temp\arduino\sketches\9ACDF38E1AB19999B2F7A36DEA1095D8\sketch\Fire2012.ino.cpp:1:
C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/core_cm7.h:1654:3: note: previous declaration as 'typedef struct CoreDebug_Type CoreDebug_Type'
 } CoreDebug_Type;
   ^~~~~~~~~~~~~~
In file included from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h7xx.h:136:0,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/cmsis.h:22,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/PinNamesTypes.h:33,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\variants\GIGA/pinmode_arduino.h:24,
                 from C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/Arduino.h:26,
                 from C:\Users\softw\AppData\Local\Temp\arduino\sketches\9ACDF38E1AB19999B2F7A36DEA1095D8\sketch\Fire2012.ino.cpp:1:
C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h747xx.h:2638:44: error: expected ')' before '*' token
 #define COMP1               ((COMP_TypeDef *) COMP1_BASE)
                                            ^
C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/cm3_regs.h:43:19: note: in expansion of macro 'COMP1'
     __IO uint32_t COMP1;                   /*!< Offset: 0x030 (R/W)  Comparator Register 1                     */
                   ^
C:\Users\softw\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.2\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/CMSIS/stm32h747xx.h:2638:44: error: expected ')' before '*' token
 #define COMP1               ((COMP_TypeDef *) COMP1_BASE)
                                            ^
C:\Users\softw\OneDrive\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/cm3_regs.h:43:19: note: in expansion of macro 'COMP1'
     __IO uint32_t COMP1;                   /*!< Offset: 0x030 (R/W)  Comparator Register 1                     */
                   ^
[... cut ...]
-----------------------------------------------------------------------------------------------------------------------------------

We tried everything but no result. This problem is so important for us. We are developing a device and we try to use WS2811 for user interactions. (errors, warnings etc.).

We think, FastLED is best solution for addressable led programming, but it should be update for as Arduino Giga R1 WiFi devices in mean time...

If you have another solutions, we and all users wait it.

Thanks a lot again 💯

zackees, 2023-03-25:
If you are trying to make a product then go with the ESP32, it's $4USD and has wifi. It's arduino compatible and FastLED works on it.

mathias-maierhofer, 2023-07-10:
Hi @donemuhendislik 

just received my giga wifi and experiencing the same problem. Did you get it to work in the meantime?

artemt1987, 2024-05-10:
Same with arduino giga and FastLED sadly not working....

schnoberts1, 2024-05-19:
Hi,

Sorry to re-heat an old thread, but I have some questions relating to the current STM32 porting effort for FastLED. Caveats - I am an experienced dev but an inexperienced embedded dev. 

I started having a poke around with the thought that porting might be quick and hit some initial issues that made me wonder how mature the existing STM32F support is (i.e. am I on a solid foundation). Example, I see this code in `led_sysdefs_arm_stm32.h`: 

```
#if defined(STM32F2XX)
#define F_CPU 120000000
#elif defined(STM32F1)
// F_CPU is already defined on stm32duino, but it's not constant.
#undef F_CPU
#define F_CPU 72000000
#else
#define F_CPU 72000000
#endif
```

However, a quick check of the STM32 docs for the STM32F1xxx line shows they do not all share the same clock frequency so they should presumably have different  F_CPU settings. That made me wonder how well tested this area was so I checked the commits related to STM32 support and it looks like it hasn't been rev'd for a couple of years. Would I be right in thinking the STM32 support is orphaned at the moment? 

I'm not looking for super smooth colour changes so I was wondering if I'd get away with FASTLED_FORCE_SOFTWARE_PINS. Is this path worth exploring do you think?

zackees, 2024-05-19:
I'd give it a try! Software bit banging should work. Let us know!

closed 2024-09-18 by zackees