FastLED/FastLED #1490: Support fo OSTW2020C1E OPTOSUPPLY

2kohm, 2023-03-28:
Hello. anybody know if this LED is supported ?
[846168](https://www.tme.eu/Document/bddfe2b10b24331bb9ab3894a93aba32/OSTW2020C1E.pdf)
is there any 2020 ( or even smaller ) led supported ?

marmilicious, 2023-03-29:
The size of the LED usually doesn't matter, it's the type of controller chip inside. (For example, here's some very tiny WS2812 pixels: [https://www.instagram.com/p/CqO2mXjsyEz/](https://www.instagram.com/p/CqO2mXjsyEz/) )

If you already have some of these pixels try running them as WS2812 pixels. If it doesn't work it won't hurt anything, they just won't light up.

Koudy86, 2023-10-19:
I tried it today, OSTW2020C1E is 2812B compatible. I used this and it works:
`FastLED.addLeds<WS2812B, LED_DATA_PIN, GRB>(leds, NUM_LEDS);  // GRB ordering is typical`

2kohm, 2023-11-24:
Thanks for the feedback, WS2812B is now also available in 1x1mm package! :)

[image]

closed 2026-09-02 by zackees