FastLED/FastLED #1492: ESP8266 Board Release 3.1.0+ produces a lot of warnings during compile.

AlbertGBarber, 2023-04-01:
Compiling for ESP8266 using the latest board releases (3.1.0 and above) produces a lot of warnings. These don't stop the compile from finishing, but they do clutter up the output. Have not tested that the code still runs. 

I've pasted a sample of the errors below. There are more, but they're all about the same, just in different parts of the code. These errors were collected by compiling the "blink" example code for a Wemos D1 mini (R2 and mini in the board selection)  with no modification using the latest FastLED and ESP8266 releases.

E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\fastspi_bitbang.h:341:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  341 |    register clock_t clockhi = FastPin<CLOCK_PIN>::hival();
      |                     ^~~~~~~
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\fastspi_bitbang.h:342:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  342 |    register clock_t clocklo = FastPin<CLOCK_PIN>::loval();
      |                     ^~~~~~~
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\fastspi_bitbang.h:357:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  357 |    register data_t datahi_clockhi = FastPin<DATA_PIN>::hival() | FastPin<CLOCK_PIN>::mask();
      |                    ^~~~~~~~~~~~~~
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\fastspi_bitbang.h:358:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  358 |    register data_t datalo_clockhi = FastPin<DATA_PIN>::loval() | FastPin<CLOCK_PIN>::mask();
      |                    ^~~~~~~~~~~~~~
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\fastspi_bitbang.h:359:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  359 |    register data_t datahi_clocklo = FastPin<DATA_PIN>::hival() & ~FastPin<CLOCK_PIN>::mask();
      |                    ^~~~~~~~~~~~~~
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\fastspi_bitbang.h:360:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  360 |    register data_t datalo_clocklo = FastPin<DATA_PIN>::loval() & ~FastPin<CLOCK_PIN>::mask();
      |                    ^~~~~~~~~~~~~~
In file included from E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\FastLED.h:68,
                 from E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\wiring.cpp:2:
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\chipsets.h: At global scope:
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\chipsets.h:88:80: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   88 |   __attribute__((always_inline)) inline static uint8_t adjust(register uint8_t data) { return ((data>>1) | 0x80) + ((data && (data<254)) & 0x01); }
      |                                                                                ^~~~
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\chipsets.h: In member function 'virtual void LPD6803Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::showPixels(PixelController<RGB_ORDER>&)':
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\chipsets.h:175:31: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  175 |             register uint16_t command;
      |                               ^~~~~~~
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\chipsets.h: In member function 'void P9813Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::writeLed(uint8_t, uint8_t, uint8_t)':
E:\Albert Barber\OneDrive\Documents\Arduino\libraries\FastLED\src\chipsets.h:349:20: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
  349 |   register uint8_t top = 0xC0 | ((~b & 0xC0) >> 2) | ((~g & 0xC0) >> 4) | ((~r & 0xC0) >> 6);

zackees, 2023-04-02:
https://github.com/FastLED/FastLED/pull/1461

m4rc77, 2023-04-07:
If you are using [PlatformIO](https://platformio.org/) you can add in your `platformio.ini` file the following lines, in order to "silence" the warnings from FastLED until a patch is available:

```
; fix in order to get rid of 'warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]' from FastLED Library <= 3.5.0
; fix in order to get rid of 'warning: this 'while' clause does not guard... [-Wmisleading-indentation]' from FastLED Library <= 3.5.0
build_flags =
    -Wno-register
    -Wno-misleading-indentation
```
**But be warned, this will also hide similar issues in your own code!**

zackees, 2023-10-03:
I think this issue is resolved and should be closed.