FastLED/FastLED #1494: Any way to flip CRGB buffer in CLEDController() object?

vortigont, 2023-04-15:
Suppose I want to switch buffers in a runtime to avoid `memcpy`-likish operations for, i.e., double buffering.
Looks like `CLEDController()` class has no way to reassign `CRGB *m_Data` member. Nor it has any way to manipulate linked list chain of `CLEDController()`, so once added any member it will stay there until restart.
Is there any hidden stones that prevents to implement a member method in parent `CLEDController()`that would allow to reassign `CRGB *m_Data` pointer at runtime? Or am I missing something?
Tnx!

vortigont, 2023-04-19:
OK, with a bit more learning found a suitable way to switch a pointer for a specific chain of LED's like this
```
CLEDController *cled;
setup(){
   cled = &FastLED.addLeds<WS2812B, IO_PIN, COLOR_ORDER>(crgb_buffer_ptr, crgb_buffer_size);
}

// then later somewhere in the code
cled->setLeds(another_crgb_buffer_ptr, buffer_size);
``` 

but still looking for a way to unlink and detach `*cled` object completely from a chain.

zackees, 2023-04-19:
Your MCU will likely perform that memcpy faster than it takes to write out ONE pixel.

Be mindful of the premature optimization trap.

vortigont, 2023-04-20:
@zackees true indeed, but it's more of a code style concerns than speed of memcpy. Good old C-style arrays works pretty fine, but wrapping it into standard containers makes life much easier, i.e. using move semantics, make it run-time resizable/configurable, drop per-byte copy `for` loops for swapping CRGB buffers, etc...
FastLED is already a set of classes anyway, right. Just curious if I'm missing something here.

vortigont, 2023-04-20:
This PR #1297 adds one missing part - obtaining a linked list ref that could be further processed, i.e. need to add a method for removing/poping nodes.

vortigont, 2023-08-31:
@robertlipe I'm trying to work my way with a code in a more flexible way. I mean, LED related libs and examples around are mostly pure C-like style, with bare pointers and tons of `#define WIDTH`, `#define HEIGHT`, `#define PIN` with a `config.h` files like another book to learn. Not that it's wrong or a bad style but it is frustrating.
For me it's easy to code in a way like defining pins and geometry/layout in a run-time, work with frame buffer structures that could be easily swapped or moved between each other. I.e. to implement on-demand overlay mixer or using a single shared buffer for different consumers. I came up with a full run-time configurable esp32 driven LED [framebuffer](https://github.com/vortigont/FireLamp_JeeUI/blob/4255c1270847a3da4e161a8adb47bd05cbf3e9dd/src/ledfb.hpp), including gpio assignment. Without a feature to manipulate a chain of `CLEDController()`s I still can't detach and release pins and `CLEDController()`, so have to use a hackish stub with a blind pixel. It works though, no complains. FastLED was born in 8-bit MCU bitbang era but modern chips are so feature rich!

As for 

> is the possibility of DMA controllers traversing objects in a CRGB while we abuse them

guess it depends on MCU and it's features. For sure you need to use some signaling or locking to mess with buffer pointers or data updating, but it's not that hard with RTOS. Mangling with memory regions that is used by DMA engines is usually OK too. By all means it's the main idea of DMA - having a shared mem for data exchange between hw and programs.

zackees, 2024-10-12:
Low level code likes static memory and passing pointers around from static data. We aren't going to change that in the core driver.

In app land where there is compositing, historical buffers and mixing, yes you want all kinds of dynamic content and smart pointers so that you don't have to deal with ownership rules and can trust that grabbing a smart pointer will keep the object alive. And also, when you need to clear an object graph, the tremendous amount of data that it uses can all be returned for new Fx to be instantiated.

As for swapping buffers around for the controller classes...

This causes more problems than fixes them. it's a better strategy to implement FastLED.remove(...) and just tear down the driver in question. The drivers are very tiny - just a hand full of bytes. The big allocation of these drivers happens to be in executable code, which can't be returned to the heap ever. The LED data is also pretty big and will dwarf the driver objects by an order of magnitude, not counting the executable code.

closed 2024-10-12 by zackees