FastLED/FastLED #1496: Beginner Question: causes of unwanted blinking ?

seb4o, 2023-04-19:
Hi everyone,

I just start playing with wled and fastled, but I have some unexpected issue during my testing.

I made a simple code to have a blue point followed by a yellow fading line.

But at some iterations I have some blinking leds (that are rather white, not blue or yellow), and I do not know from where it can come.

Can you please tell me if it can come from bad order from the controller (ESC32), bad communication, or if it is a problem with the fading function that computes and send strange values ?
(also at the very beginning the colours are nearly random, it is only after around 20 steps that I have the first led blue and the following ones yellow).
I just see no reason for that...

Thanks !!

Here is the code:

#include <FastLED.h>

#define NUM_LEDS 119 
#define DATA_PIN 15

CRGB leds[NUM_LEDS];

void setup() { 
	Serial.begin(57600);
	Serial.println("resetting");
	FastLED.addLeds<WS2812,DATA_PIN,RGB>(leds,NUM_LEDS);
	FastLED.setBrightness(200);
}

void fadeall() { for(int i = 1; i < NUM_LEDS; i++) { leds[i].fadeToBlackBy(2); } }

void loop() {
  // put your main code here, to run repeatedly:

	for(int i = 1; i < NUM_LEDS; i++) {
    leds[i+1] = CRGB::Blue;
    leds[i] = CRGB::Yellow;
    FastLED.show(); 
    fadeall();  
    delay(200);
  }

}

marmilicious, 2023-04-19:
I suspect there's something bad happening in memory since you are trying to write rgb pixel data past the end of your CRGB array. When you get to the end of your for loop leds[i+1] will equal NUM_LEDS which is a pixel that doesn't exist.

We use github "issues" here mostly just for bugs or improvements in the library itself. For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

seb4o, 2023-04-19:
Thanks for the answer.
Strangely the problem appears quite before this end of loop, but I will check !
Sorry, I did not realized that this place was not for "stupid-user-questions".
I do not know if I can remove this thread for not polluting the place...

marmilicious, 2023-04-19:
Memory errors can be difficult to recognize and cause all sorts of different random errors. After you fix that of you still have issues please post on the reddit group.

You can close this ticket and it will go away. :)

closed 2023-04-20 by seb4o