FastLED/FastLED #1497: Rpi Pico w/ FastLED periodically locks up

markthomascampbell, 2023-04-20:
I have a script that I originally created for Arduino & FastLED, and wanted to get more performance, so I switched over to a Rpi Pico.  I have it working with Arduino IDE 2.0.4, and added in the source code from the latest revision of FastLED to get it working with the Pico.  The code builds, and uploads successfully.  It also runs as expected for a while.  As part of my script, I've given it a serial interface, where I can issue commands.  Commands to change colors, speed up, slow down the light sequence, etc.  However, when I tell it to stop (which only turns a variable from true to false), and then tell it to turn back on (which turns that variable back from false to true), the Pico locks up, and the onboard LED flashes a repeating sequence of 4 fast and 4 slow green flashes.  I'm not sure what is the issue, so I am seeing if anyone here can tell what might be going on.  My first inclination is that perhaps RP2040 support is still tenuous, and not 100% compatible.  I'm also wondering if there's maybe some weirdness with the EVERY_N_MILLISECONDS statements that basically keep the script running.

I've attached my arduino IDE code in a txt file in case it helps.

[scanner_pico.txt](https://github.com/FastLED/FastLED/files/11280080/scanner_pico.txt)

marmilicious, 2023-04-20:
The repeating 4 slow 4 fast blink pattern you're seeing might be this:

https://github.com/arduino/ArduinoCore-mbed/issues/316

marmilicious, 2023-04-20:
You might try using this for the Pico in the Arduino IDE.

https://github.com/earlephilhower/arduino-pico

markthomascampbell, 2023-04-20:
Thank you Marc!  I had initially thought I'd done that one already, and indeed I had already put it in the Additional Boards Manager URLs section in Preferences, but when I went to select the board to install, I apparently selected the mbed one instead.  I just installed the correct one from Earle, and re-uploaded the sketch, and those functions are working now!!  Thanks a lot for the suggestion!

closed 2023-04-20 by markthomascampbell