FastLED/FastLED #1498: ESP32-C3 Offset Issue

Xylopyrographer, 2023-04-22:
Hi all. Wrote a sketch to test an 8 × 8 WS2812B matrix panel.

When loaded on to an ESP32-C3, it doesn't work:
- leds[1], leds[2] flash brightly on startup
- pixel walk test; leds[0] is never written
- row test; leds[] data is offset by + 1, i.e. leds[0] is at leds[1], etc.
- column test; leds[0] never lights
- fill test; leds[0] never lights
- panel test; leds[0] never lights

C3 (failed) tests on:
seed studio XIAO ESP32C3
- Chip is ESP32-C3 (revision v0.3)

ESP32-C3 DEV
- Chip is ESP32-C3 (revision v0.3)

When run on a ESP32  or an ESP32-S3 all is tickity boo. Successful tests on:
M5Stack ATOM Matrix
- Chip is ESP32-PICO-D4 (revision v1.0)

ESP32 DEV Kit
- Chip is ESP32-D0WD-V3 (revision v3.0)

seed studio XIAO ESP32S3
- Chip is ESP32-S3 (revision v0.1)

Tested using Arduino IDEv 1.8.9 & v 2.0.4
Using arduino-esp32 core v2.0.7 and 2.0.8
FastLED library v3.50
Tried with two panels. Type is [here](https://www.aliexpress.com/item/32644496853.html?spm=a2g0o.detail.1000014.2.650415edxDjLiZ&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40050.281175.0&scm_id=1007.40050.281175.0&scm-url=1007.40050.281175.0&pvid=4a5e776d-08a2-466c-8b58-195e1d9e5ac1&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40050.281175.0,pvid:4a5e776d-08a2-466c-8b58-195e1d9e5ac1,tpp_buckets:668%232846%238116%232002&pdp_ext_f=%7B%22sku_id%22%3A%2212000018168044431%22%2C%22sceneId%22%3A%2230050%22%7D&pdp_npi=2%40dis%21CAD%210.64%210.61%21%21%21%21%21%402103221f16709035562702013e70fa%2112000018168044431%21rec).

Let me know if any other info would be of use.

Sketch:
```
//  8 by 8 Matrix panel test

#include <FastLED.h>

// Define LED matrix parameters

// pick a GPIO pin...
// #define LED_PIN     26          // for std ESP32
// #define LED_PIN     22          // for ATOM MATRIX
// #define LED_PIN     5           // for ESP32-C3
// #define LED_PIN     3           // alternate test pin for ESP32-C3
// #define LED_PIN     5           // for XIAO ESP32-S3

#define LED_NUM     64
#define LED_TIPE    WS2812B
#define L_RED       0x00ff00
#define L_GREEN     0xff0000
#define L_BLUE      0x0000ff
#define L_WHITE     0xffffff
#define L_BLK       0x000000
#define A_DELAY     200UL       // # of ms delay between animations

#define STRINGIFY(x) #x
#define TOSTRING(x) STRINGIFY(x)

CRGB leds[ LED_NUM ];
CRGB tColours[] = { L_RED, L_GREEN, L_BLUE, L_WHITE };
uint8_t culrCount = sizeof( tColours ) / sizeof ( CRGB );

void setup() {

    // Initialize LED matrix
    FastLED.addLeds<LED_TIPE, LED_PIN>( leds, LED_NUM );
    FastLED.clear( true );
    FastLED.setBrightness( 10 );

    Serial.begin(115200);
    delay( 250 );
    Serial.println("\r\n\r\n");
    Serial.println( "=======================");
    Serial.println( "8 x 8 Matrix Panel Test");
    Serial.print( "   LED Type = " ); Serial.println( TOSTRING( LED_TIPE ) );
    Serial.print( "   Data Pin = " ); Serial.println( LED_PIN );
    Serial.print( "   Colour Count = " ); Serial.println( culrCount );
    Serial.println( "=======================");

}

void loop() {

    // pixel walk - sequentially light up each pixel
    for ( uint8_t culr = 0; culr < culrCount; culr++ ) {
        for ( uint8_t pxl = 0; pxl < LED_NUM; pxl++ ) {
            leds[ pxl ] = tColours[ culr ];
            FastLED.show();
            leds[ pxl ] = L_BLK;
            delay( A_DELAY );      // pause before moving to the next pixel
        }
    }
    FastLED.clear( true );

    delay( 500 );

    //  row test - sequentially light up each row
    for ( uint8_t culr = 0; culr < culrCount; culr++ ) {
        for ( uint8_t i = 0; i < 8; i++ ) {
            for ( uint8_t j = 0; j < 8; j++ ) {
                leds[ i * 8 + j ] = tColours[ culr ];
            }
            FastLED.show();
            FastLED.clear();
            delay( A_DELAY );       // pause before moving to the next row
        }
    }
    FastLED.clear( true );

    delay( 500 );

    //  column test - sequentially light up each column
    for ( uint8_t culr = 0; culr < culrCount; culr++ ) {
        for ( uint8_t i = 0; i < 8; i++ ) {
            for ( uint8_t j = 0; j < 8; j++ ) {
                leds[ i + j * 8 ] = tColours[ culr ];
            }
            FastLED.show();
            FastLED.clear();
            delay( A_DELAY );       // pause before moving to the next column
        }
    }
    FastLED.clear( true );

    delay( 500 );

    // fill test - light up all pixels
    for ( uint8_t culr = 0; culr < culrCount; culr++ ) {
        for ( uint8_t i = 0; i < LED_NUM; i++ ) {
            leds[ i ] = tColours[ culr ];
        }
        FastLED.show();
        FastLED.clear();
        delay( A_DELAY * 4 );
    }
    FastLED.clear( true );

    delay( 3000 );                  // pause before repeating the tests
}

//  EOF
```

Xylopyrographer, 2023-04-28:
[matrix_test.zip](https://github.com/FastLED/FastLED/files/11357894/matrix_test.zip)

taseraul, 2023-05-27:
I have also encountered this issue, and from my findings i concluded that some of the c3 pins are default high after the buffer is sent and the RMT relinquishes control so the pixles actually detect a ghost pixel (sometimes white, sometimes closed)

colincross, 2023-06-19:
I have a similar problem driving the neopixel on an Adafruit QT Py ESP32-C3, which has a 10k pullup on the data line.  Scoping the data line shows that it rises slowly at the end of the data sequence, and is never held low for 50uS to trigger the reset state.  Commenting out the `gpio_matrix_out` line that disconnects the output pin from the rmt controller keeps the data line low between data sequences, triggering the reset correctly.

colincross, 2023-06-24:
Removing `gpio_matrix_out` completely causes other problems, if multiple output pins are in use it leaves them all connected to the RMT controller, causing everything that is sent to be sent to all LED strings.

I tracked the problem down to a documentation issue (https://github.com/espressif/esp-idf/issues/11737), `gpio_matrix_out(pController->mPin, 0x100, 0, 0);` is only correct for some ESP32 platforms, on ESP32-C3 it needs to be `gpio_matrix_out(pController->mPin, 0x80, 0, 0);` to reconnect the pin to the GPIO controller.  Passing `0x100` selects peripheral 0 (the `SPIQ_out` line) with its output inverted.  Since I'm not using the SPI controller it's output enable is not set, so the pin ends up floating.

There doesn't seem to be an easy way to find the right value to pass to gpio_matrix_out for each platform.  One option would be to `#ifdef` the platforms that use 0x80 instead of 0x100.  The other option is to assume that the value will always be just the top bit of the field set, and compute it with `(GPIO_FUNC0_OUT_SEL_M+1)>>1`.

colincross, 2023-06-25:
https://github.com/espressif/esp-idf/issues/11737 pointed out `SIG_GPIO_OUT_IDX`, which has the correct value on all platforms, so I've updated the pull request.

ammoniak, 2023-07-08:
thx for the PR! (The fix works... hope this gets merged soon)

closed 2023-07-08 by samguyer

samm928, 2023-11-30:
What exactly was the fix in this case ? .. Flashing a new Bootloader ? .. I'm seeing the same issues on third party ESP32 -C3 and -S3 SuperMini and Zero from Tenstar Robot and Waveshare

fusionliquid, 2024-09-01:
I also have similar problem with esp32 s3.  when adding a motionsensor with FastLED, the system crashes reportedly saying RMT driver problem.    But using the same code with cheaper arduino nano, it has no problem running...

PR #1515 (2023-07-08): Use correct cancel value for gpio_matrix_out on esp32c3
The documentation for gpio_matrix_out on esp32c3 is incorrect (https://github.com/espressif/esp-idf/issues/11737), and 0x80 needs to be used instead of 0x100.  Use SIG_GPIO_OUT_IDX, which has the correct value on all platforms.

Fixes #1498
commit 237db1bb (2023-07-08): Use correct cancel value for gpio_matrix_out on esp32c3

The documentation for gpio_matrix_out on esp32c3 is incorrect
(https://github.com/espressif/esp-idf/issues/11737), and 0x80
needs to be used instead of 0x100.  Use SIG_GPIO_OUT_IDX, which
has the correct value on all platforms.

Fixes #1498