FastLED/FastLED #1500: Should FastLED be forked so that development can continue?

zackees, 2023-05-01:
We are having C++17 issues and new chipsets are coming like RGBW.

It's been over a year since FastLED has been updated and the issues to this library are going to get worse. PR's are being submitted to fix these issues but they aren't being merged. What should be done?

Solutions to this include:
  * Adding more owners that can review and greenlight changes
  * Forking FastLED and continue on with an active fork in which changes are periodically merged back into the official fork.
  
What are your thoughts?

kriegsman, 2023-05-01:
Hi @zackees - Thanks for opening up this for discussion.  @samguyer and I have had a bunch of discussions, and we'd definitely like to open up FastLED to more owners and committers, and really include more of the great developers we have making contributions now -- without a narrow bottleneck ( = just me & Sam) slowing things down.

So in short, I think I'd like to open it up bigger, as you suggested, but I think that we can do that 'in place', right here in the FastLED project.  I'm certainly in favor of that, and I'm pretty sure that @samguyer is, too.

zackees, 2023-05-01:
I'd love to help out. As mentioned in the reddit post I have over a decade in hardcore C++ working in big tech and I'd like to add more automated testing to reduce contributor friction.

ccoenen, 2023-05-02:
If this can be maintained under the existing organisation, I think this will usually be preferable. I don't have anything against a fork, but as long as the current owners/maintainers/members are fine with adding more people, the existing organisation has many benefits.

I think forking is only the "better" solution in one of these cases (which I do not think apply to FastLED):
- when the original author(s) are no longer available (i.e. single dev who is not returning for whatever reason)
- when the upstream and the fork can't agree on a direction (i.e. node.js / io.js who had different opinions about governance of the project, among other things)
- when existing members/maintainers never had important permissions for website and/or organisations (I've been on projects, where the original author vanished and nobody could add people to the Github-Org! Or where some fan registered a domain, and all of this went fine for years, and suddenly that domain got taken over by someone entirely unrelated... etc.)
- when the fork is done without notifying the source project (which may or may not be bad form, but is fair game under most licenses)
- when the existing maintainers can't vet new maintainers (for whatever reason... resources, time constraints, language barriers, I don't judge any of these!)

From what I see in this short thread, I don't think any of these apply. The **benefits** of staying with the existing FastLED org include:
- We don't fracture the community ("Oh, you're using NewLED? No, I was referring to BlorbLED, that's a fork of ...")
- Whoever just uses FastLED because of some YouTube video or instructable will just continue to follow that tutorial, and it will likely continue to work years from now (don't underestimate how freaking many people just use whatever comes up in some tutorial somewhere!)

In conclusion:
- if there are people willing and able to step up ✅
- that might even have a good reputation on the project already (or who for whatever else reason are trusted by the existing maintainer team) ✅
- and everyone's goals align reasonably well ✅
- and the current members have the ability to add project members ✅

then staying with the existing organisation benefits everyone the most.

polykoma, 2023-05-03:
Zackee is very motivated so he definitely has my vote in favour.
Also fully agree to anything worded above. <3

This is gonna be great!

kriegsman, 2023-05-29:
We just released FastLED 3.6.0.   Now this week we'll start adding more folks onto the repo as contributors -- starting with @zackees !  
I'm going to leave this issue open as a To-Do until we add a couple new team members.

Thank you, everyone, for all the support and code and care and energy!

zackees, 2023-05-29:
Thank you. It's quite an honor to be a contributor!

If you are considering others I recommend @ssilverman as a contributor.

ssilverman, 2023-08-07:
I’m still up for assisting with FastLED maintenance. I keep offering but there’s always crickets…
@samguyer @kriegsman @zackees

zackees, 2023-08-10:
I propose a zoom call once every two months. The first one would kick off expectations. I would like to write an expectations document then show it to the owners/stackholders for buy in.

ssilverman, 2023-08-11:
@samguyer was that to my comment or Zack’s?

zackees, 2023-08-11:
So how do we communicate? We are on a public forum.

ssilverman, 2023-08-11:
What about a Discord server or something?

polykoma, 2023-08-11:
If you want a Discord server going let me know. I make discords for a living and can get a dope server spun up in a fingersnip with ma template

samguyer, 2023-08-11:
@ssilverman Yes to both! 

Mark and I started a FastLED Slack channel last year, but we haven't used it a lot.

ssilverman, 2023-08-11:
@polykoma I’d say just spin it up, start with that, and it can always be changed later. I think everyone on this thread cares deeply about the project and would all love to see some movement and less stress for the two current maintainers.

samguyer, 2023-08-11:
I'm so glad there are people like you all who care about FastLED!

ssilverman, 2023-08-11:
@samguyer are you willing to share the Slack link?

ssilverman, 2023-08-11:
I started a priority list here; please request edit access if you want it:
https://docs.google.com/document/d/11uWpUgeonsuwdOZpDnWVp5qQcrxN4e8pyARHX-Hg_IA/edit

polykoma, 2023-08-11:
https://discord.gg/nqBMbAZ75M

I gotta tailor it towards this project coming days. But feel free to join already. Let me know on here whats ur nickname and what role you prefer to get (like moderator / dev tag etc)

ssilverman, 2023-08-11:
@polykoma how would you feel about capitalizing the ‘F’ in the name?

polykoma, 2023-08-11:
Hahaha, yeah sure. Done! There is a dot in front of the name too - so its easier for me to add some bots later. Will get rid of the dot after setup is done. Time to sleep now 6AM

ssilverman, 2023-08-11:
Thanks, @polykoma!

zackees, 2023-08-11:
Can we turn this discord policy off? It's currently blocking me:

[image]

polykoma, 2023-08-11:
Yeah I can take away rules and dms step. You can just click continue. Its just optional

polykoma, 2023-08-11:
Done

ccoenen, 2023-08-11:
I would like to point out that Discord or Slack are not a good place for an open source project. I could go into details, but other people have already written at length about the topic:
https://blog.mattcen.com/2023/07/04/stop-using-discord-for-your-open-source-communities/

ssilverman, 2023-08-11:
I think this is meant for some planning and not necessarily for the whole community. At least that’s my take on it.

samguyer, 2023-08-11:
@ssilverman Sorry for dropping off last night. Sounds like Discord is the choice. The Slack channel that Mark and I used was strictly for planning among the project "owners", not for public discussion.

zackees, 2023-08-11:
What do the owners think about having an experimental folder? This would give us a place to play with new code that's not part of the officially supported codebase. For example I have a PR for for a conversion of RGB -> RGBW. If people try it out and like it then it can be considered for an integration.

polykoma, 2023-08-12:
> I would like to point out that Discord or Slack are not a good place for an open source project. I could go into details, but other people have already written at length about the topic:
>
>     * [...]

While I appreciate the links, very good stuff, I dont see much alternatives beside Matrix (overwhelming) and IRC (too outdated in functionality).

I think having Reddit for actual longer term threads is great + using github for the actual development. 
Discord is a nice addition for some chatting around reddit threads or github development. I would advise to not "host" anything important on Discord beside active discussions. Since yes its a walled garden and yes its hard to extract something out of it.

So the really important stuff should stay on github.

I can connect reddit to the discord and most likely also the github. So discord becomes like a hub.

Will have time to work on the discord coming days. :)

ssilverman, 2023-09-07:
@samguyer @kriegsman I'd really like to help. How can I do this? I've been trying to volunteer for a long time now.

polykoma, 2023-09-10:
Are you on the new discord I made? Trying to gather all the active folks on there

zackees, 2023-09-12:
Oh hi @robertlipe , good to see you here.

rah2501, 2023-12-07:
> I propose a zoom call once every two months.

Are these happening?

zackees, 2023-12-12:
> Hey, @zackees. I wondered when/if you were going to recognize me. It's been a long time in programmer-years. (which are kind of like dog years.)
> 
> I'm willing to help out as I can. I made a rather serious pass over the bug database recently, trying to get some duplicates merged and dead ones closed, and have some harsh ideas (yes, based on the place of our our shared experience on how to help advance that. I'd be happy to talk about that - or anything else I can do to help - in a smaller group - that isn't on Discord. :-)

I'd personally like to see all the AVR stuff burned to the ground. I know that's where this library got it's start but AVR is being phased out and other projects like WLED are supporting multiple pins for writing out lighting without much fuss. That seems to be the future of lighting.

zackees, 2023-12-12:
> > I propose a zoom call once every two months.
> 
> Are these happening?

Not really - we have a slack channel if you are putting in a desired feature.

zackees, 2024-09-20:
> > Hey, @zackees. I wondered when/if you were going to recognize me. It's been a long time in programmer-years. (which are kind of like dog years.)
> 
> > 
> 
> > I'm willing to help out as I can. I made a rather serious pass over the bug database recently, trying to get some duplicates merged and dead ones closed, and have some harsh ideas (yes, based on the place of our our shared experience on how to help advance that. I'd be happy to talk about that - or anything else I can do to help - in a smaller group - that isn't on Discord. :-)
> 
> 
> 
> I'd personally like to see all the AVR stuff deprecated. I know that's where this library got it's start but AVR is being phased out and we need better support for the ESP32 stuff. Also parallel output, such as WLED is doing without any fuss.

Holy crao was I wrong about the AVR stuff needing to be deprecated. I now feel so stupid for saying this. People really like these extremely cheap micros and are pushing it to the limit.

zackees, 2024-09-20:
Instead of a fork, I got inducted as a contributor! 🎉

closed 2024-09-20 by zackees