FastLED/FastLED #1507: Doesn't work with Uno R4 Minima

ripred, 2023-05-30:
The library fails to compile for the Uno R4 Minima board with the Renesas processor.

zackees, 2023-05-30:
Please post the output of the compiler error

ripred, 2023-05-30:
Using an Uno R4 Minima selected as the board.
Compiling the example sketch DemoReel100.ino I get the following output using the 1.8.19 IDE:

In file included from /Users/trent/Documents/Arduino/libraries/FastLED/src/FastLED.h:51:0,
                 from /Users/trent/Documents/Arduino/libraries/FastLED/examples/DemoReel100/DemoReel100.ino:5:
/Users/trent/Documents/Arduino/libraries/FastLED/src/led_sysdefs.h:61:2: error: #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
 #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
  ^~~~~
exit status 1
Error compiling for board Arduino Uno R4 Minima.

Let me know if any other info you might need.

zackees, 2023-05-31:
Unfortunately, the new Uno uses the ` Renesas RA4M1 32-bit Cortex®-M4 processor` which appears not to be supported yet by FastLED.

antoineheseque, 2023-07-05:
Finger crossed for a support soon-ish 🔥

antoineheseque, 2023-08-30:
> These things always come down to finding someone with the right combination of skills, hardware, time, and motivation. The buik of the FastLED code is super portable, but there's the bottom layer that just has to be really cozy with the hardware in order to shove the bits out with the needed timing.
> 
> If you want it badly enough, roll up your sleeves and do it (there are plenty of examples and you presumably have the hardware and the motivation parts) or find an intern or contractor to help get it done and integrated back into the project.
> 
> Uno R4 looks to be super new, but the Arduino logo on it means it'll tend to attract capable developers quickly. That's one of the perks of the ecosystem.
> 
> I'm just saying that there are options beyond passively hoping that someone else will come along and do the work. (for free 😉)

Totally agree, but support seems already to be developed and just waiting for a review (in pull requests!)

antoineheseque, 2023-08-30:
#1523

zackees, 2024-08-30:
Confirmed that it works in latest so closing this bug.

closed 2024-08-30 by zackees