FastLED/FastLED #1510: FastLED.show() on ESP32-S2 while using RMT causes crash

ryryog25, 2023-06-07:
I'm using an ESP32-S2 and I'm trying to set up an RMT channel to output for an IR transmitter and also utilize the FastLED library.

Because FastLED uses the RMT, I've been running into a few issues. Right now I have `FASTLED_RMT_MAX_CHANNELS 1` defined 
before I include `<FastLED.h>` and then I set up the IR like so:
```
    rmt_config_t txConfig = RMT_DEFAULT_CONFIG_TX(GPIO_NUM_18, RMT_CHANNEL_3);
    txConfig.clk_div = 80; // make each tick a uS
    txConfig.mem_block_num = 1;
    txConfig.flags = 0;
    txConfig.tx_config.carrier_freq_hz = 36000;
    txConfig.tx_config.carrier_level = RMT_CARRIER_LEVEL_HIGH;
    txConfig.tx_config.carrier_duty_percent = 50;
    txConfig.tx_config.carrier_en = true;
    txConfig.tx_config.loop_en = false;
    txConfig.tx_config.idle_output_en = false;

    rmt_config(&txConfig);
    rmt_driver_install(RMT_CHANNEL_3, /*buffersize*/ 0, /*flags*/ 0); 
```

The problem comes in only when FastLED.show() is called. I have it set up so that in a loop a function for updating the LEDs, and then a function for (providing conditions are met which revolve around button presses) sending IR data. When I press the button, thus invoking the function to send IR data, shown below, the ESP crashes with a `Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).`

IR Function:
```
void send() {
    static const rmt_item32_t message[] = {
      // Start bits
      RC5_LOGIC_1,
      RC5_LOGIC_1,

      // toggle bit
      RC5_LOGIC_0,

      // address bits
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      
      // command bits
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      RC5_LOGIC_0,
      RC5_LOGIC_0,

      RMT_END_MARKER
    };

    rmt_write_items(RMT_CHANNEL_3, message, /*item_num*/ 14, /*wait_tx_done*/ false);
  }
```

With FastLED set up and LEDs added, but not calling FastLED.show() I don't get any panics. I'm thinking this is because the RMT is doing *stuff* when FastLED tries to access it, even though it shouldn't be accessing it? Honestly this is above my head and I'm just speculating based on absolutely nothing.

As an aside, I've also tried FASTLED_RMT_BUILTIN_DRIVER since there's known to be conflicts with using RMT in other places, but that just results in the ESP hanging on start.

I'm sure there's useful details I've glossed over or forgotten to add, let me know about that, heh.

ryryog25, 2023-06-07:
Actually... hmm... that's interesting. I take back my comment about `FASTLED_RMT_BUILTIN_DRIVER 1`
It just causes nothing to display on the LEDs, not hanging (Maybe something else was causing issues when I tried this earlier)

ryryog25, 2023-06-08:
Further testing shows that I just have to call FastLED.show() once. If it's never called the issue doesn't occur, but if it's called even once the IR function will always cause the ESP to crash.

zackees, 2025-01-09:
Use the new SPI clockless driver that just went into master.

closed 2025-01-09 by zackees