FastLED/FastLED #1511: Setting CHSV color inside EVERY_N_MILLISECONDS doesn't work

bmbouter, 2023-06-08:
With 3.6.0 running on esp 8266 I have an LED strip which works great with everything I use FastLED for (thanks!).

However let's say every 3 seconds I want to set LED 2 to red. It just stays black!

`EVERY_N_MILLISECONDS( 3000 ) { leds[1] = CHSV(0, 255, 255); }`

Sometime randomly after minutes, it will show up as red.

However if instead at that exact same line of code I replace it with ` leds[1] = CHSV(0, 255, 255);` it shows up perfectly as I expect (just without the delay functionality I'm looking for).

Something is wrong here, but I don't know what.

kostyamat, 2023-06-08:
EVERY_N_MILLISECONDS( 3000 ) { leds[1] = CHSV(0, 255, 255); 
FastLED.show();}

bmbouter, 2023-06-08:
Thank you for the reply. I was doing that, but the issue is indeed in my code not in FastLED.

I actually have N LED strips and I was putting the EVERY_N_MILLISECONDS in a loop which writes the same pattern to each strip. I (obviously now) was incorrectly expecting it to execute for each run through the for loop of N strips, but it was only being written to one and I don't have them all plugged in. :facepalm:

closed 2023-06-08 by bmbouter