FastLED/FastLED #1512: Esp32 Color is Changed

ruielias, 2023-06-09:
Hi,

I'm using an ESP3, on PlatformIO and using a "7Bit Ring".
When I configure it to light Red, the LEDs light Green...

And when I set it to light green, the leds light red...

What am I doing wrong?

`#include <Arduino.h>
#include <FastLED.h>

#define NUM_LEDS 7
#define DATA_PIN 13

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
    
 Serial.begin(115200);
 Serial.printf("\nStart\n");	

  FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is typical
  FastLED.setBrightness(5);
  delay(500);

  for(uint8_t x=0;x<6;x++)
  {
    leds[x] = CRGB::Red;
    FastLED.show();
    delay(25);
  }
}
void loop() { 
}`

PlatforIO:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = fastled/FastLED@^3.6.0

yesyesuk, 2023-06-09:
>> FastLED.addLeds<WS2812B, DATA_PIN, **RGB**>(leds, NUM_LEDS); // GRB ordering is typical

The "RGB" is the colour order of your string. Change that to GRB.
FastLED.addLeds<WS2812B, DATA_PIN, **GRB**>(leds, NUM_LEDS); // GRB ordering is typical

ruielias, 2023-06-09:
Hi yesyesuk,

It is already correct.
Thank you very much.

marmilicious, 2023-06-09:
Did you run the RGB calibrate example?

ruielias, 2023-06-09:
I didn't execute RGB calibrate example.
But he is already working according to the help of the "yesyesuk".

marmilicious, 2023-06-09:
Ok good. You can close this issue if it's solved.

closed 2024-11-18 by zackees