FastLED/FastLED #1517: Library does not work on Core 1 with ESP32-S3

alba-ado, 2023-07-03:
Hello,
First of all, thank you for creating this awesome library.

I am using this library inside the ESP-IDF with the ESP32-S3 board. While testing out some code, I noticed that the library does not work properly when pinned to Core 1 with the xTaskCreate. I am driving merely 16 less and it works fine on Core 0, but when I pin the task on Core 1, the lights either don't work at all or only a few LEDs get updated. I use WS2812B chips with the library. For some reason, when I set the whole frame to a fixed color it doesn't work at all. But when I set the frame with a code similar to Pride2015, just a few LEDs get updated but they work correctly. However, the rest of the LEDs don't work. I thought this was a priority issue of the tasks or because of the buffer size of the task, but changing neither made the library work on Core 1. However, when I switch the Core 0 everything works just fine. Since I have WiFi and ESP-Mesh code running on the Core 0, I don't want to bother with updating the LEDs which is a very low priority for me. This issue might be related to the one mentioned in #1438 but I'm not completely sure. That's why I opened a new issue.

For me, when I create it like this it works:
```
xTaskCreatePinnedToCore( DeviceInterface::frameTask, "Frame-Task", 2000, NULL, 2, NULL, PRO_CPU_NUM );
```

And when I create it like this, it doesn't work:
```
xTaskCreatePinnedToCore( DeviceInterface::frameTask, "Frame-Task", 2000, NULL, 2, NULL, APP_CPU_NUM );
```

I can't pinpoint the exact source of the issue since the code doesn't throw any errors, but I think the show function does not update the LEDs properly for some reason as mentioned in the other issue. The rest of the code seems to be working fine.

My task looks like this:
```
void DeviceInterface::frameTask( void * args ) {
    InterfaceState lastState = INTERFACE_STATE_UNKNOWN;

    while( true ) {
        
        if( currentState == INTERFACE_IDLE ) {
            LightFrame::frameRainbow();
        }
        
        delay(INTERFACE_LIGHT_FRAME_UPDATE_INTERVAL);
    }
}
```

But even if I create a task only for updating the LEDs, the same issue occurs.

Waiting for your help, best regards.

toybuilder, 2023-07-03:
There's other tasks running and competing for CPU cycles.  Try using I2S mode by prepending the #define before the #include.

```
#define FASTLED_ESP32_I2S 1
#include <FastLED.h>

```
In my own applications, I've found that FastLED does not work well without using I2S mode.  Downside is that if you need I2S hardware for other aspects of your system, you cannot use it for FastLED.

alba-ado, 2023-07-04:
I added the following line and it gave lots of errors. Also, I don't think the ESP32-S3 has I2S support. Et least not like the I2S hardware of the ESP32. They have changed the hardware. But I also noticed that it says:

```
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

```

So do I need to define SPI pins in some way?

Here is the full output:

```
In file included from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/fastled_esp32.h:10,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms.h:42,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/FastLED.h:62,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/include/Effects/LightFrame.h:6,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/include/BoardDrivers.h:36,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/src/System.h:4,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/main.cpp:1:
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:90:17: note: #pragma message: NOTE: ESP32 support using I2S parallel driver. All strips must use the same chipset
 #pragma message "NOTE: ESP32 support using I2S parallel driver. All strips must use the same chipset"
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/fastled_esp32.h:10,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms.h:42,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/FastLED.h:62,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/include/Effects/LightFrame.h:6,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/include/BoardDrivers.h:36,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/src/System.h:4,
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/main.cpp:1:
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::i2sInit()':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:456:34: error: 'I2S0O_DATA_OUT0_IDX' was not declared in this scope
             i2s_base_pin_index = I2S0O_DATA_OUT0_IDX;
                                  ^~~~~~~~~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:456:34: note: suggested alternative: 'I2S0O_SD1_OUT_IDX'
             i2s_base_pin_index = I2S0O_DATA_OUT0_IDX;
                                  ^~~~~~~~~~~~~~~~~~~
                                  I2S0O_SD1_OUT_IDX
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:461:34: error: 'I2S1O_DATA_OUT0_IDX' was not declared in this scope
             i2s_base_pin_index = I2S1O_DATA_OUT0_IDX;
                                  ^~~~~~~~~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:461:34: note: suggested alternative: 'I2S1O_WS_OUT_IDX'
             i2s_base_pin_index = I2S1O_DATA_OUT0_IDX;
                                  ^~~~~~~~~~~~~~~~~~~
                                  I2S1O_WS_OUT_IDX
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:470:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_msb_right = 1;
              ^~~~
              rx_conf
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:471:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_mono = 0;
              ^~~~
              rx_conf
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:472:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_short_sync = 0;
              ^~~~
              rx_conf
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:473:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_msb_shift = 0;
              ^~~~
              rx_conf
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:474:14: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
         i2s->conf.tx_right_first = 1; // 0;//1;
              ^~~~
              rx_conf
[... cut ...]
                 from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/main.cpp:1:
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/src/Interface/InterfaceController.h: In static member function 'static void DeviceInterface::frameTask(void*)':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/src/Interface/InterfaceController.h:42:20: warning: unused variable 'lastState' [-Wunused-variable]
     InterfaceState lastState = INTERFACE_STATE_UNKNOWN;
                    ^~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

alba-ado, 2023-10-09:
Any updates on this? The library simply don't work on Core 1 with esp32-s3. There are no other running tasks. Only a few LEDs get updated from time to time and the rest is not getting any updates.

zackees, 2023-10-09:
Just speculating - I think the interrupts are disabled globally while running WS2812. But when you have multi core the interrupts only get disabled for one of the chips, the one you aren't running the WS2812 on.

But I could be wrong.

alba-ado, 2023-10-10:
How can I try this? By calling the arduino function disableInterrupts before updating the LEDs?

zackees, 2023-10-11:
Try this:

```C++
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void some_function()
{
    // ... some code ...

    // Disable interrupts globally
    portDISABLE_INTERRUPTS();

    FastLED.show();

    // Re-enable interrupts globally
    portENABLE_INTERRUPTS();

    // ... remaining code ...
}
```

alba-ado, 2023-10-13:
> Just speculating - I think the interrupts are disabled globally while running WS2812. But when you have multi core the interrupts only get disabled for one of the chips, the one you aren't running the WS2812 on.
> 
> But I could be wrong.

Tried disabling the interrupts using portDISABLE_INTERRUPTS();
It didn't change anything.

samguyer, 2023-10-13:
I've seen this kind of behavior before. Are you using WiFi for anything? I think what happens is that ESP-IDF puts the whole WiFi stack on Core 1, whether you use it or not. So there's a bunch of super-high priority interrupts running on Core 1.

alba-ado, 2023-10-14:
On the contrary, the ESP WiFi works on Core 0, and the behavior is persistent whether the WiFi is turned on or not. I tried a bare minimum example without anything and it still holds true. Nothing runs on Core 1

alba-ado, 2023-10-14:
Here is the task list printed:

alba-ado, 2023-10-14:
Here is the task lis

alba-ado, 2023-10-14:
Here is the task list printed:

```
******************************************************
Task		State	Prio	Stack	Num	Core
******************************************************
ledTask        	R	1	7480	9	1
main           	X	1	2212	4	0
IDLE           	R	0	716	6	1
IDLE           	R	0	728	5	0
esp_timer      	S	22	3360	3	0
Tmr Svc        	B	3	1328	7	0
ipc1           	B	24	700	2	1
sys_evt        	B	20	2032	8	0
ipc0           	B	24	824	1	0
******************************************************
```

As seen from above, nothing runs on Core 1 except a task called ipc1. I also tried increasing the task priority of the leds above this task and it still doesn't work. Only a few leds at the beginning gets updated, rest remains unlit.

samguyer, 2023-10-14:
Oh right, I had the two cores backwards. I think Arduino sketches run on Core 1 by default and Wifi runs on Core 0. But that makes the behavior you are seeing even more puzzling because most of our users are probably using the Arduino toolchain, so they are running FastLED.show() on Core 1.

Maybe one thing to try: get one of the demo sketches (like DemoReel) and upload it to your board using the Arduino IDE. If that does not work, it means we need to look more carefully at how the S3 differs from the other ESP32 models. If it DOES work, we need to look at your code more carefully, and maybe look at the toolchain you're using.

alba-ado, 2023-10-16:
Thanks for your response. Your logic made a lot of sense to me (that the Arduino also runs on Core 1) and I started a new project to test things out. Apparently, the library does indeed work on Core 1. What was different from that example and my code was where I initialized the library. In my code, I initialized the library on the app_main() function (which runs on Core 0) and used it in a new task that runs on Core 1. When I wrote the test code, I added the library initialization on the same task and it work. So, I tested it by moving the initialization step into the same task in my code and it worked as well. What confused me a lot was a few LEDs were actually getting updated and the rest was not. Also, other libraries were again initialized on Core 0 and made to be working on Core 1, but they all ran fine without any change. However, this library requires initialization on the same core as the program that runs it. I also tried it initializing it in another task that runs on Core 1, it also works fine without a problem. I don't know what this behavior is but I am actually initializing the library before spawning the task. I guess the library init needs to be on the same core as the task, unlike other libraries I used.

closed 2023-10-16 by alba-ado

samguyer, 2023-10-16:
I was going to ask you if you were initializing on a different core from the main animation code. I think the problem has to do with the way interrupts are delivered. We set up the interrupt handler in the initialization, so I think there is a potential issue if the interrupts actually need to be delivered to the other core. Maybe it takes a little longer to reroute them, causing some of the data to get dropped. 

We should probably at least document this issue, so that others don't run into the same problem.

alba-ado, 2023-10-17:
> I was going to ask you if you were initializing on a different core from the main animation code. I think the problem has to do with the way interrupts are delivered. We set up the interrupt handler [...]

Documentation would be nice, since other libraries don't have a problem working this way. I use 7 other libraries inside the project the same way and this is the only one that caused an issue. Thanks for the help.