FastLED/FastLED #1518: Strange interaction between FastLED and ESPAsyncWebServer on ESP8266

craiglindley, 2023-07-03:
On a recent visit to a rock store I saw a lighted, color changing, crystal display stand that caught by eye. I thought I could 3D print one up and using a WEMOS D1 Mini and a single NeoPixel I had on hand I could create one with extended features. See the images below.

[image]
[image]

I decided to use FastLED and ESPAsyncWebServer to create a webpage to control the display stand.
[image]

With the web page I can turn the stand off and on, I can select two modes of operation: swepted which swepts the hue across the spectrum and stepped which switches between the 7 solid colors without fading between them.

All the webpage controls work as expected but with one strange problem. If I select the swepted mode I see all the colors of the spectrum repeated over and over. If, however, I select the stepped mode the first time through the color changes everything is fine but the second and subsequent times through the color changes the blue portion of the NeoPixel doesn't come on. If I again selected the swepted mode, things work as expected. I put a print statement in the stepped code and I can see the hue being set to 160, but the blue portion of the NeoPixel is off.

So there is some strange interaction between FastLED and the AsyncWebServer that I cannot understand.

When I first saw this problem I stepped back and wrote the following code to verify I was doing everything correctly.
```
/*
 * Crystal Light Test Code
 *
 * Hardware:
 *   Esp8266 D1 Mini
 *   A single NeoPixel
 *
 * Concept, design and implementaton by: Craig A. Lindley
 * Last Update: 07/02/2023
 */

#include <FastLED.h>

// Number of NeoPixels is 1
#define NUMBER_OF_PIXELS 1
#define PIXEL_PIN D4
#define BRIGHTNESS 128

// Define the array of leds
CRGB leds[NUMBER_OF_PIXELS];

// Number of steps in HSV color space between pure colors
#define COLOR_STEP_INC 32

/*********************************************************************/
/***                          Misc Functions                       ***/
/*********************************************************************/

void clearLED() {
  FastLED.clear(true);
  FastLED.show();
}

// Stepped display pattern
void displaySteppedPattern() {
  // Stepped up
  for (int i = 0; i < 8; i++) {
    leds[0].setHue(i * COLOR_STEP_INC);
    FastLED.show();
    delay(2000);
    clearLED();
    delay(500);
  }
}

/*********************************************************************/
/***                          Program Setup                        ***/
/*********************************************************************/

void setup() {

  // Configure and initialize FastLED library
  FastLED.addLeds<NEOPIXEL, PIXEL_PIN>(leds, NUMBER_OF_PIXELS);
  FastLED.setBrightness(BRIGHTNESS);
[... 359 lines cut ...]
  if (now - last_wifi_check_time > WIFI_TIMEOUT) {
    Serial.print("Checking WiFi... ");
    if (WiFi.status() != WL_CONNECTED) {
      Serial.println("WiFi connection lost. Restarting...\n\n\n");
      delay(3000);
      ESP.restart();
    } else {
      Serial.println("WiFi OK");
    }
    last_wifi_check_time = now;

    // Now check for Internet connectivity
    if (checkInternetConnection() == false) {
      Serial.println("WiFi connection lost. Restarting...\n\n\n");
      delay(3000);
      ESP.restart();
    }
  }
  ws.cleanupClients();
}
```

The swepted mode works fine and the stepped mode works correctly once and then the blue portion of the NeoPixel refuses to come on but the red and green portions do. 

Strange huh? I don't know if the issue is with FastLED or the async server code.

craiglindley, 2023-08-30:
> This question reallh isn't a bugreport and would thus be better suited for
> the reddit /r/FastLED group as directed. That'll get you a bigger audience
> of experts. Once you copy there, please close this issue so the maintainers
> can focus on bugreports of their own product.
>
> Offhand, I don't see any flagrant issue, but I can't debug nested state
> machine and HSV values in my head at this hour. Since I'm distrustful of
> all hardware - even my own (hell, especially my own!) - I'd try just
> alternating between CRGB::Blue, CRGB::Red, and CRGB::Green and just
> confirming that all three "halves" of the bulb are even functional.
> Software debugging with a broken Neopixel can make you really crazy. Also,
> since it's color sensitive, be sure you really do hit all the Hue values
> you think you do. Since hue blue is 240 (right?) you should also verify
> that your multiplications don't overflow a uint8 or somthing. I'd also be
> suspicious of that "224 - " if your blue value is really 240 and
> pencil-whip/debug-print that. {/shrug}
>
> Good luck.

Thanks for your reply though I have moved on from this project. I think I
did most of what you recommended with no luck. I'm sure all sections of the
NeoPixel are working because in other modes like I mentioned the blue signs
through.

craiglindley, 2023-08-30:
I'm closing this on the above suggestion as I no longer require a fix as I have done something else.

closed 2023-08-30 by craiglindley

Jaibpi96, 2023-12-12:
¡SOLUCIONADO! El problema esta en el core de ESP8266 para arduino. Haciendo downgrade a version 2.7.4 vuelve a funcionar como antes. Voy a probar versiones mas recientes pero 2.7.4 funciona con FastLed sin problemas.

Para quien no sepa cambiar la versión del core : Herramientas/Placa/Gestor de tarjetas y ahi buscamos esp8266 y cambiamos a versión mas antigua.

SEÑORES DE ESPRESSIF ME HABEIS VUELTO LOCO DURANTE DIAS...