FastLED/FastLED #1519: Library fails to compile for the IDF version v5.1-rc2 for ESP32-S3

alba-ado, 2023-07-05:
Hello, I've been trying to compile this library for the ESP-IDF version 5.1-rc2. However, I get the following errors:

```
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:130,
                 from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/fastled_esp32.h:12,
                 from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms.h:42,
                 from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/FastLED.h:62,
                 from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:7:
/home/ado/Programs/Espressif/esp-idf-v5.1/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:131:
/home/ado/Programs/Espressif/esp-idf-v5.1/components/driver/deprecated/driver/periph_ctrl.h:7:2: warning: #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions. [-Wcpp]
    7 | #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions.
      |  ^~~~~~~
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/FastLED.h:58:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:86:49: warning: 'volatile'-qualified parameter is deprecated [-Wvolatile]
   86 |         inline void set(FASTLED_REGISTER port_t val) __attribute__ ((always_inline)) { *mPort = val; }
      |                                          ~~~~~~~^~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:91:87: warning: 'volatile'-qualified parameter is deprecated [-Wvolatile]
   91 |         inline void fastset(FASTLED_REGISTER port_ptr_t port, FASTLED_REGISTER port_t val) __attribute__ ((always_inline)) { *port  = val; }
      |                                                                                ~~~~~~~^~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:94:9: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
   94 |         port_t hival() __attribute__ ((always_inline)) { return *mPort | mPinMask;  }
      |         ^~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:96:9: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
   96 |         port_t loval() __attribute__ ((always_inline)) { return *mPort & ~mPinMask; }
      |         ^~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:100:9: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
  100 |         port_t mask() __attribute__ ((always_inline)) { return mPinMask; }
      |         ^~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h: In member function 'void Pin::hi()':
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:67:67: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   67 |         inline void hi() __attribute__ ((always_inline)) { *mPort |= mPinMask; }
      |                                                            ~~~~~~~^~~~~~~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h: In member function 'void Pin::lo()':
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:69:67: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   69 |         inline void lo() __attribute__ ((always_inline)) { *mPort &= ~mPinMask; }
      |                                                            ~~~~~~~^~~~~~~~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h: In member function 'void Pin::hi(port_ptr_t)':
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:79:98: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   79 |         inline void hi(FASTLED_REGISTER port_ptr_t port) __attribute__ ((always_inline)) { *port |= mPinMask; }
      |                                                                                            ~~~~~~^~~~~~~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h: In member function 'void Pin::lo(port_ptr_t)':
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:82:98: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   82 |         inline void lo(FASTLED_REGISTER port_ptr_t port) __attribute__ ((always_inline)) { *port &= ~mPinMask; }
      |                                                                                            ~~~~~~^~~~~~~~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h: At global scope:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:290:56: warning: 'volatile'-qualified parameter is deprecated [-Wvolatile]
  290 |         inline static void set(FASTLED_REGISTER port_t val) __attribute__ ((always_inline)) { }
      |                                                 ~~~~~~~^~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:293:94: warning: 'volatile'-qualified parameter is deprecated [-Wvolatile]
  293 |         inline static void fastset(FASTLED_REGISTER port_ptr_t port, FASTLED_REGISTER port_t val) __attribute__ ((always_inline)) { }
      |                                                                                       ~~~~~~~^~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:296:16: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
  296 |         static port_t hival() __attribute__ ((always_inline)) { return 0; }
      |                ^~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:298:16: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
  298 |         static port_t loval() __attribute__ ((always_inline)) { return 0;}
      |                ^~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:302:16: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
  302 |         static port_t mask() __attribute__ ((always_inline)) { return 0; }
      |                ^~~~~~
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/fastled_esp32.h:3:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK, VALIDPIN>::toggle()':
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/fastpin_esp32.h:43:15: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   43 |       *port() ^= MASK;
      |       ~~~~~~~~^~~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: At global scope:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:166:47: warning: ignoring attribute 'section (".iram1.8")' because it conflicts with previous 'section (".iram1.1")' [-Wattributes]
[... 123 lines cut ...]
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:82:98: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   82 |         inline void lo(FASTLED_REGISTER port_ptr_t port) __attribute__ ((always_inline)) { *port &= ~mPinMask; }
      |                                                                                            ~~~~~~^~~~~~~~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h: At global scope:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:290:56: warning: 'volatile'-qualified parameter is deprecated [-Wvolatile]
  290 |         inline static void set(FASTLED_REGISTER port_t val) __attribute__ ((always_inline)) { }
[... cut ...]
      |                ^~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/FastLED/src/fastpin.h:302:16: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
  302 |         static port_t mask() __attribute__ ((always_inline)) { return 0; }
      |                ^~~~~~

```

From what I understand, they changed the low-level access methods in the 5.1 version. I can compile the library for the 4.4 version but when I try to compile it with the new IDF it fails.

Thanks for your help

iharshadec, 2023-07-19:
I am using Arduino and ESP Board Manager v2.0.9 and FastLED is not working with ESP32-S3.

alba-ado, 2023-07-20:
The F_CPU was not defined in the older version of the library. I think if you get the newer version it will work. The problem I'm having with IDF v5.1

alba-ado, 2023-09-01:
Hello,

I don't think PlatformIO can run the IDF version 5.2 since there is no 5.2 version in the esp-idf GitHub. It might be PlatformIO's own espressif core version v5.2. The latest version of the idf right now is v5.1.1, and the latest version Arduino Core supports is v4.4.5.

But I use the [release/v5.1] branch from GitHub. The reason why PlatformIO doesn't support the v5 yet is that the Arduino Core itself doesn't work with the idf fully at this moment. However, there is a development branch and I use that instead: esp-idf-v5.1-libs. 

Here are the links to those branches:

ESP-IDF: https://github.com/espressif/esp-idf/tree/release/v5.1
Arduino as a component: https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs

Here is the example I created for you:

https://github.com/alba-ado/fastled_idf_trial

You need to follow the steps on the espressif's page to install the idf and its tools. And add the Arduino repo folder to the components folder. IDF requires you to clone their repository using "--recursive" to get all of the components needed.

It builds fine when I comment on the FastLED from the CMakeLists.txt inside the main. But when I include it, it fails to compile. Here are the updated errors for the latest version of the FastLED branch;

```
In file included from /home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:130,
                 from /home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/platforms/esp/32/fastled_esp32.h:12,
                 from /home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/platforms.h:42,
                 from /home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/FastLED.h:62,
                 from /home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/noise.cpp:6:
/home/ado/Programs/Espressif/esp-idf-v5.1/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
In file included from /home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:131:
/home/ado/Programs/Espressif/esp-idf-v5.1/components/driver/deprecated/driver/periph_ctrl.h:7:2: warning: #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions. [-Wcpp]
    7 | #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions.
      |  ^~~~~~~
In file included from /home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/FastLED.h:58:
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:86:49: warning: 'volatile'-qualified parameter is deprecated [-Wvolatile]
   86 |         inline void set(FASTLED_REGISTER port_t val) __attribute__ ((always_inline)) { *mPort = val; }
      |                                          ~~~~~~~^~~
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:91:87: warning: 'volatile'-qualified parameter is deprecated [-Wvolatile]
   91 |         inline void fastset(FASTLED_REGISTER port_ptr_t port, FASTLED_REGISTER port_t val) __attribute__ ((always_inline)) { *port  = val; }
      |                                                                                ~~~~~~~^~~
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:94:9: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
   94 |         port_t hival() __attribute__ ((always_inline)) { return *mPort | mPinMask;  }
      |         ^~~~~~
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:96:9: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
   96 |         port_t loval() __attribute__ ((always_inline)) { return *mPort & ~mPinMask; }
      |         ^~~~~~
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:100:9: warning: 'volatile'-qualified return type is deprecated [-Wvolatile]
  100 |         port_t mask() __attribute__ ((always_inline)) { return mPinMask; }
      |         ^~~~~~
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h: In member function 'void Pin::hi()':
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:67:67: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   67 |         inline void hi() __attribute__ ((always_inline)) { *mPort |= mPinMask; }
      |                                                            ~~~~~~~^~~~~~~~~~~
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h: In member function 'void Pin::lo()':
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:69:67: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   69 |         inline void lo() __attribute__ ((always_inline)) { *mPort &= ~mPinMask; }
      |                                                            ~~~~~~~^~~~~~~~~~~~
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h: In member function 'void Pin::hi(port_ptr_t)':
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/fastpin.h:79:98: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
[... cut ...]
/home/ado/Projects/Espressif/FastLED_Trial/main/lib/FastLED/src/platforms/esp/32/fastpin_esp32.h:43:15: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   43 |       *port() ^= MASK;
      |       ~~~~~~~~^~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ado/Projects/Espressif/FastLED_Trial/build/log/idf_py_stderr_output_41920 and /home/ado/Projects/Espressif/FastLED_Trial/build/log/idf_py_stdout_output_41920
```

zackees, 2024-08-08:
Compiler error fixed in FastLED 3.7.1 bugfix release.

closed 2024-08-08 by zackees