FastLED/FastLED #1520: Pixels not showing when void with a for-loop is called from setup

Shumatic, 2023-07-06:
Hi

I try to figure our what i am doing wrong, and hope someone could help.
When i run the code below on a generic esp8266, no pixels are lit up.
if i uncomment either   //FastLED.show(); or   //neo(); it works
or if i move FastLED.show(); into the for loop.

i dosnt make sense to me.

`
#include <FastLED.h>
#define NUM_LEDS 60
#define DATA_PIN 4
CRGB leds[NUM_LEDS];

void setup() {
  Serial.begin(74880);
  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
  PixleVerNo();
  //FastLED.show();
  //neo();
}
void neo() {
  leds[15] = CRGB::Red;
  FastLED.show();
}
void loop() {
}
uint8_t p;
void PixleVerNo() {
  for (p = 0; p <= 6; p++) {
    Serial.printf("pix%d ", p);
    leds[p] = CRGB(0, 0, 100);  //space version digit with 20
  }
  FastLED.show();
}`

Shumatic, 2023-08-30:
@robertlipe 
Thanks for taking the time to answer :-)
Yes, as i wrote, the sketch dosnt work as is on a ESP8266, that's what bother me. 
for fun i just testet it in wokwi, there it works.......
I am just trying to grasp why it dosn't work. but i just found anoter way, so no big deal.

Baudrate is set to match ESP native baud, so i can see boot mode etc.
It is a stripped sketch, only to show the "problem" thats the reason for 60 neo etc.

closed 2023-08-30 by Shumatic

samguyer, 2023-11-17:
@Shumatic The mostly likely issue in this version is that the loop() function never returns. The reason that's a problem is that there is often other stuff going on in the processor besides your sketch. The sketch is usually run as part of a primitive operating system that looks like this:

1. Initialize runtime systems
2. Call user's setup() function one time
3. Loop forever:
     -- Call user's loop() function
     -- Take care of other tasks on the processor

When you have a loop() function that doesn't return, the "take care of other tasks" stuff never happens. On some kinds of processors it doesn't matter, but I think on the ESP processors there are things like watchdog timers, which will automatically reboot the processor if they are not reset at some regular interval (as a safety mechanism).

Shumatic, 2023-11-17:
Thanks for your kind comments
I am aware of this, and originally it wasn't an empty loop. But my understanding was that it would just light the Led as defined in setup. if I put show after calling the void, it works. So it is not a problem. I am just trying to figure out why it's working as this (or not working)

samguyer, 2023-11-17:
@Shumatic Ah ok, now I see what you're saying. Even that one call to FastLED.show() in the PixelVerNo() function does not work. I wonder if the choice of pins is giving you trouble. Can you try pin 4 or 5? The names are also very confusing for ESP8266. For example, GPIO4 is also known as D2.

Shumatic, 2023-11-18:
@samguyer 
No exactly this call should light up the pix as it do in wokwi, it is quite strange
The Pin2 was only from wokwi I have used pin4 in my hw.