FastLED/FastLED #1524: Library fails to compile under toolchain-xtensa-esp32s3 @ 12.2.0+20230208

daavery, 2023-07-12:
.pio/libdeps/M5AtomS3/FastLED/src/noise.cpp: In function 'void fill_raw_2dnoise8(uint8_t*, int, int, uint8_t, q44, fract8, int, uint16_t, int, uint16_t, uint, uint16_t)':
.pio/libdeps/M5AtomS3/FastLED/src/noise.cpp:596:100: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')
  596 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);
      |                                                                                             ~~~~~~ ^ ~~~~~~
      |                                                                                             |        |
      |                                                                                             |        int
      |                                                                                             q44 {aka q<unsigned char, 4, 4>}
In file included from .pio/libdeps/M5AtomS3/FastLED/src/pixeltypes.h:7,
                 from .pio/libdeps/M5AtomS3/FastLED/src/controller.h:9,
                 from .pio/libdeps/M5AtomS3/FastLED/src/FastLED.h:49,
                 from .pio/libdeps/M5AtomS3/FastLED/src/noise.cpp:2:
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:833:14: note: candidate: 'uint32_t q<T, F, I>::operator*(uint32_t) [with T = unsigned char; int F = 4; int I = 4; uint32_t = long unsigned int]'
  833 |     uint32_t operator*(uint32_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:834:14: note: candidate: 'uint16_t q<T, F, I>::operator*(uint16_t) [with T = unsigned char; int F = 4; int I = 4; uint16_t = short unsigned int]'
  834 |     uint16_t operator*(uint16_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:835:13: note: candidate: 'int32_t q<T, F, I>::operator*(int32_t) [with T = unsigned char; int F = 4; int I = 4; int32_t = long int]'
  835 |     int32_t operator*(int32_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:836:13: note: candidate: 'int16_t q<T, F, I>::operator*(int16_t) [with T = unsigned char; int F = 4; int I = 4; int16_t = short int]'
  836 |     int16_t operator*(int16_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
.pio/libdeps/M5AtomS3/FastLED/src/noise.cpp:596:127: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'uint' {aka 'unsigned int'})
  596 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);
      |                                                                                                                        ~~~~~~ ^ ~~~~~~
      |                                                                                                                        |        |
      |                                                                                                                        |        uint {aka unsigned int}
      |                                                                                                                        q44 {aka q<unsigned char, 4, 4>}
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:833:14: note: candidate: 'uint32_t q<T, F, I>::operator*(uint32_t) [with T = unsigned char; int F = 4; int I = 4; uint32_t = long unsigned int]'
  833 |     uint32_t operator*(uint32_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:834:14: note: candidate: 'uint16_t q<T, F, I>::operator*(uint16_t) [with T = unsigned char; int F = 4; int I = 4; uint16_t = short unsigned int]'
  834 |     uint16_t operator*(uint16_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:835:13: note: candidate: 'int32_t q<T, F, I>::operator*(int32_t) [with T = unsigned char; int F = 4; int I = 4; int32_t = long int]'
  835 |     int32_t operator*(int32_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
.pio/libdeps/M5AtomS3/FastLED/src/lib8tion.h:836:13: note: candidate: 'int16_t q<T, F, I>::operator*(int16_t) [with T = unsigned char; int F = 4; int I = 4; int16_t = short int]'
  836 |     int16_t operator*(int16_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~

there is no overload for freq44*scalex or scaley - which are prototyped as (int) but are q44 {aka q<unsigned char, 4, 4>}* uint {aka unsigned int}

this is probably due to an upgrade of gcc between  toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5  and toolchain-xtensa-esp32s3 @ 12.2.0+20230208   

if i keep toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5   then platform = espressif32@6.3.2   build the library with no problems

this is building for an M5Stack AtomS3

smurphyUSI, 2023-08-01:
I don't have an S3 to test with, but could you try the following.
In noise.cpp, on line 623 (according to github), you should find this line:

fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);

Try reversing the order of the multiply operation freq44 functions, to make sure the freq44 is always before (or after) the operator.

Can you try this:
fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, freq44*x, freq44*scalex, freq44*y, freq44*scaley, time);

And, this:
fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, scalex*freq44, y*freq44, scaley*freq44, time);

I suspect that only that second option will work.

daavery, 2023-08-02:
neither worked , the fix  is  

    fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, (uint16_t)scalex *  freq44  , y *freq44, (uint16_t)scaley *  freq44  , time);
.
.
.
    fill_raw_2dnoise16(pData, width, height, octaves-1, freq88, amplitude, skip, x *freq88 , (uint32_t)scalex * freq88 , y * freq88, (uint32_t)scaley  , time);
.
.
.
    fill_raw_2dnoise16into8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, (uint32_t)scalex *freq44, y*freq44, (uint32_t)scaley * freq44, time);

smurphyUSI, 2023-08-03:
Thanks for reporting back on this.

This means that the error is likely with incorrect automatic type conversion which aren't handled by template class 'q' in lib8tion.h.

If you would like to help in tracking down the issue, could I ask you check a few things.
In lib8tion.h, comment out these sections:
```
#if defined(FASTLED_ARM) | defined(FASTLED_RISCV) | defined(FASTLED_APOLLO3)
    /// @copydoc operator*(uint32_t)
    int operator*(int v) { return (v*i) + ((v*f)>>F); }
#endif

```
```
#if defined(FASTLED_ARM) | defined(FASTLED_RISCV) | defined(FASTLED_APOLLO3)
template<class T, int F, int I> static int operator*(int v, q<T,F,I> & q) { return q * v; }
#endif
```

Also, can see if FASTLED_RISCV get defined instead of FASTLED_XTENSA between your two different toolchain builds?

As this was more than a few weeks ago, I don't expect you to go back to check this.
If this isn't a unique problem, and is based on permanent toolchain changes, then this will come up again later.

Thanks again

Shahzad3212, 2023-08-20:
FastLED.h not suport  inlibaray

Shahzad3212, 2023-08-20:
FastLED.h not suport in libaray

LeCyberDucky, 2024-03-03:
@smurphyUSI I'm also encountering `.pio/libdeps/lilygo-t-display-s3/FastLED/src/noise.cpp:628:100: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')`. 

Is there anything I can do to help investigating this? 
I'm using: 

 - framework-arduinoespressif32 @ 3.0.0+sha.cceebb5
 - framework-arduinoespressif32-libs @ 5.1.0+sha.a5e3e0b90e.sha.6564ec9
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208

zackees, 2024-08-08:
Fixed a while ago.

closed 2024-08-08 by zackees