FastLED/FastLED #1526: the sketch cant find FastLed.h

Ilay-HH, 2023-07-23:
Hi, kinda tried everything at this point. basically i cant run any example project of the library because of this error, linking a postbin link and copying the error as well.

https://pastebin.com/fQeJZtxP

C:\Users\ilayh\AppData\Local\Temp\.arduinoIDE-unsaved2023623-32324-c7girr.myvco\DemoReel100\DemoReel100.ino:1:10: fatal error: FastLED.h: No such file or directory
 #include <FastLED.h>
          ^~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: FastLED.h: No such file or directory

I'd love for some help on the matter

smurphyUSI, 2023-08-01:
In the Arduino IDE, select "Manage Libraries..." from the Sketch menu.

Search for 'FastLED' and install either 3.5.0 or the most recent 3.6.0.

Then, add the include to your sketch.
Depending on which version of the IDE you have, it will be one of these two lines of text (not both).
#include <FastLED.h>
#include "FastLED.h"

samguyer, 2023-09-03:
Thanks @smurphyUSI and @robertlipe. @Ilay-HH can we close this ticket?

Ilay-HH, 2023-09-03:
Sorry for the delay, thanks @smurphyUSI everything is working now!
And yeah the ticket can be closed, thank you very much everyone.

closed 2023-09-03 by Ilay-HH