FastLED/FastLED #1527: When doing CFastLED::addLeds, Clang-Tidy is issuing "In template: invalid output constraint '+h' in asm"

divStar, 2023-07-27:
First: this is an awesome project and thank you very much for sharing it with us!

## TL;DR:
**Versions:**
- CLion v2023.1.3
- FastLed 3.6.0
- PlatformIO 6.1.9
- Apple Silicon M1 Pro, MacOS 13.5
- `arm-none-eabi-gcc` and `arm-none-eabi-g++` v7.2.1 20170904

**Problem:**
- Clang-Tidy reports the error "In template: invalid output constraint '+h' in asm" on `CFastLed::addLeds`
- error does not prevent build and from what it seems a successful execution on Arduino-based device
- "error occured here" reports `m0clockless.h`, line 279

I have the following code snippet (amongst others):
```c++
    CFastLED::addLeds<LED_TYPE, DATA_PIN, COLOR_ORDER>(leds, NUM_LEDS);
```

The application builds and seems to run just fine, but I thought I'd address this issue regardless.
Clang-Tidy marks the snippet above as an error (on `addLeds`), giving the following message: `In template: invalid output constraint '+h' in asm`. The "error occurred here" link in that tooltip sends me to `m0clockless.h`, line 279 (it marks `M0_ASM_ARGS` as an error, but the supposed error message stays the same).

My project is a PlatformIO (v6.1.9) project, that I am working on using CLion v2023.1.3.
Here is the `platformio.ini`:
```ini
[env:zeroUSB]
platform = atmelsam
board = zeroUSB
framework = arduino
upload_speed = 115200
upload_port = /dev/cu.usbmodem101
lib_deps =
	fastled/FastLED@^3.6.0
	arkhipenko/TaskScheduler@^3.7.0
	bblanchon/ArduinoJson@^6.21.2
	janelia-arduino/Vector@^1.2.2
	thijse/ArduinoLog@^1.1.1
	arduino-libraries/RTCZero@^1.6.0

[platformio]
description = Firmware
```

The device is an Arduino Zero based device. I can elaborate on the device if it helps.

As mentioned before, the code still compiles and runs on the device as expected.

Does anyone know where this "error" comes from and how to fix it?

divStar, 2023-08-03:
For some reason this error started to cause errors when compiling now:

```
====================[ Build | Production | zeroUSB ]============================
"/Users/igvo/Library/Application Support/JetBrains/Toolbox/apps/CLion/ch-0/231.9011.31/CLion.app/Contents/bin/cmake/mac/bin/cmake" --build /Users/igvo/CLionProjects/dialer/cmake-build-zerousb --target Production -j 8
[1/1] cd /Users/igvo/CLionProjects/dialer && platformio -c clion run -ezeroUSB
FAILED: CMakeFiles/Production /Users/igvo/CLionProjects/dialer/cmake-build-zerousb/CMakeFiles/Production 
cd /Users/igvo/CLionProjects/dialer && platformio -c clion run -ezeroUSB
Processing zeroUSB (platform: atmelsam; board: zeroUSB; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/zeroUSB.html
PLATFORM: Atmel SAM (8.2.0) > Arduino Zero (USB Native Port)
HARDWARE: SAMD21G18A 48MHz, 32KB RAM, 256KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink)
PACKAGES: 
 - framework-arduino-samd @ 1.8.13 
 - framework-cmsis @ 1.40500.0 (4.5.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 22 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FastLED @ 3.6.0
|-- TaskScheduler @ 3.7.0
|-- ArduinoJson @ 6.21.3
|-- ArduinoLog @ 1.1.1
|-- RTCZero @ 1.6.0
|-- LinkedList @ 0.0.0-alpha+sha.dac3874d28
|-- CapacitiveSensor @ 0.5.1
|-- SimpleKalmanFilter @ 0.1.0
|-- mpu6050-custom
Building in release mode
Compiling .pio/build/zeroUSB/src/led/LedTask.cpp.o
/var/folders/4l/1kxmc4190k129ggkn9snkw0c0000gn/T//ccN6mJ2t.s: Assembler messages:
/var/folders/4l/1kxmc4190k129ggkn9snkw0c0000gn/T//ccN6mJ2t.s:3727: Warning: stray `\'
/var/folders/4l/1kxmc4190k129ggkn9snkw0c0000gn/T//ccN6mJ2t.s:3727: Error: unexpected end of file in macro `dither5' definition
/var/folders/4l/1kxmc4190k129ggkn9snkw0c0000gn/T//ccN6mJ2t.s: Error: end of file inside conditional
/var/folders/4l/1kxmc4190k129ggkn9snkw0c0000gn/T//ccN6mJ2t.s:3727: Error: here is the start of the unterminated conditional
*** [.pio/build/zeroUSB/src/led/LedTask.cpp.o] Error 1
========================== [FAILED] Took 1.45 seconds ==========================
ninja: build stopped: subcommand failed.
```
What I did, was pass `SerialUSB` by reference into `LedTask.cpp` (and `LedTask.h`) - and suddenly it started breaking...

smurphyUSI, 2023-08-04:
The problem could be using Clang in that specific toolchain.
Unless you are using GCC, you have may have a compatibility issue with Extended ASM.
I don't use much Clang, but it should work with the Clang-LLVM toolchain as well.

closed 2024-09-20 by zackees