FastLED/FastLED #1528: Read data for WS2812b

SlimRG, 2023-07-30:
Can I read (not send) data from controller to WS2812b?
I want to parse this data and add some filters, so I connect controller to my Arduino and WS2812b connect to my arduino too.

marmilicious, 2023-08-15:
When the r,g,b data from CRGB leds is sent out to a LED strip any color correction and master brightness scaling is applied on the fly at that time. If you don't use any color correction in your addLeds line and also keep FastLED master brightness at 255, then the data in CRGB leds should be the same as what's sent out to the LED strip.
Thus you should be able to send what's in CRGB leds to another controller and it should be the same r,g,b data that was sent to the LED strip.

samguyer, 2023-09-03:
@SlimRG Can we close this issue?

closed 2025-01-13 by zackees