FastLED/FastLED #1531: Sketch fails to compile on Adafruit Grand Central M4 Express: "fastpin.h:261:24: error: static assertion failed: Invalid pin specified"

JNordgren42, 2023-08-25:
I'm unable to compile even the simplest sketch using FastLED on an Adafruit Grand Central M4 Express with DotStar LEDs. 

12 pixel DotStar strip connected to the Hardware SPI pins (Pin 51 Data, Pin 52 Clock). Power from board 5V, Ground to board GND. 

Arduino IDE version 2.1.1, 06/30/2023
Adafruit SAMD Boards package version 1.7.13
FastLED version 3.6.0

This configuration functions perfectly using the Adafruit DotStar library. I have confirmed the board definition is in **\FastLED\src\platforms\arm\d51\fastpin_arm_d51.h** and the pins appear to be correctly mapped, specifically:

_#define SPI_DATA 51_
_#define SPI_CLOCK 52_

_#define HAS_HARDWARE_PIN_SUPPORT 1_

I have also tried several other pin combinations that are included in the mapping and receive the same error.

Sketch:
```
#include <FastLED.h>

#define NUM_LEDS 12

#define DATA_PIN 3
#define CLOCK_PIN 2

CRGB leds[NUM_LEDS];

void setup() { 
  FastLED.addLeds<DOTSTAR, DATA_PIN, CLOCK_PIN>(leds, NUM_LEDS);
}

void loop() { 
  leds[0] = CRGB::Red; 
  FastLED.show(); 
  delay(30); 
}
```
Serial Output:

In file included from C:\Users\jason\OneDrive\Documents\USS Enterprise\Grand Central M4\FastLED\Blink\Blink.ino:1:
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/FastLED.h:665:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]
  665 | #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."
      |  ^~~~~~~
In file included from c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from C:\Users\jason\OneDrive\Documents\USS Enterprise\Grand Central M4\FastLED\Blink\Blink.ino:1:
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/FastLED.h:58,
                 from C:\Users\jason\OneDrive\Documents\USS Enterprise\Grand Central M4\FastLED\Blink\Blink.ino:1:
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<3>':
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastspi_bitbang.h:26:49:   required from 'class AVRSoftwareSPIOutput<3, 2, 10>'
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastspi.h:40:7:   required from 'class SPIOutput<3, 2, 10>'
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/chipsets.h:169:6:   required from 'class LPD6803Controller<3, 2, RGB, 10>'
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/FastLED.h:284:66:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = DOTSTAR; unsigned char DATA_PIN = 3; unsigned char CLOCK_PIN = 2]'
C:\Users\jason\OneDrive\Documents\USS Enterprise\Grand Central M4\FastLED\Blink\Blink.ino:11:63:   required from here
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastpin.h:261:24: error: static assertion failed: Invalid pin specified
  261 |  static_assert(validpin(), "Invalid pin specified");
      |                ~~~~~~~~^~
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<2>':
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastspi_bitbang.h:27:50:   required from 'class AVRSoftwareSPIOutput<3, 2, 10>'
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastspi.h:40:7:   required from 'class SPIOutput<3, 2, 10>'
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/chipsets.h:169:6:   required from 'class LPD6803Controller<3, 2, RGB, 10>'
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/FastLED.h:284:66:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = DOTSTAR; unsigned char DATA_PIN = 3; unsigned char CLOCK_PIN = 2]'
C:\Users\jason\OneDrive\Documents\USS Enterprise\Grand Central M4\FastLED\Blink\Blink.ino:11:63:   required from here
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastpin.h:261:24: error: static assertion failed: Invalid pin specified

exit status 1

Compilation error: exit status 1

JNordgren42, 2023-08-26:
Adding verbose logging:

FQBN: adafruit:samd:adafruit_grandcentral_m4
Using board 'adafruit_grandcentral_m4' from platform in folder: C:\Users\jason\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13
Using core 'arduino' from platform in folder: C:\Users\jason\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13

Detecting libraries used...
"C:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -Werror=return-type -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"Blink.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10607 -DARDUINO_GRAND_CENTRAL_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\"" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\"" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/Core/Include/" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/DSP/Include/" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS-Atmel\\1.2.2/CMSIS/Device/ATMEL/" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13\\cores\\arduino" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13\\variants\\grand_central_m4" "C:\\Users\\jason\\AppData\\Local\\Temp\\arduino\\sketches\\75C77BDE71F9463AFF121EDA6BBA8CA4\\sketch\\Blink.ino.cpp" -o nul
Alternatives for FastLED.h: [FastLED@3.6.0]
ResolveLibrary(FastLED.h)
  -> candidates: [FastLED@3.6.0]
"C:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -Werror=return-type -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"Blink.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10607 -DARDUINO_GRAND_CENTRAL_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\"" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\"" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/Core/Include/" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.4.0/CMSIS/DSP/Include/" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS-Atmel\\1.2.2/CMSIS/Device/ATMEL/" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13\\cores\\arduino" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13\\variants\\grand_central_m4" "-Ic:\\Users\\jason\\OneDrive - Associated Building Supply\\Documents\\Arduino\\libraries\\FastLED\\src" "C:\\Users\\jason\\AppData\\Local\\Temp\\arduino\\sketches\\75C77BDE71F9463AFF121EDA6BBA8CA4\\sketch\\Blink.ino.cpp" -o nul
"C:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -Werror=return-type -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"Blink.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10607 -DARDUINO_GRAND_CENTRAL_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\"" "-IC:\\Users\\jason\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.13/libraries/Adafruit_TinyUSB_Ardu [... cut ...] signed char CLOCK_PIN = 52]'
C:\Users\jason\OneDrive\Documents\USS Enterprise\Grand Central M4\FastLED\Blink\Blink.ino:11:63:   required from here
c:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED\src/fastpin.h:261:24: error: static assertion failed: Invalid pin specified

Using library FastLED at version 3.6.0 in folder: C:\Users\jason\OneDrive - Associated Building Supply\Documents\Arduino\libraries\FastLED 
exit status 1

Compilation error: exit status 1

jgildred, 2023-08-30:
> I don't have that hardware, but i've stared at those errors (eek!) and sure don't see the problem. I suspect it's the same as #1071, but I don't speak Arduino fluently. (Well, at all...)
> @jgildred was the committer of that support. Can you please offer any assistance here?

Been a while since I looked at it. May have to revert until it’s resolved as I can’t find the problem either.

samguyer, 2023-09-03:
@JNordgren42 and @jgildred Should we track this issue on the old ticket (#1071 ?)

JNordgren42, 2023-09-05:
Whatever is easiest for you to track, I'm fine either way. Also, please let
me know if there are any tests you need me to run and I'll be happy to
assist.

bwmaas, 2023-09-09:
I just submitted PR #1539 to fix a bug in `platform.h`. I didn't see this bug, but I'm testing using Neopixels.

JNordgren42, 2023-09-10:
> I just submitted PR #1539 to fix a bug in `platform.h`. I didn't see this bug, but I'm testing using Neopixels.

Thanks! Just FYI, I believe I've tried using the onboard Neopixel and get similar results (I'm not positive on that, but I'm pretty sure I did). I can try and run it again and dump the error log from that if it helps.

JNordgren42, 2023-09-15:
The updated platforms.h from PR #1539 resolved the issue. Thanks so much!

closed 2023-09-15 by JNordgren42

bwmaas, 2023-09-15:
You should maybe leave this open until #1539 is merged.

bwmaas, 2023-09-15:
> I'd suggest closing this one and emphasizing your confirmed success in
> that
> one.
>
> The castles maintainers are drowning in requests. Anything to reduce the
> flood and confirm the success of a. Existing solution seems likely to help
> them find clarity.

Sounds good