FastLED/FastLED #1535: Can you support Air001?

arilink, 2023-09-05:
I saw on the Internet or main air001 chip (https://official.openluat.com/) Its cost is very low, very suitable for ws2812 control, do not know can support?

samguyer, 2023-10-10:
Hi! I don't know anything about that processor. If it is related to ARM, ESP, or AVR, then it might be possible for you to port the code fairly quickly. Do you know what the instruction set architecture is? What compiler toolchains does it use?

We rely mostly on volunteers to developer support for new boards, although we are always happy to help!

arilink, 2023-10-11:
It is a very low cost MCU from China, using Cortex-M0+ core, it would be a great choice if it can support FastLED, I leave a website about it below, you can check it out, and I am also interested in participating in FastLED support for it (but I don't have much experience, Currently, I only use Arduino for simple development.)
https://wiki.luatos.com/chips/air001/mcu.html#id1

samguyer, 2023-10-12:
We already have support for M0-based chips, so it might be relatively easy. I can't work on it myself, but you might be able to recruit someone on the FastLED Subreddit. Lots of helpful people are in that group.

zackees, 2024-09-20:
Hi there, I'm marking this as "won't fix" because this is a very niche product line and core devs have their hands full with the mainstream features and architecture issues related to driving leds for the majority of our users.

If this becomes a mass market chip then feel free to reopen.

But with that in mind, if you do implement this chip type please do let us know and we will work with you to get your code into mainline.

closed 2024-09-20 by zackees