FastLED/FastLED #1536: "Low memory available, stability problems may occur" Message, when trying to use 8 strips

Belamat, 2023-09-06:
Hi im having an issue when compiling that outputs "Low memory available, stability problems may occur", i've tried to reading the multiple strip array guide, but the same thing happens, im I missing something or there is some limitation in that regard?
thanks!

void setup(){
FastLED.addLeds<WS2811, DATA_PIN1, BRG>(leds1, NUM_LEDS);  // GRB ordering is assumed
  FastLED.addLeds<WS2811, DATA_PIN2, BRG>(leds2, NUM_LEDS);  // GRB ordering is assumed
  
  
  FastLED.addLeds<WS2811, DATA_PIN4, BRG>(leds4, NUM_LEDS);  // GRB ordering is assumed
  
  FastLED.addLeds<WS2811, DATA_PIN5, BRG>(leds5, NUM_LEDS);  // GRB ordering is assumed
  
  FastLED.addLeds<WS2811, DATA_PIN6, BRG>(leds6, NUM_LEDS);  // GRB ordering is assumed
  
  FastLED.addLeds<WS2811, DATA_PIN7, BRG>(leds7, NUM_LEDS);  // GRB ordering is assumed
  
  FastLED.addLeds<WS2811, DATA_PIN8, BRG>(leds8, NUM_LEDS);  // GRB ordering is assumed
}

if i only declare one FastLED object in the setup the code does not show that warning and it works with one strip but when trying with 8 it dont, still have to try some other things but currently not with the device but trying to figure it out so when i get to the place I'll be more ready to tackle the problem.

Thanks!

Belamat, 2023-09-06:
using an arduino nano, btw

alastaira, 2023-09-07:
Well, the Arduino Nano only has 2kB of RAM, which is sufficient to store the RGB values of one array of LEDs, but not enough (or, barely enough) for 8x that amount of LEDs. I'm not quite sure how much more there is else to explain?

Belamat, 2023-09-07:
hi @alastaira, thanks for answering: )

Solved it with this code implementation following a fastLed Multiple Controller Example:
With this approach only 28% of the dynamic memory is being used.

#include "FastLED.h"

#define NUM_LEDS 45
#define NUM_STRIPS 8

CRGB leds[NUM_LEDS];
CLEDController *controllers[NUM_STRIPS];
uint8_t gBrightness = 255;
 
void setup() { 
  //controllers[0] = &FastLED.addLeds<WS2811,13,BRG>(leds, NUM_LEDS);
  controllers[1] = &FastLED.addLeds<WS2811,2,BRG>(leds, NUM_LEDS);
  controllers[2] = &FastLED.addLeds<WS2811,3,BRG>(leds, NUM_LEDS);
  controllers[3] = &FastLED.addLeds<WS2811,5,BRG>(leds, NUM_LEDS);
  controllers[4] = &FastLED.addLeds<WS2811,6,BRG>(leds, NUM_LEDS);
  controllers[5] = &FastLED.addLeds<WS2811,8,BRG>(leds, NUM_LEDS);
  controllers[6] = &FastLED.addLeds<WS2811,9,BRG>(leds, NUM_LEDS);
  controllers[7] = &FastLED.addLeds<WS2811,11,BRG>(leds, NUM_LEDS);
  controllers[0] = &FastLED.addLeds<WS2811,12,BRG>(leds, NUM_LEDS);
 // controllers[8] = &FastLED.addLeds<WS2811_400,12,BRG>(leds, NUM_LEDS);
  FastLED.clear();
  FastLED.show();
}

void loop() { 
//   draw led data for the first strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Red);
  controllers[0]->showLeds(gBrightness);
 //  draw led data for the second strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Green);
  controllers[1]->showLeds(gBrightness);

  // draw led data for the third strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Red);
  controllers[2]->showLeds(gBrightness);

//   draw led data for the first strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Black);
  controllers[3]->showLeds(gBrightness);

  // draw led data for the first strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Blue);
  controllers[4]->showLeds(gBrightness);

  // draw led data for the second strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Black);
  controllers[5]->showLeds(gBrightness);

  // draw led data for the third strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Red);
  controllers[6]->showLeds(gBrightness);

  // draw led data for the first strand into leds
  fill_solid(leds, NUM_LEDS, CRGB::Blue);
  controllers[7]->showLeds(gBrightness);
}

closed 2023-09-07 by Belamat

Bennyggcc, 2025-01-08:
Hi, @Belamat @alastaira, I would like to ask, how can 8 LED strips simultaneously light up? I follow your code and analyze the I/O logic of esp32, shown as the figure. It can be seen that the only 4 strips light up at the same time. 
[image]

sutaburosu, 2025-01-08:
@Bennyggcc this issue discusses reducing memory usage on the Nano. The Nano always sends each pin sequentially.

How many pins you can send in parallel depends on exactly which ESP32 model you have, which versions of ESP32 board support and FastLED are installed, and which peripheral you are using: RMT or I2S. It varies from 1 to 16. For support questions such as yours, https://reddit.com/r/FastLED is a better place. Remember to include the entire part number of your ESP, e.g. ESP32-S3-WROOM-1

Bennyggcc, 2025-01-09:
Thanks for your reply and guidance.