FastLED/FastLED #1537: I’d like to volunteer time to help add RGBW+ support

ssilverman, 2023-09-07:
Greetings. I’d love to help architect RGBW+ into the system. I’m not sure what the status is of Mr. Garcia’s experimental RGBW branch, but I’d like to help integrate it.

This is a re-request of https://github.com/FastLED/FastLED/issues/914 because maybe new issues have more visibility to the maintainers than comments on old issues.

@samguyer @kriegsman would either of you be willing to have a conversation about this? I’d really like to help advance the state of the project. Let’s chat about it by voice, if you’re willing? I believe I have the expertise needed to assist getting this done robustly.

polykoma, 2023-09-10:
Hey Hey - actually you should see and follow up on the SK thread here where active RGBW discussion is taking place right now.
https://github.com/FastLED/FastLED/issues/1279

I actually just right now finished soldering a test setup with 120x  SK6822 RGBW leds and tried to use the last sketch i posted there which i found on the interwebs.
https://gist.github.com/tuxuser/45d06bfc2eb8b92de2f8c6fe0dded28d

Sadly this doesnt do the trick yet. Maybe the code needs adaption. Im sadly no coder but a tinkerer 🗡️

kriegsman, 2023-09-13:
Cross-referencing this here:  I've (finally) started work on the low-level FastLED driver for four-channel (RGBW) WS281x LED strips.  I'll be tracking that work over on this Issue:  https://github.com/FastLED/FastLED/issues/1540

Once the LED driver is up and running on 8-bit AVR, which I'm doing first because it's the least-capable microcontroller we support, we'll need help 'porting' the same approach to ARM (will require a little asm code) and ESP (probably little/no asm?)

closed 2024-12-18 by zackees