FastLED/FastLED #1538: Porting for Arduino UNO R4 please.

yoyojacky, 2023-09-08:
Here is my test code on arduino UNO R4 : 
```bash

#include <FastLED.h>

#define LED_PIN     6
#define NUM_LEDS    18
#define BRIGHTNESS  64
#define LED_TYPE    WS2811
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

// This example shows several ways to set up and use 'palettes' of colors
// with FastLED.
//
// These compact palettes provide an easy way to re-colorize your
// animation on the fly, quickly, easily, and with low overhead.
//
// USING palettes is MUCH simpler in practice than in theory, so first just
// run this sketch, and watch the pretty lights as you then read through
// the code.  Although this sketch has eight (or more) different color schemes,
// the entire sketch compiles down to about 6.5K on AVR.
//
// FastLED provides a few pre-configured color palettes, and makes it
// extremely easy to make up your own color schemes with palettes.
//
// Some notes on the more abstract 'theory and practice' of
// FastLED compact palettes are at the bottom of this file.

CRGBPalette16 currentPalette;
TBlendType    currentBlending;

extern CRGBPalette16 myRedWhiteBluePalette;
extern const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM;

void setup() {
    delay( 3000 ); // power-up safety delay
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );
    
    currentPalette = RainbowColors_p;
    currentBlending = LINEARBLEND;
}

void loop()
{
    ChangePalettePeriodically();
    
    static uint8_t startIndex = 0;
    startIndex = startIndex + 1; /* motion speed */
    
    FillLEDsFromPaletteColors( startIndex);
    
    FastLED.show();
    FastLED.delay(1000 / UPDATES_PER_SECOND);
}

void FillLEDsFromPaletteColors( uint8_t colorIndex)
{
    uint8_t brightness = 255;
    
    for( int i = 0; i < NUM_LEDS; ++i) {
        leds[i] = ColorFromPalette( currentPalette, colorIndex, brightness, currentBlending);
        colorIndex += 3;
    }
}

// There are several different palettes of colors demonstrated here.
//
// FastLED provides several 'preset' palettes: RainbowColors_p, RainbowStripeColors_p,
// OceanColors_p, CloudColors_p, LavaColors_p, ForestColors_p, and PartyColors_p.
//
// Additionally, you can manually define your own color palettes, or you can write
// code that creates color palettes on the fly.  All are shown here.

void ChangePalettePeriodically()
{
    uint8_t secondHand = (millis() / 1000) % 60;
    static uint8_t lastSecond = 99;
    
    if( lastSecond != secondHand) {
        lastSecond = secondHand;
        if( secondHand ==  0)  { currentPalette = RainbowColors_p;         currentBlending = LINEARBLEND; }
        if( secondHand == 10)  { currentPalette = RainbowStripeColors_p;   currentBlending = NOBLEND;  }
        if( secondHand == 15)  { currentPalette = RainbowStripeColors_p;   currentBlending = LINEARBLEND; }
        if( secondHand == 20)  { SetupPurpleAndGreenPalette();             currentBlending = LINEARBLEND; }
        if( secondHand == 25)  { SetupTotallyRandomPalette();              currentBlending = LINEARBLEND; }
        if( secondHand == 30)  { SetupBlackAndWhiteStripedPalette();       currentBlending = NOBLEND; }
        if( secondHand == 35)  { SetupBlackAndWhiteStripedPalette();       currentBlending = LINEARBLEND; }
        if( secondHand == 40)  { currentPalette = CloudColors_p;           currentBlending = LINEARBLEND; }
        if( secondHand == 45)  { currentPalette = PartyColors_p;           currentBlending = LINEARBLEND; }
        if( secondHand == 50)  { currentPalette = myRedWhiteBluePalette_p; currentBlending = NOBLEND;  }
        if( secondHand == 55)  { currentPalette = myRedWhiteBluePalette_p; currentBlending = LINEARBLEND; }
    }
}

// This function fills the palette with totally random colors.
void SetupTotallyRandomPalette()
{
    for( int i = 0; i < 16; ++i) {
        currentPalette[i] = CHSV( random8(), 255, random8());
    }
}

// This function sets up a palette of black and white stripes,
// using code.  Since the palette is effectively an array of
// sixteen CRGB colors, the various fill_* functions can be used
// to set them up.
void SetupBlackAndWhiteStripedPalette()
{
    // 'black out' all 16 palette entries...
    fill_solid( currentPalette, 16, CRGB::Black);
    // and set every fourth one to white.
    currentPalette[0] = CRGB::White;
    currentPalette[4] = CRGB::White;
    currentPalette[8] = CRGB::White;
    currentPalette[12] = CRGB::White;
    
}

// This function sets up a palette of purple and green stripes.
void SetupPurpleAndGreenPalette()
{
    CRGB purple = CHSV( HUE_PURPLE, 255, 255);
    CRGB green  = CHSV( HUE_GREEN, 255, 255);
    CRGB black  = CRGB::Black;
    
    currentPalette = CRGBPalette16(
                                   green,  green,  black,  black,
                                   purple, purple, black,  black,
                                   green,  green,  black,  black,
                                   purple, purple, black,  black );
}

// This example shows how to set up a static color palette
// which is stored in PROGMEM (flash), which is almost always more
// plentiful than RAM.  A static PROGMEM palette like this
// takes up 64 bytes of flash.
const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM =
{
    CRGB::Red,
    CRGB::Gray, // 'white' is too bright compared to red and blue
    CRGB::Blue,
    CRGB::Black,
    
    CRGB::Red,
    CRGB::Gray,
    CRGB::Blue,
    CRGB::Black,
    
    CRGB::Red,
    CRGB::Red,
    CRGB::Gray,
    CRGB::Gray,
    CRGB::Blue,
    CRGB::Blue,
    CRGB::Black,
    CRGB::Black
};

```
when compiling: 
```bash
C:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4/bin/arm-none-eabi-g++ -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 -DPROJECT_NAME="C:\Users\52pidev1\AppData\Local\Temp\arduino\sketches\87A926BAA596FC67FEBA1B4AF18DA433/sketch_sep8a.ino" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb @C:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\variants\MINIMA/defines.txt -DCFG_TUSB_MCU=OPT_MCU_RAXXX -IC:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2/cores/arduino/tinyusb -IC:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/api/deprecated -IC:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/api/deprecated-avr-comp -IC:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino -IC:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\variants\MINIMA -Id:\Arduino\libraries\FastLED\src -iprefixC:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2 @C:\Users\52pidev1\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\variants\MINIMA/includes.txt C:\Users\52pidev1\AppData\Local\Temp\arduino\sketches\87A926BAA596FC67FEBA1B4AF18DA433\sketch\sketch_sep8a.ino.cpp -o C:\Users\52pidev1\AppData\Local\Temp\3842959541\sketch_merged.cpp
In file included from d:\Arduino\libraries\FastLED\src/FastLED.h:51:0,
                 from C:\Users\52pidev1\AppData\Local\Temp\.arduinoIDE-unsaved202388-18568-1900ul8.3clni\sketch_sep8a\sketch_sep8a.ino:1:
d:\Arduino\libraries\FastLED\src/led_sysdefs.h:61:2: error: #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
 #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
  ^~~~~

Using library FastLED at version 3.6.0 in folder: D:\Arduino\libraries\FastLED 
exit status 1

Compilation error: exit status 1
```
Seems like it does not compatible with FastLED library on arduino UNO R4, could you please port for arduino UNO R4?

chw90, 2023-09-24:
Have you tried using the latest master branch that includes [this commit for UNO R4 support by Martino Facchin](https://github.com/FastLED/FastLED/pull/1523/commits/b1f3a6613ea678187c270f4839ed653937924eec)?

For my R4 Wifi and W2812b strip with 144 LEDs, this is working as desired:

```
#include <FastLED.h>

#define NUM_LEDS 144
#define DATA_PIN 6

CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
}

void loop() {
  for ( int i = 0; i < NUM_LEDS; i++ ) {
    leds[i] = CRGB::Red;
    FastLED.show();
    delay(20);
    leds[i] = CRGB::Black;
    FastLED.show();
    delay(1);
  }
}
```

eMUQI, 2023-10-18:
> Have you tried using the latest master branch that includes [this commit for UNO R4 support by Martino Facchin](https://github.com/FastLED/FastLED/pull/1523/commits/b1f3a6613ea678187c270f4839ed653937924eec)?您是否尝试过使用包含此提交的最新主分支来获得 Martino Affin 的 UNO R4 支持？
> 
> For my R4 Wifi and WS2812B strip with 144 LEDs, this is working as desired:对于我的 R4 Wifi 和带有 144 个 LED 的WS2812B条，这是按预期工作的：
> 
> ```
> #include <FastLED.h>
> 
> #define NUM_LEDS 144
> #define DATA_PIN 6
> 
> CRGB leds[NUM_LEDS];
> 
> void setup() {
>   FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
> }
> 
> void loop() {
>   for ( int i = 0; i < NUM_LEDS; i++ ) {
>     leds[i] = CRGB::Red;
>     FastLED.show();
>     delay(20);
>     leds[i] = CRGB::Black;
>     FastLED.show();
>     delay(1);
>   }
> }
> ```

Just for reference, I tested the latest code from the master branch. The code performs as expected on my R4 Wifi and WS2812B strip with 8 LEDs.

Thank you for your reminder @chw90.

eMUQI, 2023-11-15:
Hey FastLED Team,

@kriegsman @jasoncoon @zackees 

Hope you're all doing great! I've been playing around with the Arduino UNO R4 and am really excited about the potential updates for FastLED support. Noticed it's been a bit since we chatted about it. Any idea when the new release might drop? Just curious if there's any tricky bugs or stuff that's holding things up.

Really appreciate all the work you put into this. It's awesome to see how FastLED keeps evolving!

toybuilder, 2023-11-15:
If you download the master branch and install it manually into your Arduino's libraries directory, it will work.  I have an installation with 889 LEDs and it has been working quite well on the Uno R4.

ocack, 2024-01-03:
Hello,

I have a UNO R4 and a WS2811 strip led.
The actual FastLed release doesn't work with this configuration.
I tried without success this ticket : #1554 #1523 

I can help because I have the hardware and I'am not a novice in C Programmation.
I just need renesas and WS2811 specification.

Any help would be appreciated.
Regards

ocack, 2024-01-29:
Can anybody help me for this issue ?
Difficult to beleive that I am the only one that interfaces an Arduino R4 with FastLed and a strip WS2811 !?
I would like to help the community to move forward on that issue.
Which tests can I do ?
Which documentation do I need for trying to fixing that ?
Regards

yoyojacky, 2024-08-22:
I have test it out, but still no luck, so that I have changed the library from fastLED to adafruit_neopixel..... it works for me.

zackees, 2024-08-30:
Builds fine for us:

[image]

zackees, 2024-08-30:
Just built with no errors on platformio. So not sure what the problem was, or what fixed it.

closed 2024-08-30 by zackees