FastLED/FastLED #1540: Add RGBW support to the LED driver code

kriegsman, 2023-09-13:
This is the new main 'tracking' Issue for **RGBW** support in the FastLED LED drivers, which _is_, at long last, getting worked on.

LED driver support for RGBW means:
The minimum functionality that is required to drive a four-channel 8-bit-per-channel LED strip (R,G,B,W), with support for the FastLED-standard master brightness control, color correction, temperature correction, and temporal dithering.

First target will be WS281x, driven by classic AVR microcontrollers (classic Arduino Uno, Leonardo, etc.)  We're doing this one first because it is by far the hardest, and the one that has been holding up the works literally for years.  The existing 3-channel AVR assembly code is exquisitely crafted and tuned -- and we may keep that code exactly unchanged for 3-channel (RGB) strips.  Adding a fourth channel (W) will cost us some clock cycles, but at this point it's worth it.  Better to have a four-channel LED driver that works, which we can optimize later, than to have nothing working at all.

Once we have the AVR version up and working well enough, we'll do ARM and ESP drivers that follow the approach used in the AVR LED driver.  Whatever works on the least-capable microcontrollers (8-bit AVR) will presumably be not too difficult to implement on newer 32-bit MCUs.

Things that are _not_ covered as part of this tracking Issue (which, again, is just for the LED driver code) :
- more or less than 8 bits of data per channel 
- more than four channels e.g. R G B W UV... or whatever
- HSV/RGB/RGBW color converters
- new RGBW-savvy color correction
- adding RGBW support to all the library functions that currently support RGB, e.g. Palettes, fill_ functions, etc.  All of that is needed, and all of that will come, hopefully with lots of help from lots of contributors.  But this tracking Issue is just about the LED driver code.

kriegsman, 2023-09-13:
Just so folks know, I've spent many hours over the last couple of weeks finally starting to unpack the existing 2013-vintage AVR assembly code for driving three-channel (RGB) WS281x LEDs, finding bugs(!) in it, deconstructing it, and starting to rebuild it so that it can drive four-channel (RGBW) LEDs.

I have some code up and running at the very very most basic level, but it is outputting four channels of data per pixel from an Arduino Uno.  Once it's even slightly presentable, I'll check in a new branch and other folks can start to help test it.

@focalintent (Dan) wrote some amazing AVR assembly code back in 2013 to drive RGB strips even on an AVR ATtiny at 8MHz, and ever since then we'd sort of been dreading having to tear it out and re-write it all to handle RGBW.  In the spring of 2019, Dan and I casually started work on it a little bit, but when I lost Dan in the fall of 2019, it completely took the wind out of my sails for doing this work.   Now, four difficult years later, I've decided that I'm going to take a crack at it.  I'm feeling cautiously optimistic, and with some good reason: I have a hacked apart/together version of the code lighting up some LEDs on my lab bench right next to me.  On the one hand, Dan would roll his eyes at how I've decided to do this, but on the other hand, he'd probably laugh, too, and say "Well, if it works, let's just ship it and see what happens!"

So.  Progress is finally happening.  And the AVR assembly code still needs all kinds of TLC.  But we're getting going again now for real.

kuriousDesign, 2023-10-09:
Hey Kriegsman, 

> Just so folks know, I've spent many hours over the last couple of weeks finally starting to unpack the existing 2013-vintage AVR assembly code for driving three-channel (RGB) WS281x LEDs, finding [...]

Is there anything available to test now? I'm working with SK6812 RGBW right now, i could use either an esp32 or an Uno or Mega.

zackees, 2023-10-09:
We now have three people who want to re-write the code to handle 4-channel WS2811 code.

@kriegsman @kuriousDesign @ssilverman

zackees, 2023-10-15:
Got an interesting message on reddit about RGB -> RGBW conversion and some algorithms. Bad news is it's all in floating point. So probably a non starter:

```
If you are familliar with GLSL syntax, take a look at this (I know I'm late to the party):

https://www.shadertoy.com/view/NsjBzt

There are two different algorithms creating theorically perfect RGBW matches (if there is any) for any unbound RGB input.

The first algorithm (RGBW_max) tries to maximize the value of the white component (better CRI), and the other (RGBW_min) minimises it. RGBW_min doesn't do anything if your color can be represented with normalized RGB.

Both these algorithm will try to at least match the percieved luminance of the target color if it is impossible to perfectly reproduce the full color only with the available primaries.

You can linearly interpolate between the results of these two algorithm to get the full range of all possible RGBW quadruplets that correspond to the desired RGB color.

Both of these algorithm works under these assumptions:

the relative luminance of each primary is known (lum)
input colors are expressed in the linear color space associated with your 3 colored primary light sources (hopefully some kind of RGB), normalized from 0 to 1 (0=no emission, 1 = full brightness)
the color of the W primary is also expressed in that space, and it can have values exceeding 1 depending on its actual brightness relative to the other primaries.
the chromaticity of the W primary is contaned inside that color space's gamut (no negative components)
```

kuriousDesign, 2023-10-16:
I think at a minimum, I'm looking for an algorithm which will allow me to
apply the NighDriverLedStrip color affects to RGBW strips without
necessarily controlling the W values of those strips. When I performed some
testing a few weeks ago I was able to control the rgb but I had to
oscillate the order or RGB for each successive led. Is there already a
chipset that I can use which handles this problem?

zackees, 2023-10-18:
Note, after further discussion it was discovered the above algorithm requires 9-bit component lighting.

JoshHiggins92, 2024-01-30:
Hope you're doing well kriegsman. I'm happy and willing to test anything you may have made any progress with!
I've got a Arduino Nano Every ATmega328P @20mhz, I'm sure it can handle the additional load if you're looking to optimize later.

kuriousDesign, 2024-01-31:
I found a work around but I'm limited in the number of leds.?

JoshHiggins92, 2024-02-02:
What's seems to be the limitation?

kuriousDesign, 2024-02-05:
has there been recent work on this issue that can be tested?

LunaCodeGirl, 2024-02-16:
Is there a working branch or PR for this? Can try to help some as well. I'm assuming there are some common data types that first need to be updated and then the work to update each of the different chipsets could be done in parallel. I'm working with an nRF52840, would be willing to give it a try updating that chipset's code. lmk.

zackees, 2024-02-16:
From some other comments that I've seen, you may be able to just overallocate a CRGB strip.  So if you have 12 lights that's going to be 12 + 1/3*12 = 12 + 4 = 16 CRGB pixels. Every 4th pixel will be white or one of the other components.

If this works then make sure your strip you are addressing is a multiple of three.

abc123me, 2024-04-07:
Wanted to throw my name in the bin, I have a WS2814 strip and an Arduino to go with, I wrote my own library but it sucks, would love to help test and develop the RGBW support anyway that I can!

dogofpavlov, 2024-06-27:
Is there a working Branch for this?

Le0ganess, 2024-07-14:
Hello, I've created a small library to add RGBW, I'm still a beginner in all things programming but I hope it can help you program your LEDs. You just need to add the fastLED library and the library I've put here: [](https://github.com/Le0ganess/RBGW-with-fastLED)

closed 2024-09-18 by zackees

directionless, 2025-11-01:
This was closed, but AFAICT FastLED only supports RGBW on esp32 bit boards.

zackees, 2025-11-02:
All other platforms support it via [RGBWEmulated.ino](https://github.com/FastLED/FastLED/blob/master/examples/RGBWEmulated/RGBWEmulated.ino)

directionless, 2025-11-02:
For white, should [`RGBWEmulated.ino`](https://github.com/FastLED/FastLED/blob/master/examples/RGBWEmulated/RGBWEmulated.ino) use the white LED, or the RGB? Squinting, it looks like it's using the RGB. Which makes me think the emulated support is not using the white, just sending an empty value as padding. 

Is that correct? 

If so, I think this issue should remain open. I, and I suspect the larger community, would love to see real RGBW support in fastled.

zackees, 2025-11-03:
RGBWEmulated does RGBW spoofing, sending RGBW data reinterpreted casted as a higher count RGB data strip + padding down the wire.

Trust me it works.