FastLED/FastLED #1543: UNO R4 WiFi NeoPixel interrupts(?) cause artifacts on a strip

Tai-Min, 2023-09-27:
Running this code causes random color / flickering on the whole strip:
```cpp
#include <Arduino.h>
#include <FastLED.h>

namespace
{
  CRGB leds1[49];
  CLEDController *ctrl1 = &FastLED.addLeds<NEOPIXEL, 7>(leds1, 49);
}

void setup()
{
 Serial.begin(9600);

}

uint8_t bright = 0;
void loop()
{
  fill_solid(leds1, 49, CRGB::Green);
  //noInterrupts();
  ctrl1->showLeds(bright);
  //interrupts();
  bright++;
  delay(10);
}
```
From my tests it seems that using any peripheral based on interrupts such as Serial or I2C (I've tested it with LiquidCrystal_I2C library) causes the strip to go wild randomly. Uncommenting noInterrupts(); and interrupts() above seems to fix the issue.

I've tested it using PlatformIO

samguyer, 2023-10-10:
Hello! The latest main branch has support for Uno R4. But it's not in a regular release yet.

sparky650, 2023-10-24:
I had a similar issue, and was able to fix it by adjusting the bit timing in clockless_arm_renesas.h. I tested your code and was able to replicate the flicker with the old timing. The new timing gave flicker free output with and without interrupts. Here is the PR: https://github.com/FastLED/FastLED/pull/1554

closed 2024-09-20 by zackees