FastLED/FastLED #1545: APA102 / Dotstar Gamma Bit shifting gamma correction formula

zackees, 2023-09-28:
# High Bit Depth Algorithm for APA102/Dotstar LEDs

This document outlines an algorithm to achieve higher bit depth on the APA102 chipset (and possibly its sks clone), leveraging the global brightness setting unique to each LED. The primary exploit revolves around manipulating the 5-bit global brightness setting alongside bit-shifting operations on the R,G,B color components to enhance the relative resolution, particularly at the lower end where it's crucial.

This algorithm is fast, using only fast integer math like bitshift and comparisons.

## Introduction

The APA102 LEDs come with a per-LED global brightness setting, expressed in 5 bits. A fascinating characteristic is observed where a shift in the global brightness proportionally alters the brightness of each color component by doubling it.

Illustratively, consider the format RGBV(8,8,8,5), where V represents the Global brightness setting, and the numbers denote bit-depth. It's observed that on shifting the Global brightness, each of the color components shift equivalently:

```
RGBV(R << 1, G << 1, B << 1, V) == RGBV(R, G, B, V << 1)
```

For example

```
RGBV(2, 2, 2, 2) == RGBV(4, 4, 4, 1)
```

## Algorithm

Follow the steps outlined below:

1. **Gamma Correction**:
   - Transform RGB(8,8,8) to RGB(16, 16, 16) through gamma correction.
   
2. **Initialization**:
   - Set V8 = 0xff >> 3.
   
3. **Bit Shifting Loop**:
   - While conditions `(V8 >> 1) > 1` and `(max32(R16,G16,B16) << 2) < int32(0xffff)` are met:
     - Execute `R16 = R16 << 2`
     - Execute `B16 = B16 << 2`
     - Execute `G16 = G16 << 2`
     - Execute `V8 = V8 >> 1`
   
4. **Conversion Back**:
   - Convert RGB16 + V8 back to RGB + V5:
     - `R8 = R16 / 255`
     - `G8 = G16 / 255`
     - `B8 = B16 / 255`
   
5. **Output**:
   - Output as RGBV(R8, G8, B8, 5bits(V8))

## Notes
- Initial color components are in 8-bits (r, g, b).
- Brightness is represented in 5-bits.
- The conversion from 8-bit to 16-bit is illustrated as follows:
   - RGB(0xFF, 0xFF, 0xFF) -> Gamma Correction -> RGB16(0xFFFF, 0xFFFF, 0xFFFF)

## Future Exploration

Exploration into a multiplication-based method that scales the 8-bit color values while adjusting the 5-bit global brightness accordingly is deemed worthwhile.

zackees, 2023-09-28:
Here is a chat-gpt implementation in C++

```C++
#include <cstdint>
#include <algorithm>

// Assuming gamma correction function is defined as follows:
void gammaCorrect(uint8_t r8, uint8_t g8, uint8_t b8, uint16_t& r16, uint16_t& g16, uint16_t& b16) {
    r16 = uint16(r8) * r8; // Placeholder
    g16 = uint16(g8) * g8; // Placeholder
    b16 = uint16(b8) * b8; // Placeholder
}

struct RGBV {
    uint8_t r;
    uint8_t g;
    uint8_t b;
    uint8_t v; // 5-bit value
};

RGBV highBitDepthCorrection(uint8_t r8, uint8_t g8, uint8_t b8) {
    // Step 1: Gamma Correction
    uint16_t r16, g16, b16;
    gammaCorrect(r8, g8, b8, r16, g16, b16);

    // Step 2: Initialization
    uint8_t v8 = 0xff >> 3;

    // Step 3: Bit Shifting Loop
    while ((v8 >> 1) > 1 && (std::max({r16, g16, b16}) << 2) < 0xffff) {
        r16 = r16 << 2;
        g16 = g16 << 2;
        b16 = b16 << 2;
        v8 = v8 >> 1;
    }

    // Step 4: Conversion Back
    uint8_t r8_final = r16 / 255;  // or bitshift >> 8
    uint8_t g8_final = g16 / 255;
    uint8_t b8_final = b16 / 255;

    // Step 5: Output
    RGBV output;
    output.r = r8_final;
    output.g = g8_final;
    output.b = b8_final;
    output.v = v8;  // Ensure this value is within 5 bits

    return output;
}

int main() {
    uint8_t r8 = 0xff;
    uint8_t g8 = 0xff;
    uint8_t b8 = 0xff;

    RGBV correctedValues = highBitDepthCorrection(r8, g8, b8);

    // Output the corrected values
    // ...

    return 0;
}
```

Joshfindit, 2024-01-23:
Apologies if I'm out of line, I just ran across this issues during research and noticed it's resolved by the linked (merged) PR.

zackees, 2024-01-25:
Yeah I had to shift scaling post gamma instead of pre gamma.