FastLED/FastLED #1550: Hardware SPI Correct Pin On Mega2560

Mac898, 2023-10-11:
Hi,

I was just reading the wiki page on [SPI Hardware Or Bit banging](https://github.com/FastLED/FastLED/wiki/SPI-Hardware-or-Bit-banging).

It says pin 53 is the clock on ATMega2560 based boards. I presume this is the Arduino pinout number.
Except if we look at the Arduino Mega 2560 board the SCK pin is pin 52 with 53 being chip select.
[image]

I am designing a PCB with this in mind and just wanted to verify that pin 52 is the correct pin and update the wiki if this is correct.

Kind Regards,
Mac

zackees, 2024-09-20:
Please update our wiki. We will accept the PR right away.

Mac898, 2024-09-21:
I dont think it is possible to make a pull request to update a wiki page?