FastLED/FastLED #1551: APA102: black shows blue

GvBon, 2023-10-12:
All my small (5mm width) APA102 ledstrips show blue when they are supposted to be black for the leds that are initialised. So in below example 10 leds are blue when they should be black. The rest of the strip really is black. The 10 leds do blink so they do receive the data they are suppost to. When I connect a 'normal' and bigger ledstrip APA102 everything works fine. Also when I use another library ( https://github.com/pololu/apa102-arduino) the smaller ledstrip also works fine. 
For my code I need it to be fastled as I am using a lot of the fantastic functions fastled offers. A combination of both APA102 libraries is not possible I found out.
How can I resolve this? 

`#include <FastLED.h>
#define NUM_LEDS 10
#define DATA_PIN 1
#define CLOCK_PIN 0
CRGB leds[NUM_LEDS];
void setup() {
  FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, GBR>(leds, NUM_LEDS);  // BGR ordering is typical
}
void loop() {
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB::Red;
  }
  FastLED.show();
  delay(500);
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB::Black;
  }
  FastLED.show();
  delay(500);
}
`

zackees, 2024-09-19:
Are you using the blue pill board? Others complained of this exact same problem. If so then try the latest master branch and see if that fixes your problem (assuming you still have a board around).

zackees, 2024-09-20:
This issues matches the blue pill / maple mini problems which are fixed in next weeks 3.7.8 release.

Since the reporter didn't specify their chipset, and has not responded in two days, I'm going to assume that this is a blue pill / maple mini and the issue has been resolved. If this is not the case then feel free to reopen this issue.

Cheers.

closed 2024-09-20 by zackees