FastLED/FastLED #1552: Please implement SPI hardware for ESP32-S3 (Feature Request)

alba-ado, 2023-10-13:
Hello! Thank you for this library.

While reviewing your library about the issue #1517, I found out that the ESP32-S3 SPI implementation is not present. As a result, it defaults to bit banged output as it prints a warning in the console. However, I changed the SPIClass as the Arduino SPI class itself in the fastspi_esp32.h file, the program works successfully (although does not resolve the mentioned issue). This works because any pin on the ESP32-S3 can act as a hardware SPI unlike the classic ESP32. So there is no VSPI or HSPI hardware SPI classes present. Instead, using the default SPI class seems to work.

But I don't know how can I implement this properly in the library, so I didn't send a pull request. (I may send it in the future if I can understand all the related macros about hardware SPI) For this reason, I am asking you to add hardware SPI support for the S3 module. Changing the fastspi_esp32.h may not be enough to properly support it for the hardware, so I would be happy if you can implement these changes.

Best Regards

SCarter858, 2023-11-23:
I think the problem is not that the -S3 is not supported, but that the whole ESP32 hardware SPI support is implemented with a sort of fatal flaw.

Perhaps you didn't put "#define FASTLED_ALL_PINS_HARDWARE_SPI" before you #included the FastLED library?  It's not well documented.  On my ESP32-S3 with FastLED version 3.6.0, the warning message then turns to a message that HW SPI is now being used.  BUT...  if you look at your SPI clock pin you'll see that it bursts out only one byte at a time, with a uSec delay between each byte.  Most people are probably driving a strip, so for 128 LEDs, the FastLED.show call should burst out about 4k bits.  This "HW support" may use a 40MHz clock, but the overhead of setting up a new DMA for each byte rather than one for the full transfer kills the effective throughput...  It's faster than bit-banging, but not by much...

I've been trying to dig through the code, but don't understand FastLED well enough to see if FastLED.show() is fundamentally a per-byte operation or if this is an easy fix to the code...

alba-ado, 2023-11-24:
> I think the problem is not that the -S3 is not supported, but that the whole ESP32 hardware SPI support is implemented with a sort of fatal flaw.
>
> Perhaps you didn't put "#define [...]

My problem was caused by the use of VSPI and HSPI classes inside of the library. ESP-IDF for ESP32-S3 doesn't have a VSPI or HSPI class. This was due to the old ESP32 having fixed SPI ports (Hence the name VSPI and HSPI) (although changeable with some drawbacks) but the new ESP32-S3 doesn't have any such port. So it just uses the default SPI class. As I stated, changing the class fixes the error, but I didn't create a pull request because I don't know what else needs to be changed. ESP32-S3 module needs an implementation seperate from the ESP32 module for this reason in the library. 

But it can be fixed by just changing some macros inside of the fastspi_esp32.h file:

```
#ifndef FASTLED_ESP32_SPI_BUS
    #define FASTLED_ESP32_SPI_BUS VSPI
#endif

static SPIClass ledSPI(FASTLED_ESP32_SPI_BUS);

#if FASTLED_ESP32_SPI_BUS == VSPI
    static uint8_t spiClk = 18;
    static uint8_t spiMiso = 19;
    static uint8_t spiMosi = 23;
    static uint8_t spiCs = 5;
#elif FASTLED_ESP32_SPI_BUS == HSPI
    static uint8_t spiClk = 14;
    static uint8_t spiMiso = 12;
    static uint8_t spiMosi = 13;
    static uint8_t spiCs = 15;
#elif FASTLED_ESP32_SPI_BUS == FSPI  // ESP32S2 can re-route to arbitrary pins
    #define spiMosi DATA_PIN
    #define spiClk CLOCK_PIN
    #define spiMiso -1
    #define spiCs -1
#endif

```

SCarter858, 2023-11-24:
I agree that the ESP32-S3 changed the pinout and nomenclature of the SPI ports, and that the library should be updated to fix that.  

But what was your problem that "can be fixed by changing macros"?  Was it just the compiler defaulting to bit-banged output vs. giving the message "Hardware SPI Support Added"?  If so, yes, that can be accomplished with the right combination of #define's and such.  And it does indeed invoke the hardware SPI driver.

But I believe it then feeds that SPI hardware driver (aka sets up a DMA transfer) for each byte individually.  The overhead of setting up a new DMA transfer for each byte is huge, and limits the SPI throughput to something like 5MHz.  Even using random GPIO pins rather than the optimal ones these macros aim to select should allow  40MHz transfers...

Are you able to get these higher data rates the hardware SPI should allow?

alba-ado, 2023-11-27:
I'm not at my work computer at the moment, but from what I remember it defaulted to bitbanged output and when I forced the hardware SPI it failed to compile saying that VSPI is not found or such. So I edited the VSPI to be SPI and it worked that way. When I have time I will investigate this further.

gunicsba, 2023-12-25:
> Perhaps you didn't put "#define FASTLED_ALL_PINS_HARDWARE_SPI" before you #included the FastLED library? 

This is what you get when you try it:

```
In file included from c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:6,
                 from c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/platforms.h:42,
                 from c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/FastLED.h:62,
                 from D:\AgOpenGPS\AOG_RC\Modules\ESP32 Rate\RC_ESP32\RC_ESP32.ino:33:
c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:2:17: note: #pragma message: ESP32 Hardware SPI support added
 #pragma message "ESP32 Hardware SPI support added"
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:51:35: error: 'VSPI' was not declared in this scope
     #define FASTLED_ESP32_SPI_BUS VSPI
                                   ^~~~
c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:54:24: note: in expansion of macro 'FASTLED_ESP32_SPI_BUS'
 static SPIClass ledSPI(FASTLED_ESP32_SPI_BUS);
                        ^~~~~~~~~~~~~~~~~~~~~
c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:51:35: note: suggested alternative: 'SPI'
     #define FASTLED_ESP32_SPI_BUS VSPI
                                   ^~~~
c:\Users\Balazs\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:54:24: note: in expansion of macro 'FASTLED_ESP32_SPI_BUS'
 static SPIClass ledSPI(FASTLED_ESP32_SPI_BUS);
                        ^~~~~~~~~~~~~~~~~~~~~

exit status 1

```

SCarter858, 2023-12-27:
I don’t think my description of the library problem was clear enough.  

Yes, putting this #define in the code will cause FASTLED to report that “ESP32 HW SPI support was added”.  And it does appear to now use the HW SPI registers to send data.  But if you put a scope on the SPI clock, you’ll see that it bursts out only 8 bits at a time, followed by a long setup for the next 8 bits.  The ESP32 HW SPI support uses DMA, and it looks like this “HW support” is doing a separate DMA setup for each byte (the Espressif documentation says this setup takes about 1uS, which is what I see on the scope).  That’s faster than bit-banging, but not by much.  

A “typical” 1m LED strip needs something like 4k bits to refresh when FastLED.show(); is called. If properly implemented, the HW should burst this out with a single DMA transfer at up to an 80MHz clock (faster than most LEDs can accept).  This
implementation limits things to about 5MHz (effective rate), and takes a bunch of CPU power to keep refreshing the HW for each byte.

alba-ado, 2024-01-08:
> I don’t think my description of the library problem was clear enough. Yes, putting this #define in the code will cause FASTLED to report that “ESP32 HW SPI support was added”. And it does appear to [...]

The problem you are talking about seperate from the one I reported because we can't get it to compile since the VSPI is not defined for the S3 boards.

coryking, 2024-03-10:
@SCarter858 I think your thing should be a seperate issue. I have a patch for this particular issue on my own fork. I'd be willing to do a pull request on it if people think it looks good.
https://github.com/coryking/FastLED/tree/esp32s3-fix

Now as to @SCarter858 , I see the same thing when I put this stuff under my oscilloscope as well. There is 8 bits transferred, then a very long delay, then 8 more bits.... this adds up to a substantial amount of time.

If you can't get to it, I can capture it on my oscilloscope and file a ticket.  It is a pretty big issue for me because I need to drive a string of 30 LEDs super hard.

coryking, 2024-03-10:
Created a pull request (#1608) that "works for me" on my various ESP32S3 projects.

Given the S3 seems to be quite flexible with pin assignments, I wonder if it makes sense to just default to hardware SPI if using an S3?

coryking, 2024-03-11:
And to put a pin in things, I created Issue #1609 for tracking the issues brought up by @SCarter858. Doing some research on that right now...

samguyer, 2024-03-12:
@coryking Thank you -- I added some comments to your PR, but I'd like to get this feature added soon!

coryking, 2024-03-15:
Hey @samguyer, sorry about that code reformat!  I've closed that pull request and opened another one against the latest version of master.... see #1610

zackees, 2024-09-04:
Since this appears to have been added in #1610 and #1615 i'm going to go ahead and close this feature request.

closed 2024-09-04 by zackees

zackees, 2024-11-14:
Something noteworthy here is that I added a bulk transfer method for ESP32 FastLED spi implementation starting with FastLED 3.9.3.

You can enable it like this.

```C++
#define FASTLED_ESP32_SPI_BULK_TRANSFER 1
#include "FastLED.h"
```