FastLED/FastLED #1553: Blink Arduino sample fails to compile on ESP32S3

sukesh-ak, 2023-10-21:
Sample used without any changes:
https://github.com/FastLED/FastLED/tree/master/examples/Blink

My custom sample which was working earlier, is not compiling anymore. So tested with the Blink sample and same issues.
You don't need hardware to test it. Set the MCU to S3 device and try compiling.

Compiler Error output. 
```
Compiling libraries...
In file included from c:\Users\sukes\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from C:\Users\sukes\AppData\Local\Temp\.arduinoIDE-unsaved2023921-15596-mlo18f.socmm\Blink\Blink.ino:5:
c:\Users\sukes\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling library "FastLED"
"C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_TINYS3 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"TINYS3\"" "-DARDUINO_VARIANT=\"um_tinys3\"" -DARDUINO_PARTITION_default_8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:tinys3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,PartitionScheme=default_8MB,DebugLevel=none,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/defines" "-IC:\\Users\\sukes\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023921-15596-mlo18f.socmm\\Blink" -iprefix "C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/include/" "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/includes" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/qio_qspi/include" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\cores\\esp32" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\variants\\um_tinys3" "-Ic:\\Users\\sukes\\Documents\\Arduino\\libraries\\FastLED\\src" "@C:\\Users\\sukes\\AppData\\Local\\Temp\\arduino\\sketches\\0BC9D72F78793541A3C87EDF8B9A09E7/build_opt.h" "c:\\Users\\sukes\\Documents\\Arduino\\libraries\\FastLED\\src\\colorutils.cpp" -o "C:\\Users\\sukes\\AppData\\Local\\Temp\\arduino\\sketches\\0BC9D72F78793541A3C87EDF8B9A09E7\\libraries\\FastLED\\colorutils.cpp.o"
"C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_TINYS3 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"TINYS3\"" "-DARDUINO_VARIANT=\"um_tinys3\"" -DARDUINO_PARTITION_default_8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:tinys3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,PartitionScheme=default_8MB,DebugLevel=none,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/defines" "-IC:\\Users\\sukes\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023921-15596-mlo18f.socmm\\Blink" -iprefix "C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/include/" "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/includes" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/qio_qspi/include" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\cores\\esp32" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\variants\\um_tinys3" "-Ic:\\Users\\sukes\\Documents\\Arduino\\libraries\\FastLED\\src" "@C:\\Users\\sukes\\AppData\\Local\\Temp\\arduino\\sketches\\0BC9D72F78793541A3C87EDF8B9A09E7/build_opt.h" "c:\\Users\\sukes\\Documents\\Arduino\\libraries\\FastLED\\src\\bitswap.cpp" -o "C:\\Users\\sukes\\AppData\\Local\\Temp\\arduino\\sketches\\0BC9D72F78793541A3C87EDF8B9A09E7\\libraries\\FastLED\\bitswap.cpp.o"
"C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_TINYS3 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"TINYS3\"" "-DARDUINO_VARIANT=\"um_tinys3\"" -DARDUINO_PARTITION_default_8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:tinys3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,PartitionScheme=default_8MB,DebugLevel=none,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/defines" "-IC:\\Users\\sukes\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023921-15596-mlo18f.socmm\\Blink" -iprefix "C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/include/" "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/includes" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/qio_qspi/include" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\cores\\esp32" "-IC:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha2\\variants\\um_tinys3" "-Ic:\\Users\\sukes\\Documents\\Arduino\\libraries\\FastLED\\src" "@C:\\Users\\sukes\\AppData\\Local\\Temp\\arduino\\sketches\\0BC9D72F78793541A3C87EDF8B9A09E7/build_opt.h" "c:\\Users\\sukes\\Documents\\Arduino\\libraries\\FastLED\\src\\FastLED.cpp" -o "C:\\Users\\sukes\\AppData\\Local\\Temp\\arduino\\sketches\\0BC9D72F78793541A3C87EDF8B9A09E7\\libraries\\FastLED\\FastLED.cpp.o"
"C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\sukes\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-6b1f40b9bf/esp32s3/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_TINYS3 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"TINYS3\"" "-DARDUINO_VARIANT=\"um_tinys3\"" -DARDUINO_PARTITION_default_8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:tinys3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,PartitionScheme=default_8MB,DebugLevel=none,PSRAM=enabled,LoopCore=1,Ev [... cut ...]
  373 |     RMT.chnconf0[channel].apb_mem_rst_n = 1;
      |                           ^~~~~~~~~~~~~
c:\Users\sukes\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:374:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  374 |     RMT.chnconf0[channel].apb_mem_rst_n = 0;
      |                           ^~~~~~~~~~~~~

Using library FastLED at version 3.6.0 in folder: C:\Users\sukes\Documents\Arduino\libraries\FastLED 
exit status 1

Compilation error: exit status 1

```

jeffcrouse, 2023-10-23:
Pardon if I should have started a new issue, but I think my issue is related. I'm trying to run the Blink example on a [Seeed Studio XIAO ESP32C3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html) and I get similar errors
```
c:\Users\jeff\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::startOnChannel(int)':
c:\Users\jeff\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:266:29: error: 'RMTMEM' was not declared in this scope
  266 |         mRMT_mem_start = & (RMTMEM.chan[mRMT_channel].data32[0].val);
      |                             ^~~~~~
c:\Users\jeff\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
c:\Users\jeff\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:347:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  347 |     gpio_matrix_out(pController->mPin, 0x100, 0, 0);
      |     ^~~~~~~~~~~~~~~
      |     gpio_iomux_out
```
I have FastLED 3.6.0. I just updated my esp32 board definitions to 3.0.0-alpha2 and have selected 'XIAO_ESP32C3' as my board, per the instructions at https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/

munkerench, 2023-10-24:
None of the samples compile after 3.0.0-alpha2

samguyer, 2023-11-05:
OK, I think there are two separate issues here. One is regarding the q44 template. The other is regarding ESP32C3 support. I might create separate master issues for these two problems and link your specific examples.

sukesh-ak, 2023-11-06:
This issue was with dev version of Arduino 3 (IDF 5.1). 
I went back to released version to make it work. 

So currently its not an issue for me but maybe there is some breaking change in ESP32 Arduino v3.

samguyer, 2023-11-06:
Yeah, I have a couple of reports now of compilation failing on the latest version of ESP-IDF. I'll take a look and see what we need to do. Thanks!

P-R-O-C-H-Y, 2023-11-07:
@samguyer You can take a look to the [Migration guide](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html) to have quick overview what have been changed. Also API in docs + examples have been updated too.

rodolfoacostacastro, 2024-01-07:
Experimenting similar issues here with the Arduino IDE v.2.2.1 :(

ricardok-sp, 2024-02-17:
I understand this is the same issue reported in #1555

ricardok-sp, 2024-03-04:
@ylnest2018 When I downgrade the board to ESP32 2.0.11, I can compile and everything works fine. Apparently, it is an issue with Espressif's ESP32 v3.0.0 on Arduino IDE.

ylnest2018, 2024-03-04:
are you able to tell me how that is done or link me to something?
I feel like i mightve upgraded my board (as the phrase seems familiar) to have another function work that didnt, but I feel that was like a 3.5 or something i cant quite recall

ricardok-sp, 2024-03-05:
No, no... maybe I explained it in a wrong way here, sorry. 
I mean: the board driver, in the Arduino IDE board manager (Tools -> Board -> Board Manager). 
When Espressif's esp32 v. 2.0.11 is installed, the code compiles well. When the latest version is installed (currently 3.0.0-alpha3), the problem shown by the OP still happens.

ylnest2018, 2024-03-05:
> No, no... maybe I explained it in a wrong way here, sorry. I mean: the board driver, in the Arduino IDE board manager (Tools -> Board -> Board Manager). When Espressif's esp32 v. 2.0.11 is [...]

ah i see. so is there a way to revert to the older working version?

ricardok-sp, 2024-03-05:
That's it!
I understand v3.0.0 is still in alpha3 because of some uncompatibilities such as this one.

ylnest2018, 2024-03-05:
noooooooooiiiccccceee it worked 🤣

sorry! im still learning my way around everything

PR #1619 (2024-05-27): Fix ambiguous overload compile errors
Partially fixes #1553, #1555 , #1605 

This only fixes the issues in noise.cpp. There are other errors when building for esp32 with the new 3.0 core mostly related to upgrading from espidf 4.x to 5.x which has a new driver model. The RMT and I2S code needs to be updated, but that's a larger change for another day :)
closed 2024-05-27 by kriegsman