FastLED/FastLED #1555: Library not working with ESP Alpha 3.0

munkerench, 2023-10-24:
Since the esp32 alpha 3 release the library does not build.

FastLED\src\noise.cpp: In function 'void fill_raw_2dnoise8(uint8_t*, int, int, uint8_t, q44, fract8, int, uint16_t, int, uint16_t, int, uint16_t)':
\Arduino\libraries\FastLED\src\noise.cpp:623:100: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')
  623 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);

samguyer, 2023-11-05:
Sounds more like a compiler problem. What are you using to build the code? And what version?

samguyer, 2023-11-06:
Following up: I think we know what the issue is, and we'll work on a fix. Thanks!

munkerench, 2023-11-06:
Great!

bas2807, 2024-01-03:
Any update on this? Since I need the ADC continuous mode driver, I cannot revert back to a previous version of arduino-esp32. 
I only use CHSV and fill_solid functions. Could I make adjustments to my local library copy to temporarily fix it?

Many thanks for your great contibutions!

kidreck, 2024-01-10:
> Following up: I think we know what the issue is, and we'll work on a fix. Thanks!

Also on ESP32 Alpha3 and it does seem there is an issue with the compatibility/types of the operands

samguyer, 2024-02-01:
@berkin88 Have you tried using the current head of the repo?

ricardok-sp, 2024-02-17:
@samguyer, I have got the same issue running on Blink example using a Lolin S2 Mini (ESP32-S2) with Arduino IDE. Both IDE and Fastled updated. The error:

> h:\My Drive\Producoes\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:2:17: note: '#pragma message: ESP32 Hardware SPI support added'
    2 | #pragma message "ESP32 Hardware SPI support added"
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
h:\My Drive\Producoes\Arduino\libraries\FastLED\src\noise.cpp: In function 'void fill_raw_2dnoise8(uint8_t*, int, int, uint8_t, q44, fract8, int, uint16_t, int, uint16_t, int, uint16_t)':
h:\My Drive\Producoes\Arduino\libraries\FastLED\src\noise.cpp:623:86: error: invalid cast from type 'q44' {aka 'q<unsigned char, 4, 4>'} to type 'int'
  623 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x* (int)freq44, freq44 * scalex, y*(int)freq44, freq44 * scaley, time);
      |                                                                                      ^~~~~~~~~~~
h:\My Drive\Producoes\Arduino\libraries\FastLED\src\noise.cpp:623:106: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')
  623 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x* (int)freq44, freq44 * scalex, y*(int)freq44, freq44 * scaley, time);
      |                                                                                                   ~~~~~~ ^ ~~~~~~
      |                                                                                                   |        |
      |                                                                                                   |        int
      |                                                                                                   q44 {aka q<unsigned char, 4, 4>}

LeCyberDucky, 2024-03-03:
> @berkin88 Have you tried using the current head of the repo?

Simply adding master to my PlatformIO project makes it fail to build. 

I'm using: 

 - framework-arduinoespressif32 @ 3.0.0+sha.cceebb5
 - framework-arduinoespressif32-libs @ 5.1.0+sha.a5e3e0b90e.sha.6564ec9
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208

I run into errors such as these: 

`.pio/libdeps/lilygo-t-display-s3/FastLED/src/noise.cpp:628:100: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')`

`.pio/libdeps/lilygo-t-display-s3/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:266:29: error: 'RMTMEM' was not declared in this scope`

`.pio/libdeps/lilygo-t-display-s3/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:307:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'`

Is there any workaround for this at this point?

planevina, 2024-03-26:
I have the same problem

tueddy, 2024-04-26:
Is there any news on this? Meanwhile Arduino 3.0 RC1 has been released and the final release should be out in a few weeks.
Unfortunately, FastLED cannot be compiled for the new version.
Thank you very much!

tueddy, 2024-05-04:
With latest commit + https://github.com/FastLED/FastLED/pull/1619 FastLED compiles and works on Arduino 3-RC1, tested with PlatformIO. I use a 24 NeoPixel with different color animation, everything is fine!

closed 2024-08-30 by zackees