FastLED/FastLED #1557: No information on how to port FastLED to a new platform

rah2501, 2023-11-03:
I've searched the website, the wiki, the FAQ and found no information on how to port FastLED to a new platform.  I would expect at least a paragraph on what one needs to do in the codebase in order to add a new platform.

zackees, 2023-11-04:
If you attempt to compile it you'll get a compiler error pointing you to the correct files.

The platforms can be found in the platform directory.

samguyer, 2023-11-04:
@rah2501 I can probably help put this documentation together. I did the ESP32 support, so I had to go through the process recently (relatively recently, anyway). There's also a considerable difference between adding support for a variant of an existing platform (e.g., a different ARM board) vs a totally new platform. What are you hoping to do?

rah2501, 2023-11-05:
> What are you hoping to do?

To be clear, I created this issue to track the lack of documentation rather than seek help for my own particular porting efforts.  The appropriate place for those discussions would be a separate issue I think.

>  There's also a considerable difference between adding support for a variant of an existing platform (e.g., a different ARM board) vs a totally new platform. 

Can I suggest for consideration, that you could write two responses, one each for both possible answers that I could give, and those responses could be included in the README.md? :-)  That way, you're helping both me and others.

samguyer, 2023-11-05:
Do you need this documentation right now, or not? I think it's a great idea to get this information together, but I'm unclear on what the impetus is at the moment.

rah2501, 2023-11-05:
> Do you need this documentation right now, or not?

That depends on what's in the documentation and the documentation doesn't exist :-)  So difficult to answer..

> I'm unclear on what the impetus is at the moment

The impetus for me to create this issue was the knowledge that in future, people will need to port the project and will find an absence of information on porting, as I did.

samguyer, 2023-11-05:
What I mean, specifically, is: are you currently working on a port or planning to do so in the near future? Do you have a specific platform target in mind?

rah2501, 2023-11-05:
> What I mean, specifically, is: are you currently working on a port or planning to do so in the near future? Do you have a specific platform target in mind?

I don't think my personal activities are relevant to this issue.

samguyer, 2023-11-05:
It is relevant, in the following sense: this is an entirely volunteer-run project, so many of the new features and content come from user contributions, or some collaboration between us and the users who need those features. It is very much a community effort. So, if you want new documentation, you are probably going to need to engage with us a bit more.

samguyer, 2023-11-06:
Actually, it looks like there is an existing (brief) porting document that covers the basics:

https://github.com/FastLED/FastLED/blob/master/PORTING.md

rah2501, 2023-11-06:
> if you want new documentation

I don't want new documentation for my benefit, I want to help the FastLED project identify a barrier-to-entry which can be lowered, benefitting all.

zackees, 2023-11-06:
If we spell it out on how to port, then there could be a lot of people attempting to do this but won't be advanced enough to actually get the platform ported and will abandon the work, but not before asking for a lot of help.

My personal assumption is that if you are advanced enough to compile for a toolchain, then you'll figure out the missing files pretty quickly. But I'm open to having a minority opinion on this.

rah2501, 2023-11-07:
> If we spell it out on how to port, then there could be a lot of people attempting to do this but won't be advanced enough to actually get the platform ported and will abandon the work, but not before asking for a lot of help.

O_o  WTF?!

samguyer, 2023-11-14:
@rah2501 Did you look at the PORTING.md file? It's obviously a little thin, but it is there. Do you want to keep this issue open?

rah2501, 2023-11-14:
I've made a pull request to add a pointer to the PORTING.md which would close this issue.

samguyer, 2023-11-14:
OK, excellent. Thank you.

PR #1559 (2023-12-18): [README.md] Add a pointer to PORTING.md
Make a start (however miniscule) on providing information about porting FastLED to new platforms.

Fixes: #1557
commit e55ef4dd (2023-12-18): [README.md] Add a pointer to PORTING.md

Provide a pointer to PORTING.md in the README in order to help out
anyone looking there for information.

Fixes: #1557
closed 2023-12-18 by kriegsman