FastLED/FastLED #1558: I2S in ESP32 S2

androot2, 2023-11-04:
Compiling error in Arduino 2.2.1 and FastLED library version 3.5.0:

in file: clockless_i2s_esp32, function i2sInit()
lines:
 i2s = &I2S1;
periph_module_enable(PERIPH_I2S1_MODULE);
interruptSource = ETS_I2S1_INTR_SOURCE;
i2s_base_pin_index = I2S1O_DATA_OUT0_IDX;
and
i2s->clkm_conf.clka_en = 0;

zackees, 2023-11-04:
Please dump the entire error

androot2, 2023-11-04:
In file included from c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:10,
                 from c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms.h:42,
                 from c:\Users\andro\Documents\Arduino\libraries\FastLED\src/FastLED.h:62,
                 from M:\Ardunio\ESP\lozko_WS2812_ESP32_OTA\lozko_WS2812_ESP32_OTA.ino:39:
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:90:17: note: #pragma message: NOTE: ESP32 support using I2S parallel driver. All strips must use the same chipset
 #pragma message "NOTE: ESP32 support using I2S parallel driver. All strips must use the same chipset"
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:10,
                 from c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms.h:42,
                 from c:\Users\andro\Documents\Arduino\libraries\FastLED\src/FastLED.h:62,
                 from M:\Ardunio\ESP\lozko_WS2812_ESP32_OTA\lozko_WS2812_ESP32_OTA.ino:39:
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::i2sInit()':
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:458:20: error: 'I2S1' was not declared in this scope
             i2s = &I2S1;
                    ^~~~
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:458:20: note: suggested alternative: 'I2S0'
             i2s = &I2S1;
                    ^~~~
                    I2S0
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:459:34: error: 'PERIPH_I2S1_MODULE' was not declared in this scope
             periph_module_enable(PERIPH_I2S1_MODULE);
                                  ^~~~~~~~~~~~~~~~~~
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:459:34: note: suggested alternative: 'PERIPH_I2S0_MODULE'
             periph_module_enable(PERIPH_I2S1_MODULE);
                                  ^~~~~~~~~~~~~~~~~~
                                  PERIPH_I2S0_MODULE
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:460:31: error: 'ETS_I2S1_INTR_SOURCE' was not declared in this scope
             interruptSource = ETS_I2S1_INTR_SOURCE;
                               ^~~~~~~~~~~~~~~~~~~~
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:460:31: note: suggested alternative: 'ETS_I2S0_INTR_SOURCE'
             interruptSource = ETS_I2S1_INTR_SOURCE;
                               ^~~~~~~~~~~~~~~~~~~~
                               ETS_I2S0_INTR_SOURCE
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:461:34: error: 'I2S1O_DATA_OUT0_IDX' was not declared in this scope
             i2s_base_pin_index = I2S1O_DATA_OUT0_IDX;
                                  ^~~~~~~~~~~~~~~~~~~
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:461:34: note: suggested alternative: 'I2S0O_DATA_OUT0_IDX'
             i2s_base_pin_index = I2S1O_DATA_OUT0_IDX;
                                  ^~~~~~~~~~~~~~~~~~~
                                  I2S0O_DATA_OUT0_IDX
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_i2s_esp32.h:488:24: error: 'volatile union i2s_dev_s::<unnamed>' has no member named 'clka_en'
         i2s->clkm_conf.clka_en = 0;
                        ^~~~~~~
In file included from c:\Users\andro\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from M:\Ardunio\ESP\lozko_WS2812_ESP32_OTA\lozko_WS2812_ESP32_OTA.ino:39:
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/fastspi.h: At global scope:
c:\Users\andro\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

samguyer, 2023-11-05:
@androot2 The pragma messages are just informative (you can ignore them). But the errors are more interesting. What kind of ESP32 board do you have? There are a few variations of the ESP32 now, and they have slightly different hardware peripherals.

androot2, 2023-11-06:
I'm sorry for this bolded message. It was done by github, when I copy&paste that. I know that pragma message is only information.
It's compiled on board ESP32 S2 mini, this purple one from Wemos.

samguyer, 2023-11-06:
Following up: I think the latest version of the ESP-IDF is breaking our driver, so we'll need to look into that. Thanks!

closed 2025-01-13 by zackees