FastLED/FastLED #1562: Feature Request: `CFastLed::canShow(void)`

jojo2357, 2023-11-09:
In order to reduce busy-waiting, it would be helpful to determine if the following will block execution:

https://github.com/FastLED/FastLED/blob/3a03742a09aeb219a065954d7f85be9cdd2582f0/src/FastLED.cpp#L55-L56

While it is currently possible to keep my own counter to determine if blocking will happen, it would be much cleaner to recycle what is already there.

samguyer, 2024-03-12:
Thanks @robertlipe . @jojo2357 Does this address your need?

zackees, 2024-09-20:
Sending commands from core to driver is required and challenging enough across product lines.

Reading back that state from driver to core is sometimes even harder. Partial implementation means different behaviors across chipsets.

Additionally there are some work arounds, such as using interrupts in your code to drive background processing, or measuring the last frame time and assuming this will be roughly the same. Could we do this? Not reliably as some of the Attiny0/1 boards have defined CLOCK_SOURCE=0 meaning they don't even have a millis counter.

closed 2024-09-20 by zackees