FastLED/FastLED #1563: First led is blinking and other errors in the strip

nedasoo, 2023-11-10:
Hey,
probably a hundreth issue about this problem, but few days of research are not giving me any fixes.
I'm building a project that consists of a WS2811 strip, a esp8266 (nodemcu v1) module and a socket.io server. I will create a frontend that would send which pixels to light up and what color to the server, and the server transmits this data to the esp8266 which turns on the LEDs. However, my strip starts acting weirdly when I add WIFI connection. First LED pretty much flickers all the time, but if there a lot of color updates on the strip, the entire strip flickers. I tried putting together a simple FastLED "Cylon" example with wifi connection and attached the result video.

Code:
```#include <Arduino.h>

#include <ESP8266WiFi.h>
#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>
#include <ArduinoJson.h>

#define LED_COUNT 150
#define LED_PIN D1
CRGB leds[LED_COUNT];
//CLEDController *ledController;

#define USE_SERIAL Serial
bool WifiConnected = false;
bool SocketConnected = false;

void setup() {
  USE_SERIAL.begin(115200);
  FastLED.addLeds<WS2811, LED_PIN, RGB>(leds, LED_COUNT);
  FastLED.setBrightness(150);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFi.setPhyMode(WIFI_PHY_MODE_11G);
  WiFi.begin("SSID", "PASS");
}

int num=0;
void loop() {

  static uint8_t hue = 0;
  for(int i = 0; i < LED_COUNT; i++) {
		// Set the i'th led to red 
		leds[i] = CHSV(hue++, 255, 255);
		// Show the leds
		FastLED.show(); 
		// now that we've shown the leds, reset the i'th led to black
		// leds[i] = CRGB::Black;
		fadeall();
		// Wait a little bit before we loop around and do it again
		delay(10);
	}

  for(int i = (LED_COUNT)-1; i >= 0; i--) {
		// Set the i'th led to red 
		leds[i] = CHSV(hue++, 255, 255);
		// Show the leds
		FastLED.show();
		// now that we've shown the leds, reset the i'th led to black
		// leds[i] = CRGB::Black;
		fadeall();
		// Wait a little bit before we loop around and do it again
		delay(10);
	}
}

void fadeall() { for(int i = 0; i < LED_COUNT; i++) { leds[i].nscale8(250); } }```

In below video you can see how initially it works fine but as soon as it connects to the wifi, it starts blinking and displaying wrong colors.
VIDEO 1 (running cylon example, starts glitching upon connecting to wifi) : https://streamable.com/r2ms7d
VIDEO 2 (script making leds turn on one by one, first led blinking green): https://streamable.com/dpv9nl
VIDEO 3 (data sent from socket.io server, turns leds one by one. First led wrong color, last one is red even though nothing is set to red): https://streamable.com/j2tftb

If I remove `#define FASTLED_ALLOW_INTERRUPTS 0` the entire strip goes crazy and looks like its not even grounded properly.

**What could be the problem and possible solutions?** 

I have a spare esp8266 module laying around, I was thinking maybe I could use one for receiving the data and other for setting the LEDs, both controllers connected via I2C? Would that be better?

samguyer, 2023-11-10:
Hi @nedasoo . Yes, this is a long-standing issue with WiFi. It uses very frequent, high-priority interrupts that tend to severely disrupt the timing of the clockless LED strips, like the WS281x family. We don't have a great solution right now -- I usually recommend trying the allow-interrupts-0 thing that you already have in there. It might be worth posting a question on the FastLED Subreddit to see how others are dealing with it.

samguyer, 2023-11-10:
Also, just FYI, there is a "fade all" function in the library already:

fadeToBlackBy(leds, LED_COUNT, 10);

nedasoo, 2023-11-10:
> Hi @nedasoo . Yes, this is a long-standing issue with WiFi. It uses very frequent, high-priority interrupts that tend to severely disrupt the timing of the clockless LED strips, like the WS281x [...]

Thanks for your answer, do you by any chance know if any LED libraries also have the issue?
I'll try to post in the subreddit.

samguyer, 2023-11-12:
@nedasoo I don't think anyone has a good solution to the interrupt problem on the ESP8266, in general. But I'm surprised that disabling interrupts doesn't work better (although it might mess up the WiFi).

Shumatic, 2023-11-13:
Should the above code show the issue?
I have tried it on a generic 8266, i dont see any issue.

samguyer, 2023-11-14:
@Shumatic I wonder if it depends on how much WiFi traffic is actually going in/out of the device (although the sketch above doesn't show any)

Shumatic, 2023-11-14:
@samguyer 
I have added my credentials to the sketch, and it runs fine, both before and after conn. And also without define interrupt
I have 4 neoclock with ntp and a webportal, and I haven't seen any glitches with that either. Thought fastled.show is in loop.

Jaibpi96, 2023-12-12:
¡SOLUCIONADO! El problema esta en el core de ESP8266 para arduino. Haciendo downgrade a version 2.7.4 vuelve a funcionar como antes. Voy a probar versiones mas recientes pero 2.7.4 funciona con FastLed sin problemas.

Para quien no sepa cambiar la versión del core : Herramientas/Placa/Gestor de tarjetas y ahi buscamos esp8266 y cambiamos a versión mas antigua.

 SEÑORES DE ESPRESSIF ME HABEIS VUELTO LOCO DURANTE DIAS...

zackees, 2024-08-31:
> ¡SOLUCIONADO! El problema esta en el core de ESP8266 para arduino. Haciendo downgrade a version 2.7.4 vuelve a funcionar como antes. Voy a probar versiones mas recientes pero 2.7.4 funciona con [...]

😂 Worth translating :

```
Solved! The issue is with the ESP8266 core for Arduino. Downgrading to version 2.7.4 makes it work like before. I'm going to try more recent versions, but 2.7.4 works with FastLED without any problems.

For those who don't know how to change the core version: go to Tools/Board/Board Manager, then search for ESP8266 and switch to an older version.

Espressif team, you've driven me crazy for days...
```

closed 2024-08-31 by zackees