FastLED/FastLED #1566: Running 2 strips at the same time

joebataz, 2023-11-24:
So is it possible to run 2  strips at the same time?
I have 2 strips (APA102C), one is 240 LEDs, the second is 120 LEDs. I want  pattern to update in the smaller strip every other time the larger strip updates.
I have the code working using the Adafruit libraries but it's MUCH too slow. Since I am running on a Giga I can't use FastLed. I have TONS of code that I have written for FastLed and I can't use ANY of it on the Giga which I wouldn't use if I could run 2 strips as I describe above.
If there IS code for this usage I would really appreciate a pointer to it.

Thanks,

Joe B

samguyer, 2023-11-27:
Hi @joebataz ! You can definitely drive multiple strips in parallel using FastLED -- it's one of the big selling points. I would recommend going to the FastLED subreddit for community support in getting it coded up. 

What is the Giga platform? Is it not (yet) supported on FastLED? Or is there some other reason FastLED doesn't run on it?

imirzov, 2024-03-01:
I have tested FastLed with Arduino Pro Mini, Nano and Mega 2560. It works perfectly. Yep, unfortunately Giga is not yet supported. But anyway, FastLed is a great library. Dear authors, thank you so much!

samguyer, 2024-03-12:
@joebataz Did you get it working? Can we close this issue, or do you still need help?

joebataz, 2024-03-13:
Yes and no. I had to switch over to a Mega 2560 which was considerably slower but it did work.
Sorry for the delay in updating!
Best,
Joe B

marmilicious, 2024-03-13:
@joebataz  If your project allows for it, definitely look into using a Teensy or an ESP32. Great speed. Not all the flavors of ESP32 controllers are supported by FastLED yet, but the Teensys are.  
www.prjc.com for the Teensy boards.

If you didn't yet, come visit the FastLED reddit group if you have questions. Lot of people there to help out.
https://www.reddit.com/r/FastLED

joebataz, 2024-03-14:
Hi Marc,

The reddit interface makes me crazy! Started writing a post and wanted to go back and pull something from another post (copy/paste) and spent 10 minutes trying  to get back to my saved draft post. Never found it!
So let me ask you; using either a Teensy or ESP32 can I directly drive the Clock and Data busses on an APA102C strip. I have a 5V 60A supply set up as power to the strips (8 @ 200 LEDs). Most of what I see are WS2xxxxx Posts.

Thanks! Back to see if I can figure reddit out………

Joe B

marmilicious, 2024-03-14:
@joebataz  Yes, you can use a Teensy or ESP32 to control your APA102C pixels. All the code you've written will work fine. 

Reddit was a bit frustrating for me at first too. Do give it a few more goes, share your project, and ask about issues you run into there.

closed 2024-10-31 by zackees