FastLED/FastLED #1569: Where is the full documentation?

jwhendy, 2023-12-02:
I returned to a project I wrote last year (Christmas star) and am fuzzy on fastled usage. I recall a nice, fully documented code base so that I could study e.g. `fill_rainbow()`. I'm now finding no such thing?

I find this circular reference:
- http://fastled.io/docs/index.html: "Check out the official site http://fastled.io for links to documentation, issues, and news"
- that takes me to http://fastled.io/, where I click what seems like the obvious "Documentation" in the upper right menu
- that takes me right back to http://fastled.io/docs/

The wiki also links to an "API Reference" but that yields this not found page (but looks like it would get me to the circular link above). https://github.com/FastLED/FastLED/wiki/API-Reference

Is there full documentation available anymore?

marmilicious, 2023-12-02:
While at https://fastled.io/docs/ and search for `fill_rainbow` for example you'll get to:
https://fastled.io/docs/class_c_pixel_view.html#a1d540a02f3c7e98564050448ec251fbe

What sort of format are you remembering?

Also, feel free to ask questions on the FastLED subreddit.

jwhendy, 2023-12-03:
That is equivalent to what I remember. I admit to groking the TOC and not seeing something intuitive to "here's the full documentation." Sorry for the noise, for whatever reason the existence of Modules and Classes didn't click, and the current docs page struck me as a very high level overview. The circular more info -> back to this overview, plus the dead link made me wonder if the function level docs were broken.

Thanks for the assistance.

closed 2023-12-03 by jwhendy